GreenWaves-Technologies/gap_sdk #170: PMSIS performance counter not working using FreeRTOS

plat-zhaw, 2020-10-15:
Hi,

I wrote a simple program for the GAPuino V2 to measure the performance of a for-loop. This is the program code (main.c);
```
#include "pmsis.h"

// running on fabric controller
void fabric_controller(void) {
    uint32_t loop_cnt = 0;

    printf("start loop performance measurement\n");
    pi_perf_conf((1 << PI_PERF_CYCLES) | (1 << PI_PERF_ACTIVE_CYCLES));
    pi_perf_reset();
    pi_perf_start();
    for (loop_cnt = 0; loop_cnt < 1000000; loop_cnt++);
    pi_perf_stop();
    uint32_t act_cycles = pi_perf_read(PI_PERF_ACTIVE_CYCLES);
    uint32_t tot_cycles = pi_perf_read(PI_PERF_CYCLES);
    printf("loop finished with counter = %i\n", loop_cnt);
    printf("loop performance: active cycles = %i, total cycles = %i\n", act_cycles, tot_cycles);

    pmsis_exit(0);
}

// program entry
int main(void) {
    printf("\n\n\t *** PMSIS performance measurement ***\n\n");
    return pmsis_kickoff((void*)fabric_controller);
}
```
This the content of the make file:
```
APP		                = pmsis_perf
# App sources
APP_SRCS               = main.c
# App includes
APP_INC	         	=
# Compiler flags
APP_CFLAGS       	=
# Linker flags
APP_LDFLAGS      	=
# Custom linker
APP_LINK_SCRIPT		=
# execute the PMSIS make file of the SDK
include $(RULES_DIR)/pmsis_rules.mk
```

When I build and compile the program using FreeRTOS, the performance measurement does clearly not work:
```
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2/pmsis_perf$ make clean
rm -f /home/immanuel/Documents/GAP8/gapuino_v1_2/pmsis_perf/BUILD/GAP8_V2/GCC_RISCV/flash.img
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2/pmsis_perf$ make PMSIS_OS=freertos platform=board io=host all
    CC main.c
    ASM  gap8_iet.S
    ASM  startup_gap8.S
    ASM  cluster_core.S
    ASM  port_asm.S
    CC  system_gap8.c
    CC  gap8_it.c
    CC  gap_io.c
    CC  string.c
    CC  stdlib.c
    CC  errno.c
    CC  cl_malloc.c
    CC  cl_to_fc_delegate.c
    CC  fc_to_cl_delegate.c
    CC  cl_team.c
    CC  hyperbus_cl_internal.c
    CC  uart_cl_internal.c
    CC  cl_dma_irq.c
[... 155 lines cut ...]
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
Loading binary through JTAG
Info : tcl server disabled
Info : telnet server disabled
Warn : Burst read timed out

         *** PMSIS performance measurement ***

start loop performance measurement
loop finished with counter = 1000000
loop performance: active cycles = 10000236, total cycles = 10000229
```

I am using the SDK release 3.7.1

Is this a bug? Or am I doing something wrong?

On another note: I observed that I can use the PMSIS_API with different OS's (FreeRTOS, PULP-OS, MBED-OS). Is it possible to use no OS? How would I do that and can I use PMSIS_API without an OS?

Thanks for the help

sivasivarajah, 2020-10-15:
Hi @plat-zhaw ,

The perf mesured correct in this case because of compiler optimizations. In your test, you are using a for loop without any actions in the loop, so the compiler has optimized and erased the loop from bimary(which is useless then). You should put a `volatile` to the variable you use to control the for loop, this will avoid aggressive compiler optimization.

PMSIS is indeed designed to work on multiple OS, for the moment AliOs, FreeRTOS, Pulp-Os. Mbed-Os is not supported any more.
You need an OS to use PMSIS, it does not work alone. To put it simple, it's a layer between application code and OS.

Regards.

plat-zhaw, 2020-10-16:
HI @sivasivarajah ,

Thanks for the fast answer. You are quite right about the compiler optimizing my loop away. And yes, I did think about that before posting this here. I tried several things as well in order to trick the compiler into not removing the loop, but I guess the compiler saw through all my attempts. And later I thought it can not be an optimization issue, since the loop was not removed using PULP-OS. While nosing around in the makefile freeRTOS_rules.mk I discovered that the flag for size optimization is set there, which is only executed if you specify FreeRTOS as the PMSIS_OS. That might explain why it did not happen when I used PULP-OS (or I just did not find the flag, because the make files are quite complex).

Anyway, I tried using the `volatile` keyword and it worked fine. Thanks for the tip. It never occured to me to use the volatile keyword to prevent the compiler from optimizing busy-loops away.
```
void fabric_controller(void) {
    volatile uint32_t loop_cnt = 0;

    printf("start loop performance measurement\n");
    pi_perf_conf((1 << PI_PERF_CYCLES) | (1 << PI_PERF_ACTIVE_CYCLES));
    pi_perf_reset();
    pi_perf_start();
    for (loop_cnt = 0; loop_cnt < 10000000; loop_cnt++);
    pi_perf_stop();
    uint32_t act_cycles = pi_perf_read(PI_PERF_ACTIVE_CYCLES);
    uint32_t tot_cycles = pi_perf_read(PI_PERF_CYCLES);
    printf("loop finished with counter = %i\n", loop_cnt);
    printf("loop performance: active cycles = %i, total cycles = %i\n", act_cycles, tot_cycles);

    pmsis_exit(0);
}
```
And the console output while running the program on the board:
```
(base) immanuel@ubuntu1-VirtualBox:~/Documents/GAP8/gapuino_v1_2/pmsis_perf$ make platform=board io=host run
gapy --target=gapuino_v2 --platform=board --work-dir=/home/immanuel/Documents/GAP8/gapuino_v1_2/pmsis_perf/BUILD/GAP8_V2/GCC_RISCV    run --exec-prepare --exec --binary=/home/immanuel/Documents/GAP8/gapuino_v1_2/pmsis_perf/BUILD/GAP8_V2/GCC_RISCV/pmsis_perf 
Launching execution with command:
gap8-openocd -c 'gdb_port disabled; telnet_port disabled; tcl_port disabled' -f interface/ftdi/gapuino_ftdi.cfg -f target/gap8revb.tcl -f tcl/jtag_boot.tcl -c 'gap8_jtag_load_binary_and_start "/home/immanuel/Documents/GAP8/gapuino_v1_2/pmsis_perf/BUILD/GAP8_V2/GCC_RISCV/pmsis_perf" elf'
Open On-Chip Debugger 0.10.0+dev-00031-g6ec2d931 (2020-10-02-13:21)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
Loading binary through JTAG
Info : tcl server disabled
Info : telnet server disabled

         *** PMSIS performance measurement ***

start loop performance measurement
loop finished with counter = 10000000
loop performance: active cycles = 110406001, total cycles = 110406093
```

What I usually do to prevent the compiler from removing my busy-loops is using inline assembly like this:
```
void fabric_controller(void) {
    uint32_t loop_cnt = 0;

    printf("start loop performance measurement\n");
    pi_perf_conf((1 << PI_PERF_CYCLES) | (1 << PI_PERF_ACTIVE_CYCLES));
[... 35 lines cut ...]
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
Loading binary through JTAG
Info : tcl server disabled
Info : telnet server disabled

         *** PMSIS performance measurement ***

start loop performance measurement
loop finished with counter = 10000000
loop performance: active cycles = 20073453, total cycles = 20073553
```
Interestingly, I observed that using the assembly instruction  I get approximately what I expected in terms of used cycles for the loop. Because there is probably an instruction which compares, jumps and increments in just one cycle. Adding the "nop" makes about two instructions per loop-cycle (not counting the loops overhead and clean-up). Using the volatile I get around 11 clock cycles per loop-cycle. There is probably some additional stuff the compiler does when you use the `volatile` keyword.

**About the OS:**
If I understood correctly it is not possible to use the PMSIS_API and PMSIS_BSP without an OS. Is that correct?
Secondly, how do I program the GAP9 without any OS? Is there an example for that? Because I could not find it.
Third, if the MBED-OS is not supported anymore it would be good to remove it from the documentation and the various stuff from the SDK (because it is still there and I tried to compile my code using PMSIS_OS=mbed, which of course did not work).

Thanks very much for the help.

sivasivarajah, 2020-10-20:
Hi @plat-zhaw, indeed using nop instructions are better for those loops, using `volatile` keyword forces to read/write from memory(and not cache/regs) thus if the variable was in L2 RAM, access time is between 8 and 10 cycles.

That is correct, an OS is needed for PMSIS_API and PMSIS_BSP.
We do not provide any soft to control microcontroller without OS(since it is needed to handle time, IRQ,...), but Pulp-Os may be an alternative. It is an OS built for GAP8, optimized for the events(while FreeRTOS is fully preemptive, with quantum of exec).
Do you have needs to use without OS?

Thanks for the info, doc about Mbed should be removed.

Regards.

plat-zhaw, 2020-10-21:
Hi @sivasivarajah, thanks for the your answer.

So you basically say the PULP-OS is optimized for the GAP8 and should have less overhead than FreeRTOS? What I noticed is that if I compile my program with FreeRTOS the image size is considerably larger than when I compile with PULP-OS (image size with PULP-OS is 28704 and with FreeRTOS is 32800 for the program posted above).

In the PULP-OS doc there is some explanation about the code in the SDK folder examples/native/pulpos/helloworld. This example uses only functions provided be the rt_api to execute some stuff on the cluster core. The same can be done using PMSIS_API (exmples/pmsis/helloworld is practically the same program). Is the rt_api faster (less overhead) than the PMSIS_API? Or what are the differences?

What I am trying to do is to execute a certain program and for the execution of that program it is crucial to use as less energy as possible. Because my energy source is from energy harvesting only. My reasoning is that every OS introduces some overhead which of courses uses energy to execute and there are some latencies (at this point not the main problem). So to really get the best performance in terms of energy consumption for a certain program I would like  to program the GAP8 without any OS which takes up resources. Further more I do not need an OS.

Thanks for the help.

sivasivarajah, 2020-10-23:
Hi @plat-zhaw ,

Indeed like you've observed, PulpOs is more optimized for GAP8. FreeRTOS is a widely used RTOS by various chips and microcontrollers while PulpOs is designed for GAP8 like platform. FreeRTOS will likely consume more energy because of its preemptive feat, although with latest updates and tickless feature, it will consume less(less IRQ from SysTick therfore less instructions executed, core sleeps more).
I think it depends also if you need kernel management in your application(threads, timing, and other sync mechanism). If it is needed FreeRTOS seems suitable, otherwise PulpOs is fine.

RT_API is the old API used on PulpOs, it is not maintained any more, instead consider using PMSIS_API. PMSIS_API will act as a wrapper to RT_API when an implementation exists otherwise. In terms of perf, both API are nearly equivalent.

In any, use PMSIS  API for your application, you can then measure perf between PulpOs and FreeRTOS.

plat-zhaw, 2020-10-23:
Dear @sivasivarajah,

Thanks for the really useful information.

Since the RT_API is not maintained anymore please update the SDK doc of the PULP-OS. The example in the doc uses exclusievly the RT_API. Based on that I just spent two days developing code using the RT_API, which I know now, thanks to you, is good for nothing. That is really frustrating. Please don't take this personally. I just wrote this to you since you are the one who answered me and I have no idea who is responsible for the outdated documentation. But you guys should really fix that and try in the future to keep the doc up to date. Because information is already sparse enough as it is and as I just told you it can lead to a lot of wasted time. Furthermore, it leaves me with a bad aftertaste, because now I have the feeling, that I can not rely on the doc and never know if it's accurate or not.

As I said, please don't take this personally. I'm really grateful for your help and the information you provided. Tank you.

Regards

sivasivarajah, 2020-10-23:
Hi @plat-zhaw ,

Even if RT_API is not maintained, you can still use them with PulpOs. It is still kept for backward compatibility that is why the example you mentioned is still in examples/native/pulpos because it uses native API(RT_API), but indeed all new features and implementations will only be PMSIS_API, examples are in examples/pmsis.

Regards.

plat-zhaw, 2020-10-26:
Hi @sivasivarajah,

I understand und respect that. Nonetheless, in my opinion it would be good to mention in the SDK's doc, that the RT_API is deprecated and one should use PMSIS_API for future development.

Regards

closed 2020-11-03 by sivasivarajah