GreenWaves-Technologies/gap_sdk #171: rt_cluster_call default stack sizes

plat-zhaw, 2020-10-21:
Hi,

I just started programing using rt_api on a GAPuino V2 board. The function `rt_cluster_call` is used to execute some stuff on the cluster. There you have to specify the stacks for the cluster master and slaves or if NULL the default is used. Now, the documentation states:
```
** \brief Call a function that will execute on the cluster.
 *
 * This will wake-up the master core (core 0) of the specified cluster and make it execute the specified function.
 * This function is intended to be used for coarse-grain job delegation to the cluster side, and thus the stack
 * used to execute this function must also be specified.
 * When the function starts executing on the cluster, the other cores of the cluster are also available
 * for parallel computation. Thus the stacks for the other cores (called slave cores) can also be specified,
 * as well as the number of cores which can be used by the function on the cluster (including the master).
 *
 * Note that this enqueues a function execution. To allow cluster executions to be pipelined 2 can be enqueued at the same time.
 * If 2 functions are queued, as soon as the first is finished, the master core immediately continues with the next one,
 * while the fabric controller receives the termination notification and can enqueue a new execution, in order
 * to keep the cluster busy.
 *
 * \param call    A pointer to the cluster call structure, needed by the runtime to manage the call. This is only needed if more than 2 calls are enqueued at the same time, otherwise it can be NULL.
 * \param cid     The identifier of the cluster on which to call the function.
 * \param entry   The function to be called.
 * \param arg     The argument of the function to be called.
 * \param stacks  The stacks of the cores. This must contain the master stack and one stack for each slave core. If NULL the stack will be allocated and freed automatically. The free occurs the next time that rt_cluster_call is invoked.
 * \param master_stack_size  Stack size of the master core which will execute the function. If stacks is NULL can be 0 to indicate to use the default master stack size. See rt_cl_master_stack_size_get().
 * \param slave_stack_size   Stack size of each core associated to the execution of the function. If stacks is NULL can be 0 to indicate to use the default slave core stack size. See rt_cl_slave_stack_size_get().
 * \param nb_pe   Number of cores involved in the execution of the function. Can be 0 to select the maximum number of cores.
 * \param event   An event to specify how to be notified when the cluster function has returned. If NULL, the function will only return when this is done. Otherwise the event can be used to either call a callback or to wait on the event afterwards.
 */
int rt_cluster_call(rt_cluster_call_t *call, int cid, void (*entry)(void *arg), void *arg, void *stacks, int master_stack_size, int slave_stack_size, int nb_pe, rt_event_t *event);
```
But I can't find the two functions `rt_cl_master_stack_size_get()` and `rt_cl_slave_stack_size_get()` anywhere and of course the compiler tells me, that it doesn't know them. How can I get these default stack sizes?

Any help is much appreciated, thanks.

sivasivarajah, 2020-10-23:
Hi @plat-zhaw ,

RT_API is not maintained any more, you should use PMSIS_API instead.

When sending a task to cluster, the cluster_task will have info about master and slave cores stack size.
Default size are 2kB for master core and 1kB for slave cores.

Regards.

plat-zhaw, 2020-10-23:
Dear @sivasivarajah,

Thanks you for the answer. I didn't know that one should not use the RT_API and I just wasted two days coding. Please have a look at the issue "PMSIS performance counter not working using FreeRTOS" in which I made more remarks about the RT_API issue.

Regards

closed 2020-10-23 by plat-zhaw