GreenWaves-Technologies/gap_sdk #172: Transpose convolution compatibility

gemenerik, 2020-10-22:
I am trying to import a custom CNN in nntool, which has transpose convolutions (also known as deconvolution, upconvolution), and get the following warning:

`new_tflite_graph_all - graph has unknown operator TRANSPOSE_CONV and cannot be properly processed`

Running the latest gap_sdk. Are these operations not supported (yet)?

Is there any documentation on supported operations?

sousoux, 2020-10-22:
We do not support TRANSPOSE_CONV at present however this month's release will include RESIZE_BLINEAR and already support RESIZE_NEAREST_NEIGHBOUR which can be used (followed by a conv) to get the same kind of effect.

You can see the operations currently supported looking at [here](https://github.com/GreenWaves-Technologies/gap_sdk/blob/18fb0e0e362eb13c3dcf4a4f7d8a2f4aa26bd647/tools/nntool/importer/tflite/new_tflite_graph_all.py#L1571
) and [here](https://github.com/GreenWaves-Technologies/gap_sdk/blob/18fb0e0e362eb13c3dcf4a4f7d8a2f4aa26bd647/tools/nntool/importer/tflite/new_tflite_graph_all.py#L115).

gemenerik, 2020-10-22:
Thanks for you reply!

I would hereby suggest implementation of the `TRANSPOSE_CONV` operation, as it is part of many nets with decoder architectures (image segmentation, super resolution, optical flow estimation, GANs), some of which more applicable on GAP8 than others.

For those interested, [this article](https://medium.com/apache-mxnet/transposed-convolutions-explained-with-ms-excel-52d13030c7e8) explains transpose convolutions pretty well. A transpose convolution can be done by fully zero padding the input  (e.g. 3 layers of zero padding of the input for a 4x4 filter/kernel, 1 layer for a 2x2 kernel) and taking the regular convolution with 'valid' padding. If your transpose convolution has a stride, then you can add zeros between values of the input matrix (1 layer for a stride of 2). 

Effectively that means that if you have a stride of 1 you could manually pad your tensor using TensorFlow's pad function, which is supported (or even easier; use 'same' padding) and apply a regular conv2d. However, there is no point in using this, as a transpose convolution with stride=1 is nothing but a regular convolution, and does 0 upscaling.

If you have (and you likely do) a stride of >1; I don't except manually adding zeroes in between input tensor values being compatible with nntool (please correct me if I'm wrong). So for now, I don't see another option than using sousoux's suggestion and/or wait for an updated sdk.

closed 2020-10-22 by gemenerik