GreenWaves-Technologies/gap_sdk #174: Errors making sdk 3.7.2

gemenerik, 2020-10-30:
I ran into several errors making the latest sdk (3.7.2), using `make sdk`.

`make[2]: Entering directory '/home/rik/gap_sdk/rtos/pulp/archi_pulp'`
`scons`
`scons: Reading SConscript files ...`
`ImportError: No module named configparser:`
`  File "/home/rik/gap_sdk/rtos/pulp/archi_pulp/SConstruct", line 4:`
`    import pulp_config as plpconfig`
`  File "/home/rik/gap_sdk/tools/gap-configs/python/pulp_config.py", line 19:`
`    import json_tools as js`
`  File "/home/rik/gap_sdk/tools/gapy/json_tools.py", line 23:`
`    import configparser`
`Makefile:2: recipe for target 'build' failed`
`make[2]: *** [build] Error 2`
`make[2]: Leaving directory '/home/rik/gap_sdk/rtos/pulp/archi_pulp'`
`Makefile:7: recipe for target 'do.archi' failed`
`make[1]: *** [do.archi] Error 2`
`make[1]: Leaving directory '/home/rik/gap_sdk/rtos/pulp'`
`Makefile:97: recipe for target 'pulp-os' failed`
`make: *** [pulp-os] Error 2`

The install refuses to use my activated Conda install, nor my base Conda install and instead uses `/usr/bin/python`, hence the error. The problem was fixed by installing `configparser` to said Python install. I'm sure other people might run into this problem.

Also missing a `requests` dependency for the Conda env.

Finally, my build fails in the making of openocd (still `make sdk`), which I haven't managed to fix.
`git submodule update --init --recursive tools/gap8-openocd`
`cd tools/gap8-openocd && ./bootstrap`
`+ aclocal`
`+ libtoolize --automake --copy`
`+ autoconf`
`+ autoheader`
`+ automake --gnu --add-missing --copy`
`Makefile.am:46: warning: wildcard $(srcdir: non-POSIX variable name`
`Makefile.am:46: (probably a GNU make extension)`
`Setting up submodules`
`Generating build system...`
`Bootstrap complete. Quick build instructions:`
`./configure ....`
`cd tools/gap8-openocd && ./configure --prefix=/home/rik/gap_sdk/install/workstation --program-prefix=gap8-`
`checking for makeinfo... makeinfo`
`checking for a BSD-compatible install... /usr/bin/install -c`
`checking whether build environment is sane... yes`
`checking for a thread-safe mkdir -p... /bin/mkdir -p`
`checking for gawk... no`
`checking for mawk... mawk`
`checking whether make sets $(MAKE)... yes`
`checking whether make supports nested variables... yes`
`checking for gcc... gcc`
`checking whether the C compiler works... yes`
`checking for C compiler default output file name... a.out`
`checking for suffix of executables... `
`checking whether we are cross compiling... no`
`checking for suffix of object files... o`
`checking whether we are using the GNU C compiler... yes`
`checking whether gcc accepts -g... yes`
`checking for gcc option to accept ISO C89... none needed`
`checking whether gcc understands -c and -o together... yes`
`checking for style of include used by make... GNU`
`checking dependency style of gcc... none`
`checking for gcc option to accept ISO C99... none needed`
`checking for ranlib... ranlib`
`checking for pkg-config... /usr/bin/pkg-config`
`checking pkg-config is at least version 0.23... yes`
`checking build system type... configure: error: cannot guess build type; you must specify one`
`Makefile:132: recipe for target 'openocd' failed`
`make: *** [openocd] Error 1`

SuperN1ck, 2020-11-01:
> I'm sure other people might run into this problem.

I did when using the bitcraze docker.

Yaooooo, 2020-11-01:
@gemenerik Thanks a lot. 
I will tag this as bug, and make sure "configparser" in the requirements list in the next release. 

For your openocd build problem you mentioned above, it seems the build cannot detect your system. When I run the command:
`cd tools/gap8-openocd && ./configure --prefix=/home/rik/gap_sdk/install/workstation --program-prefix=gap8-`

I can see it detect my system type:
.......
checking build system type... **x86_64-pc-linux-gnu**

I am using Ubuntu 20.04, may I ask yours? 
You can also compile the openocd by yourself as mentioned in the README: https://github.com/GreenWaves-Technologies/gap8_openocd

gemenerik, 2020-11-02:
> For your openocd build problem you mentioned above, it seems the build cannot detect your system. When I run the command:
> `cd tools/gap8-openocd && ./configure --prefix=/home/rik/gap_sdk/install/workstation --program-prefix=gap8-`

This results in the same error

> I am using Ubuntu 20.04, may I ask yours?

I am running Ubuntu 18.04

> You can also compile the openocd by yourself as mentioned in the README: https://github.com/GreenWaves-Technologies/gap8_openocd

This made me realize I already compiled `openocd` as one of the first steps in the install instructions, as I did in previous sdk builds, in a separate folder, and so I just skipped openocd for the gap_sdk part (so `make all autotiler nntool` instead of `make flash`). So for some reason the gap8_openocd included in tools currently doesn't work for me.

gemenerik, 2020-11-25:
On Ubuntu 20.04 I do not get the error for `make sdk`, either.

Yaooooo, 2020-11-25:
Hi @gemenerik 
Thanks for your fix, I just review the build process, indeed this is required in downloading the autotiler lib.
This package is included by default in 20.04.
I will merge your pull request, and fix it internally to ensure it is fixed in the future release.

closed 2020-11-25 by Yaooooo

LorenzoLamberti94, 2021-02-23:
I had the issue that configparser was installed in my conda env, but still the gap sdk was not finding the packet.
installing configparser with apt-get solved the issue