GreenWaves-Technologies/gap_sdk #175: Mbed SPI async transfer won't call callback

Gualor, 2020-11-01:
Hi,
I am using a `Gapuino v2` board with `Mbed OS` in the `PlatformIO` IDE. 
I'm currently working on a project which requires sampling data at a constant rate (by using a data ready interrupt) from an SPI device (ADS1299) and I need to send batches of data asynchronously via UART. The issue lies with the async nature of the messages I'm exchanching via UART (using rosserial protocol) which could break any SPI blocking function (such as SPI::write) and for this reason I require a function such as `SPI::transfer`.

Using SPI::transfer in the [example](https://github.com/GreenWaves-Technologies/gap_sdk/blob/3.4/examples/native/mbed/test_driver_c%2B%2B/test_SPI_Asynch/test_SPI_Asynch.cpp) provided in the gap_sdk does not give any problem, given the fact that is given (uint8_t *)NULL as an rx buffer that will trigger a callback no matter what. But when I modify the example to work with a receiver buffer no callback function is ever called. I will provide the modified example below:

```c++
#include "mbed.h"

SPI spi(SPI1_MOSI, SPI1_MISO, SPI1_SCLK, SPI1_CSN0_A9); // mosi, miso, sclk, ssel

// Procedure to read ID register on ADS1299, works fine with SPI write
uint8_t buffer_tx[4] = {0x11, 0x20, 0x00, 0x00};
uint8_t buffer_rx[1];

static void spi_callback(int events)
{
    printf("This function is never called!\n");
}

int main() {
    // Setup the spi for 8 bit data, high steady state clock,
    // second edge capture, with a 1MHz clock rate
    spi.format(8, 3);
    spi.frequency(4000000);

    event_callback_t event;

    event = spi_callback;

    spi.transfer((uint8_t *)buffer_tx, sizeof(buffer_tx), (uint8_t *)buffer_rx, sizeof(buffer_rx), event);

    wait(0.01);

    return 0;
}
```

Unfortunately I couldn't find any other examples, could very well be I'm using this function inappropriately.
Thank you in advance for your support, it will be greatly appreciated!

Yaooooo, 2020-11-05:
Dear @Gualor,

We appologize that we don't support and maintain mbed os anymore, and we don't have enough energy to maintain this.
If you would like to use PULP OS or FreeRTOS, we are very happy to support you to port that.
What I can suggest here is:

1. Test RX and TX saperately. which means in this example, you put 1 transfer for RX, and 1 transfer for TX. Then check if it works. SInce it's async way, it should be the same as what you do here.  
2. Dig into the SPI transfer driver implementation in mbed-os.

Sorry again that we cannot do more for your problems on mbed.

Gualor, 2020-11-05:
Dear @Yaooooo,

I know you don't support Mbed anymore, but thank you for your reply! Suggestion #1 really helped as now I can at least write to the registers. The issue now is that after I execute an SPI read (which gives the exact result : 0x3e) all the other attempts to read/write failed, i seems that the status of the SPI is always busy.  

```c++
#include "mbed.h"

SPI spi(SPI1_MOSI, SPI1_MISO, SPI1_SCLK);

uint8_t buffer_tx[1];
uint8_t buffer_rx[1];

int main() 
{
    spi.format(8, 1);
    spi.frequency(10000000);

    buffer_rx[0] = 0x00;
    buffer_tx[0] = 0x06;
    printf("%d\n", spi.transfer((uint8_t *)buffer_tx, 1, (uint8_t *)NULL, 0, 0)); // write
    buffer_tx[0] = 0x11;
    printf("%d\n", spi.transfer((uint8_t *)buffer_tx, 1, (uint8_t *)NULL, 0, 0)); // write
    buffer_tx[0] = 0x20;
    printf("%d\n", spi.transfer((uint8_t *)buffer_tx, 1, (uint8_t *)NULL, 0, 0)); // write
    buffer_tx[0] = 0x00;
    printf("%d\n", spi.transfer((uint8_t *)buffer_tx, 1, (uint8_t *)NULL, 0, 0)); // write
    printf("%d\n", spi.transfer((uint8_t *)NULL, 0, (uint8_t *)buffer_rx, 1, 0)); // read
    printf("result: %x\n", buffer_rx[0]);

    // same 4 writes and 1 read as before 
    buffer_rx[0] = 0x00;
    buffer_tx[0] = 0x06;
    printf("%d\n", spi.transfer((uint8_t *)buffer_tx, 1, (uint8_t *)NULL, 0, 0));
    buffer_tx[0] = 0x11;
    printf("%d\n", spi.transfer((uint8_t *)buffer_tx, 1, (uint8_t *)NULL, 0, 0));
    buffer_tx[0] = 0x20;
    printf("%d\n", spi.transfer((uint8_t *)buffer_tx, 1, (uint8_t *)NULL, 0, 0));
    buffer_tx[0] = 0x00;
    printf("%d\n", spi.transfer((uint8_t *)buffer_tx, 1, (uint8_t *)NULL, 0, 0));
    printf("%d\n", spi.transfer((uint8_t *)NULL, 0, (uint8_t *)buffer_rx, 1, 0));
    printf("result: %x\n", buffer_rx[0]);

    return 0;
}
```

The output of this code is the following:

```
Start execution on GAP8
0    <---- all writes until first read complete and returned 0
0
0
0
0
result: 3e    <---- first read is successful
-1    <---- all other writes after first read failed and returned -1
-1
-1
-1
-1
result: 0    <---- second (and later) reads return wrong result
```

Could be that maybe the SPI rx interrupt is not cleared after the first read? If I never read, spi write seems to work fine.
Since me and my team are required to use Mbed, you are the only help we can use to solve this issue, if you have any other suggestion this would be highly appreciated!

Yaooooo, 2020-11-05:
Pressed wrong button.
Be careful on what you are doing. Since this is a async api, all the transfer will be pushed to uDMA without waiting. And in a HW 
 channel (uDMA), we can maximum enqueue 2 tasks. So max 2 tasks for TX, 2 tasks for RX.
If the previous tasks haven't finished, the next enqueue will crash the previous one.

Gualor, 2020-11-06:
Dear @Yaooooo ,

Yes I know I should be careful about timing, but I don't think timing is the issue here as I demonstrate in the following code:

```c++
#include "mbed.h"

SPI spi(SPI1_MOSI, SPI1_MISO, SPI1_SCLK);

void eventCb(int event);
volatile bool flag = false;
event_callback_t event = eventCb;

// Notify when spi trasfer is completed
void eventCb(int event)
{
    if (event == SPI_EVENT_COMPLETE)
    {
        printf("Transfer completed\n");
        flag=true;
    }
    else
    {
        printf("SPI event status: %d\n", event);
    }
}

// Async write function
void write(uint8_t value)
{
    uint8_t buffer_tx[1] = {value};
    spi.transfer((uint8_t *)buffer_tx, 1, (uint8_t *)NULL, 0, event, SPI_EVENT_ALL);
    while(flag==false); // wait spi
    flag=false;
}

// Async read function
uint8_t read(void)
{
    uint8_t buffer_rx[1];
    spi.transfer((uint8_t *)NULL, 0, (uint8_t *)buffer_rx, 1, event,SPI_EVENT_ALL);
    while(flag==false); // wait spi 
    flag=false;
    return buffer_rx[0];
}

int main() 
{

    spi.format(8, 1);
    spi.frequency(10000000);

    /**
     * all writes are successful
     */
    write(0x06);
    write(0x11);
    write(0x20);
    write(0x00);

    /**
     * stuck in while loop inside read(),
     * ISR is never triggered and flag never
     * set to true.
     */
    uint8_t reg = read();

    printf("Reg: 0x%x\n", reg);

    return 0;
}
```

Which gives the following output:
```
Transfer completed
Transfer completed
Transfer completed
Transfer completed
```
The callback of the read function is never called, in fact in the previous example code the SPI status was "always busy". So as long as the SPI is busy the callback won't be triggered, and that does never happen for some reason.
Any suggestion on how I could debug this? Thanks again

Yaooooo, 2020-11-08:
Hi,

I am not saying that there's no bugs in our implementation. 
However, there's several things you really need to be careful:
1, Mbed OS is a preemptive system, but not reactive system. (Even it has some asynchronous api)
2, When we use Async api and reactive system, we should NEVER block the scheduler. 

There's one thing I don't understant, you are using asynchronous api in a blocking way:
```
    while(flag==false); // wait spi 
    flag=false;
```
In this case, why bother to use asynchronous api?

As you may know, mbed-os is a native preemptive system, it has something on high level for "asychronous" mode, but it's not a pure reactive sys. So it loose lots of benefits of a reactive sys. 

My suggestion is: connect on an Oscilloscope and read the signal MISO and MOSI.

Gualor, 2020-11-09:
Dear @Yaooooo ,
Thank you very much for the response, since in my implementations I had to use thread signal to unlock from an ISR a thread to read data from the ADS1299 at a constant 250Hz (without losses), I couldn't just use an if statement, I just hoped the interrupt would arrive quickly (like it has done with the SPI write commands). The thing I could've done would be dropping the signal and use a boolean flag instead with an if statement and yield at the end of the thread.

Anyway, given the amount of work it would have required to use Mbed (even though it is super useful for quick prototyping), I decided to follow your advice and use freeRTOS instead, requires a little bit of extra work to setup but I got it running in no time.

Thank you very much for the great work, have a nice day!

closed 2020-11-09 by Gualor