GreenWaves-Technologies/gap_sdk #176: Concats with multiple versions of the same input are not supported

gemenerik, 2020-11-10:
Quantized during TensorFlow Lite conversion to a `.tflite` model.
I get the following error loading into `nntool`;

`NotImplementedError: concats with multiple versions of the same input are not supported. This is normally a graph design problem.`
`EXCEPTION of type 'NotImplementedError' occurred with message: 'concats with multiple versions of the same input are not supported. This is normally a graph design problem.'`

I am a bit confused what is meant by this exception. What I can think of is either;

1. Literally what the exception says; I am branching off and concatenating back together. Note that this is desired and not a 'graph design problem' as the exception calls it. Visualized in `netron`:
> [image]
2. The quantization in the TensorFlow Lite conversion does not only add quantization and dequantization at the network input and output, but adds _requantizations_ in my graph before concatenations, but not on all branches that feed into them. So `nntool` maybe thinks there are different types being fed?

> [image]
> All branches requantized

> [image]
> All but one branch requantized

I truly hope it's not the first one, otherwise the types of networks I am looking at are impossible to implement altogether. At least, for now.

Thanks for the help!

gemenerik, 2020-11-11:
I managed to get the requantizations on all branches, and the error persists, so it is likely the branching off and concatenating. Still it is curious that the error does not occur when loading an unquantized model, does it not interpret the architecture before?
>[image]

* Since there is, as far as I know, no reasonable workaround; where can I best leave suggestions for features?

sousoux, 2020-11-25:
Sorry. I only just saw this. Is it possible for you to give us a model with this problem?

gemenerik, 2020-11-25:
No worries! I am working on a network based on FlowNetS for optical flow estimation.

The following zip includes:
- `models.py` where the model is defined in TensorFlow
- `flownetslike`; a saved model dir with `.pb` saved model
- `flownetslike.tflite`; a converted (quantized) tflite model.

[flownetslike.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5596446/flownetslike.zip)

The most important reason to want concatenations of multiple versions of the same input is because it allows one to do optical flow predictions at various resolutions, and concatenate these back onto the feature maps. This is conventional for all state of the art optical flow CNNs, and helps solve the aperture problem while maintaining a high resolution final flow prediction.

gemenerik, 2020-11-26:
This issue is similar to if not the same as the problem mentioned in the [Jun 4th comment](https://github.com/GreenWaves-Technologies/gap_sdk/issues/127#issuecomment-638641809) by @FrontMage on issue #127

sousoux, 2020-12-08:
This was a bug. Fixed in sources. Will be pushed shortly.

closed 2020-12-08 by sousoux