GreenWaves-Technologies/gap_sdk #177: Unable to use freeRTOS with AIdeck

Gualor, 2020-11-15:
Hi,
I recently purchased an AIdeck from bitcraze in order to port a freeRTOS based project I previously developed for the GAPuino v2 to the smaller AIdeck board. 

The Aideck board works fine with almost every example, however, when I specify `PMSIS_OS = freertos` in the Makefile and I run the make command `make clean all` I get the following output with the error `undefined reference to semihost_write0`:
```
rm -f /home/gualor/thesis/git/camlin-thesis/ai_deck/uart_send_char/BUILD/GAP8_V2/GCC_RISCV/flash.img
    CC test.c
    ASM  startup_gap8.S
    ASM  gap8_iet.S
    ASM  cluster_core.S
    ASM  port_asm.S
    CC  gap8_it.c
    CC  system_gap8.c
    CC  gap_io.c
    CC  string.c
    CC  stdlib.c
    CC  errno.c
    CC  cl_malloc.c
    CC  cl_to_fc_delegate.c
    CC  fc_to_cl_delegate.c
    CC  cl_team.c
    CC  hyperbus_cl_internal.c
    CC  uart_cl_internal.c
    CC  cl_dma_irq.c
    CC  fc_event.c
    CC  fll.c
    CC  gpio.c
    CC  pad.c
    CC  pmu.c
    CC  pmu_internal.c
    CC  pwm.c
    CC  pwm_internal.c
    CC  rtc.c
    CC  rtc_internal.c
    CC  timer.c
    CC  hyperbus.c
    CC  hyperbus_internal.c
    CC  cpi.c
    CC  cpi_internal.c
    CC  dmacpy.c
    CC  dmacpy_internal.c
    CC  i2c.c
    CC  i2c_internal.c
    CC  i2s.c
    CC  i2s_internal.c
    CC  spi.c
    CC  spi_internal.c
    CC  uart.c
    CC  uart_internal.c
    CC  perf_internal.c
    CC  pi_log.c
    CC  event_kernel.c
    CC  cl_l1_malloc.c
    CC  fc_l1_malloc.c
    CC  l2_malloc.c
    CC  malloc_external.c
    CC  malloc_internal.c
    CC  pi_malloc.c
    CC  mem_slab.c
    CC  device.c
    CC  pmsis_task.c
    CC  pmsis_backend_native_task_api.c
    CC  port.c
    CC  printf.c
    CC  list.c
    CC  queue.c
    CC  tasks.c
    CC  timers.c
    CC  event_groups.c
    CC  stream_buffer.c
    CC  FreeRTOS_util.c
/home/gualor/thesis/git/camlin-thesis/ai_deck/uart_send_char/BUILD/GAP8_V2/GCC_RISCV//home/gualor/gap_sdk/rtos/freeRTOS/vendors/gwt/gap8/src/device/system_gap8.o: In function `system_init':
/home/gualor/gap_sdk/rtos/freeRTOS/vendors/gwt/gap8/src/device/system_gap8.c:126: undefined reference to `semihost_exit'
/home/gualor/thesis/git/camlin-thesis/ai_deck/uart_send_char/BUILD/GAP8_V2/GCC_RISCV//home/gualor/gap_sdk/rtos/freeRTOS/vendors/gwt/gap8/src/driver/gap_io.o: In function `__semihost_buffer_write_exec':
/home/gualor/gap_sdk/rtos/freeRTOS/vendors/gwt/gap8/src/driver/gap_io.c:365: undefined reference to `semihost_write0'
/home/gualor/gap_sdk/rtos/freeRTOS/vendors/gwt/gap8/src/driver/gap_io.c:365: undefined reference to `semihost_write0'
/home/gualor/gap_sdk/rtos/freeRTOS/vendors/gwt/gap8/src/driver/gap_io.c:368: undefined reference to `semihost_write0'
collect2: error: ld returned 1 exit status
make: *** [/home/gualor/gap_sdk/rtos/freeRTOS/vendors/gwt/rules/freeRTOS_rules.mk:352: /home/gualor/thesis/git/camlin-thesis/ai_deck/uart_send_char/BUILD/GAP8_V2/GCC_RISCV/test] Error 1
```
I don't believe is a bug with the example itself, in fact even with all the code taken out from the source file such as:
```c
int main(void)
{
    return 0;
}
```
with the following Makefile (only the PMSIS_OS line is added to the example):
```Makefile
APP = test
APP_SRCS += test.c 

APP_CFLAGS += -O3 -g -Werror

PMSIS_OS = freertos # <-- compile only if commented

export GAP_USE_OPENOCD=1

include $(RULES_DIR)/pmsis_rules.mk
```
this does not compile.

My setup is the following:
- OS: ubuntu 20.04
- gap_sdk: latest release
- board: AIdeck
- JTAG programmer: Olimex ARM USB TINY H

Thank you in advance for your time.

Best regards,
Lorenzo Gualniera

Yaooooo, 2020-11-15:
Hi,

I confirm there may be some problem on release-v3.5, with AI Deck and freertos.
You can checkout to the latest SDK and use that insead.
However, there's a regression on the wifi streamer on the lateast sdk with ai-deck, it's under fixing.
Besides of that, it should works fine. 

Best,
Yao

Gualor, 2020-11-15:
Hi, 
thank you for the quick response! As I said, I am already at the latest release of the `gap_sdk` (which I believe it to be 3.7). Are the issues with freeRTOS addressed as well in the fix? If not, is there any workaround to get freeRTOS working?
Thank you again.

Best,
Lorenzo Gualniera

Yaooooo, 2020-11-15:
Hi,
I tried the last version of SDK from my end with AIDeck, and I cannot reproduce the problem.
I use this example: gap_sdk/examples/pmsis/helloworld

Gualor, 2020-11-16:
Hi,
to make sure nothing on my PC was clashing with the build, I resumed an old laptop on which I installed a fresh version of Ubuntu 20.04 (updated and upgraded) to reproduce the error, I'll list below all the steps I've done in order to get the error I mentioned:
1) Built latest release of `gap_sdk` following all the steps for the minimal installation with just freeRTOS (ai_deck.sh is sourced before the build)
2) Compiled the same example in ~/gap_sdk/examples/pmsis/helloworld with command `make clean all PMSIS_OS=freertos`, the compilation then failes with the same output I provided in my first message.

[ Note ]: interestingly if I source gapuino_v2.sh (with the same exact build) before making the helloworld example, the compilation is successful. Are there any differences/downsides in sourcing `gapuino_v2.sh` instead of `ai_deck.sh`?

Thank you again @Yaooooo .

Best,
Lorenzo Gualniera

Yaooooo, 2020-11-16:
Hi,

I think it may due to the "minimal installation" and/or a source file missing in the BSP configuration.
FYI, the environment configuration is used for different BSP. 
As you know, on different board, the devices are connected to GAP8 differently, therefore, we use such a configuration mechanism to let the env know which board you are using. 

I quickly do a grep in rtos, and you can see that:

> (base) yao@yao:~/Project/SDK/gap_sdk_github/rtos$ grep -rnI "semihost_write0"
pulp/pulp-os/libs/io/semihost.h:104:void semihost_write0(const char *print_string);
pulp/pulp-os/libs/io/semihost.c:6:void semihost_write0(const char *print_string)
pulp/pulp-os/libs/io/io.c:438:    semihost_write0(__rt_putc_host_buffer);
freeRTOS/vendors/gwt/gap8/include/driver/semihost.h:104:void semihost_write0(const char *print_string);
freeRTOS/vendors/gwt/gap8/src/driver/gap_io.c:78:    semihost_write0((const char *) buffer);
pmsis/pmsis_bsp/fs/host_fs/semihost.h:104:void semihost_write0(const char *print_string);
pmsis/pmsis_bsp/fs/host_fs/semihost.c:6:void semihost_write0(const char *print_string)

The source of `semihost_write0`, where you have this error, is in `pmsis/pmsis_bsp/fs/host_fs/semihost.c`. 
And I believe the error comes from there, this source file hasn't been included in the AI_DECK bsp.

For pulp_os, since this src is in pulp-os libs, so it's not a problem.

Best,
Yao

Yaooooo, 2020-11-16:
Hi,

I found the issue, and fixed it.
I push the fix to the branch 3.8.0_dev (which is a branch prepare for the next release): https://github.com/GreenWaves-Technologies/gap_sdk/commit/43b4b22a810647330d45af857f84c99581846d13

Feel free to checkout to that, it should be fine now.

BR,
Yao

closed 2020-11-16 by Yaooooo

Gualor, 2020-11-16:
> Hi,
> 
> I found the issue, and fixed it.
> I push the fix to the branch 3.8.0_dev (which is a branch prepare for the next release): [43b4b22](https://github.com/GreenWaves-Technologies/gap_sdk/commit/43b4b22a810647330d45af857f84c99581846d13)
> 
> Feel free to checkout to that, it should be fine now.
> 
> BR,
> Yao

Thank you @Yaooooo that's amazing news! I confirm that the built is now successful with freeRTOS.

Best regards,
Lorenzo Gualniera

LorenzoLamberti94, 2021-03-01:
Hi Yao!
Did you push this fix anywhere? I cannot see in the main 3.8.0 dev or 3.8.1 branches.
There are several students facing this problem and fixinf this by hand :)