GreenWaves-Technologies/gap_sdk #178: Unclear instructions on using OpenOCD with GAPuino

JosseVanDelm, 2020-11-16:
Hi there!

I'm trying to deploy [microTVM](https://tvm.apache.org/2020/06/04/tinyml-how-tvm-is-taming-tiny) on the GAPuino (v1) board.
Apparently this requires an OpenOCD bridge to connect through JTAG. But i find the documentation on using OpenOCD very unclear.
I've already gone through the initial setup of the SDK and i've been able to produce this helloworld output:
```
❯ make platform=board io=host run
gapy --target=gapuino --platform=board --work-dir=/home/josse/Thesis/tryout-gap8/gap_sdk/examples/pmsis/helloworld/BUILD/GAP8/GCC_RISCV    run --exec-prepare --exec --binary=/home/josse/Thesis/tryout-gap8/gap_sdk/examples/pmsis/helloworld/BUILD/GAP8/GCC_RISCV/test 
/home/josse/Thesis/tryout-gap8/gap_sdk/tools/gapy/partition/__init__.py:91: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if nargs is not 2:
Launching execution with command:
gap8-openocd -c 'gdb_port disabled; telnet_port disabled; tcl_port disabled' -f interface/ftdi/gapuino_ftdi.cfg -f target/gap8.tcl -f tcl/jtag_boot.tcl -c 'gap8_jtag_load_binary_and_start "/home/josse/Thesis/tryout-gap8/gap_sdk/examples/pmsis/helloworld/BUILD/GAP8/GCC_RISCV/test" elf'
Open On-Chip Debugger 0.10.0+dev-gb84d97ec4 (2020-11-11-19:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 9 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 9 is passed to adv_dbg_unit debug unit
Loading binary through JTAG
Info : tcl server disabled
Info : telnet server disabled
Warn : Burst read timed out

	 *** PMSIS HelloWorld ***

Entering main controller
[32 0] Hello World!
[0 3] Hello World!
[0 0] Hello World!
[0 2] Hello World!
[0 5] Hello World!
[0 6] Hello World!
[0 7] Hello World!
[0 4] Hello World!
[0 1] Hello World!
Cluster master core exit
Test success !
```

I've found this information on OpenOCD on the [greenwaves website](https://greenwaves-technologies.com/setting-up-sdk/)
> OpenOCD for Gap8 is now used instead of plpbridge. There are a few applications which require OpenOCD, as they are using OpenOCD semi-hosting to transfer files with the workstation.
>You have to install the system dependencies required by OpenOCD that you can find here: http://openocd.org/doc-release/README
>There are different cables setup by default for each board. In case you want to use a different cable, you can define this environment variable:
>`export GAPY_OPENOCD_CABLE=interface/ftdi/olimex-arm-usb-ocd-h.cfg`

It tells me that I can use OpenOCD but it doesn't really state how to actually use it.
I have no experience with OpenOCD so I wouldn't know how to proceed.
I'm on Arch Linux so i just installed openocd through my package manager.
But now I don't know what to do...
Also I connected the GAPuino over USB, is this something to take into account? Is this a different cable i have to set?

The [tools/gap8-openocd](https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/tools/gap8-openocd) folder of this repo just contains the standard installation instructions for openocd.
I don't know which installation instructions to follow here.

The [tools/gap8-openocd](https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/tools/gap8-openocd-tools) has this readme which I find very unclear. A lot of stuff is mentioned but I'm really not sure what is meant with some of the references.
> # How to install openocd
>* You will find in this archive a version of openocd ported to support GAP8 basic operations.
>* Installation follows the straightforward ./configure && make && make install procedure
>* You might if you so witch add a PREFIX=My_Prefix option to change default install target (/usr)
>
> # How to use it to run a binary over jtag
> * Now that you have openocd installed you might run binaries and do some basic debug (halt core, monitor regs...)
> * To load a binary in L2 memory and execute using jtag proceed as follow:
> ** In a first terminal execute, considering you are in gap_tools archive: 
> *** openocd -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f target/gap8revb.tcl -f ./tcl_scripts/jtag_boot.tcl
> ** In second terminal run:
> *** telnet localhost 4444
> ** in the telnet terminal invoke command as follows
> *** gap8_jtag_load_binary_and_start /PATH/TO/YOUR/BINARY.elf elf
> *** this will load and run your binary. You can halt the core with the command "halt" or reset with "reset" (will erase the program)

I find this very confusing!
*  > You will find in this archive a version of openocd ported to support GAP8 basic operations.
  
* I don't know what is meant with `this archive` --> is it the gap8-openocd-tools folder? 

* >  Installation follows the straightforward ./configure && make && make install procedure
* I don't know what makefiles from what repository i should use. There is no makefile in the tools/gap8-openocd-tools/ folder.
Also the website above just says to install it via the package manager, while this README mentions a makefile...

* When I run the `openocd -f interface/ftdi/olimex-arm-usb-ocd-h.cfg -f target/gap8revb.tcl -f ./tcl_scripts/jtag_boot.tcl` command inside the gap8-openocd-folder (which is what I think I am supposed to do)
I get this output:
```
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:60: Error: Can't find target/gap8revb.tcl
in procedure 'script' 
at file "embedded:startup.tcl", line 60
```
Also there is no jtag_boot.tcl script inside the tcl_scripts folder. What steps did I miss? What is the actual procedure? Or is there still something wrong with my setup if you look at the helloworld output?
Any help with this is much appreciated! Thanks in advance!

JosseVanDelm, 2020-11-16:
Woops I just see that this was already mentioned in #115
But maybe this issue is more concrete?
Sorry!

Yaooooo, 2020-11-16:
Hi,

We are sorry that our document is not clear and leads your/others misunderstanding.
We definitly need to improve that, and more than welcome if others can contribute.
To make it clear, there are 2 ways to use openOCD with our system:

1. Since the openOCD has been integrated into the SDK with Makefile, therefore, the GAP SDK user no need to deal with natif openOCD command. These make targets are available for you to use openOCD with your board:

- `make flash`: this will use openocd to flash the flashimage.
- `make run`: this command will execute openocd and boot GAP8 from jtag.
- `make all run`: this will build your application, create and flash the flashimage, then execute it.

2, You can use openOCD natif commands as you are trying above. In this case, you need to pay attention on:

- which openocd you are using: maybe you have more than 1 in your system
- where are the necessary scripts, like gap8revb.tcl.

As your problem above, it seems the openocd you are using, doesn't has all the scripts for GAP8. 

Hope this help you clear.
Yao

JosseVanDelm, 2020-11-16:
Hi @Yaooooo,

Thank you very much for your quick reply.
I think it is very nice that OpenOCD has been integrated in the makefile so that developers don't need to worry about setting it up.
However for my purpose (deploying microTVM) I only need to use OpenOCD as TVM should take care of everything else. (At least that is what I believe, currently I haven't got it to run yet).

I've managed to get an OpenOCD instance running myself by looking into the helloworld example. I'll write down my steps here so others having the same issue can refer to it.

## To create a GAPuino OpenOCD session:
1. Make sure during the setup of the SDK to Install the gap8 openocd build. [instructions.](https://github.com/GreenWaves-Technologies/gap_sdk#ubuntu-1804)
2. source the appropriate configuration e.g. in the repository's root folder call `source sourceme.sh` and then select the appropriate board. In my case that was `7 - GAPuino`. This has to be done for every new terminal session you open an openocd session in.
3. The mistake I made was that I tried to use the regular `openocd`, but I had to use `gap8-openocd`
I found this command in the helloworld example:
```
gap8-openocd -c 'gdb_port disabled; telnet_port disabled; tcl_port disabled' -f interface/ftdi/gapuino_ftdi.cfg -f target/gap8.tcl -f tcl/jtag_boot.tcl
```
I don't think i need to disable any port, so I just use:
```
gap8-openocd -f interface/ftdi/gapuino_ftdi.cfg -f target/gap8.tcl -f tcl/jtag_boot.tcl
```
This gives the following output:
```
Open On-Chip Debugger 0.10.0+dev-gb84d97ec4 (2020-11-11-19:35)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 9 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : starting gdb server for gap8.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 9 is passed to adv_dbg_unit debug unit
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

I will not close this issue as I still feel that the documentation is currently lacking on just using OpenOCD.
I don't currently feel comfortable contributing such a part of the guide as I'm not an expert on embedded development, and I also don't have any other boards except for the GAPuino v1 to test it on.
If I get my head around how it works I might create a pull request suggesting better documentation.
If I get stuck again I'll try to post my developments here for later reference.

Best regards.

Yaooooo, 2020-11-16:
Hi,

No worry about contribution, it's our responsability to make the doc better.
For your problem, it's correct that you don't need to disable any port. And your command is correct to connect with openOCD.
However, it miss a command to load the binary and boot from jtag. 
Here's my command openocd to run it in terminal with helloworld example:

> (base) yao@yao:~/Project/SDK/gap_sdk_gitlab/examples/pmsis/helloworld$ gap8-openocd -f interface/ftdi/gapuino_ftdi.cfg -f target/gap8revb.tcl -f tcl/jtag_boot_entry.tcl -c 'gap8_jtag_load_binary_and_start "/home/yao/Project/SDK/gap_sdk_gitlab/examples/pmsis/helloworld/BUILD/GAP8_V3/GCC_RISCV/test" elf 0x1c000080'
Open On-Chip Debugger 0.10.0+dev-gb84d97ec4 (2020-10-24-11:36)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
0
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : starting gdb server for gap8.cpu on 3333
Info : Listening on port 3333 for gdb connections
Loading binary through JTAG
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : Burst read timed out
>
> 
> 
> 	 *** PMSIS HelloWorld ***
> 
> Entering main controller
> [32 0] Hello World!
> Cluster master core entry
> [0 4] Hello World!
> [0 5] Hello World!
> [0 1] Hello World!
> [0 2] Hello World!
> [0 7] Hello World!
> [0 0] Hello World!
> [0 6] Hello World!
> [0 3] Hello World!
> Cluster master core exit
> Test success !

As you can see the command here, we need:
`-f tcl/jtag_boot_entry.tcl -c 'gap8_jtag_load_binary_and_start`
for load the binary and boot from jtag.

There's another thing concern me, it's you mentioned you are using a **GAPuino V1**. If it's the V1 version, it may have some hw issues with openOCD. To double confirm that, could you please contact me through this email: support@greenwaves-technologies.com

Please send us your command detail:
- Command date
- Contact name and email address
- Order name

Best regards,
Yao

JosseVanDelm, 2020-11-17:
Hi @Yaooooo 

Thanks for your elaborate response and your example openocd command!
I'll contact greenwaves support immediately.

closed 2021-01-21 by Yaooooo