GreenWaves-Technologies/gap_sdk #181: NNTOOL import tflite model error

Manucar, 2020-11-25:
Dear Greenwaves Technologies,

I am trying to import my convolutional autoencoder model into NNTOOL. I have generated 2 .tflite model from my .h5 model (not quantized and quantized) with respectively the following pieces of code:

[NOT QUANTIZED]
```python
import tensorflow as tf
from tensorflow import keras

model = keras.models.load_model('cae.h5')

converter = tf.lite.TFLiteConverter.from_keras_model(model)
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS,tf.lite.OpsSet.SELECT_TF_OPS]

tfmodel = converter.convert() 
open ('cae.tflite' , "wb") .write(tfmodel)
```
and when i try to open this model inside NNTOOL i get this error:
```sh
(NNT) open cae.tflite 
open - opening graph file cae.tflite
new_tflite_graph_all - Importing TFLITE model version 3
Traceback (most recent call last):
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 98, in do_open
    self.__open_graph(args.nnfile, args.tensor_file, args.load_quantization, args.load_dequantized)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 71, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/importer.py", line 51, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1756, in create_graph
    self.create_subgraph(0)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1714, in create_subgraph
    self.add_operator(graph, graph_index, op, i)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1673, in add_operator
    in_node, out_node = self.SWITCH_ADD_FUNCTIONS[op_name](
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1438, in add_pack
    in_shape = Dim.unnamed(self.remove_batch_dim(input_tensor.shape), is_ordered=True)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 444, in remove_batch_dim
    if dim[self.batch_dimension] != 1:
IndexError: index 0 is out of bounds for axis 0 with size 0
EXCEPTION of type 'IndexError' occurred with message: 'index 0 is out of bounds for axis 0 with size 0'
```
while for the quantized model:

[QUANTIZED]

```python
import tensorflow as tf
from tensorflow import keras

model = keras.models.load_model('cae.h5')

converter = tf.lite.TFLiteConverter.from_keras_model(model)
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS,tf.lite.OpsSet.SELECT_TF_OPS]
converter.optimizations = [tf.lite.Optimize.DEFAULT]

tfmodel = converter.convert() 
open ('caeq.tflite' , "wb") .write(tfmodel)
```
I get this error instead:
```sh
(NNT) open caeq.tflite -q 
open - opening graph file caeq.tflite
new_tflite_graph_all - Importing TFLITE model version 3
Traceback (most recent call last):
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 98, in do_open
    self.__open_graph(args.nnfile, args.tensor_file, args.load_quantization, args.load_dequantized)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 71, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/importer.py", line 51, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1756, in create_graph
    self.create_subgraph(0)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1714, in create_subgraph
    self.add_operator(graph, graph_index, op, i)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1673, in add_operator
    in_node, out_node = self.SWITCH_ADD_FUNCTIONS[op_name](
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 903, in add_fully_connected
    self.load_filter_parameters(node, input_tensors, output_tensors)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 710, in load_filter_parameters
    raise NoQuantizationError("quantization not present in tflite file")
importer.tflite.new_tflite_graph_all.NoQuantizationError: quantization not present in tflite file
EXCEPTION of type 'NoQuantizationError' occurred with message: 'quantization not present in tflite file'

```
This is my .h5 model [cae.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5598154/cae.zip)
I am on a Ubuntu 20.40 machine with the latest sdk (master).

If you any suggestions on this please let me know.

Best regards,

Manuel

sousoux, 2020-11-25:
Could you put the converted tflite file from the first script into this issue? The second error is normal. Using -q asks nntool to load the quantization from the tflite file and it will not be there.

Manucar, 2020-11-25:
Here it is [cae_tflite.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5598273/cae_tflite.zip).
However i have not fully understand your second observation, could you please elaborate it a bit more?

Best regards,

Manuel

sousoux, 2020-11-25:
-q attempts to load quantization information created during the TOCO conversion. With the way you are using TOCO the graph will not be quantized.

There appears to be a custom operation in your converted graph called FlexConv3D. Not sure what this is but we do not support it currently.

Manucar, 2020-11-25:
I used a standard [Conv3D](https://keras.io/api/layers/convolution_layers/convolution3d/) keras layer in my .h5 model, seems  that during conversion to .tflite model this FlexConv3D pops up instead of the standard Conv3D layer of keras.

I was wondering, do you support 3D convolutions?

Best regards,

Manuel

sousoux, 2020-11-25:
I guess this is using 3D Convolutions and somewhere google has added some hidden (and undocumented) conversion code of a tensorflow Conv3D to a custom function called FlexConv3D perhaps linked to this: https://github.com/deepmind/kinetics-i3d project.. We have no way of really knowing what is in this. We will complain more politely in the next version.

sousoux, 2020-11-25:
I think I found some docs on this https://www.tensorflow.org/lite/guide/ops_select. This needs an appropriate kernel implementation in nntool and in the autotiler generators. We do not plan on supporting this currently but I can understand that it is interesting for gesture recognition / accelerometer type applications. We would need a little more info about your application to bump this up on our priority list.

closed 2020-11-25 by sousoux