GreenWaves-Technologies/gap_sdk #182: NNTOOL import tflite model error

Manucar, 2020-11-27:
Dear GreenWaves Technologies,

I redesigned my network architecture in order to use Conv2D instead of Conv3D as in my previous [issue](https://github.com/GreenWaves-Technologies/gap_sdk/issues/181), however i get the exact same error while importing the .tflite model genereted with this piece of code:

[NOT QUANTIZED]
```python
import tensorflow as tf
from tensorflow import keras

model = keras.models.load_model('cae_2d_model.h5')

converter = tf.lite.TFLiteConverter.from_keras_model(model)
converter.target_spec.supported_ops = [tf.lite.OpsSet.TFLITE_BUILTINS,tf.lite.OpsSet.SELECT_TF_OPS]

tfmodel = converter.convert() 
open ('cae_2d_model.tflite' , "wb") .write(tfmodel)
```
and when i try to open this model inside NNTOOL i get this error (i set debug true):
```sh
(NNT) open cae_2d_model.tflite 
open - opening graph file cae_2d_model.tflite
new_tflite_graph_all - Importing TFLITE model version 3
Traceback (most recent call last):
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 98, in do_open
    self.__open_graph(args.nnfile, args.tensor_file, args.load_quantization, args.load_dequantized)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 71, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/importer.py", line 51, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1756, in create_graph
    self.create_subgraph(0)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1714, in create_subgraph
    self.add_operator(graph, graph_index, op, i)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1673, in add_operator
    in_node, out_node = self.SWITCH_ADD_FUNCTIONS[op_name](
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 1438, in add_pack
    in_shape = Dim.unnamed(self.remove_batch_dim(input_tensor.shape), is_ordered=True)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite/new_tflite_graph_all.py", line 444, in remove_batch_dim
    if dim[self.batch_dimension] != 1:
IndexError: index 0 is out of bounds for axis 0 with size 0
EXCEPTION of type 'IndexError' occurred with message: 'index 0 is out of bounds for axis 0 with size 0'

```

This is my .h5 model [cae_2d_model.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5608152/cae_2d_model.zip) and this my .tflite converted model [cae_2d_model_tflite.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5608156/cae_2d_model_tflite.zip).

I am on a Ubuntu 20.04 machine with the latest sdk (master).

I don't see any unsupported type of operations/activations in my .tflite model, so i'm wondering what could be wrong in my implementation.
If you have any suggestions on this please let me know.

Best regards,

Manuel

sousoux, 2020-11-27:
This was a missing feature for constant folding on our side. It will be supported in the next release. Should go out sometime next week. There are still strange things in your graph:

Here you have a filter that is larger than the input. Perhaps a bug on your side. Definitely not optimal.
[image]

Also are you placing batch normalization correctly. It is getting converted as mul / adds. It's probably in the wrong place in your original graph.
[image]

Manucar, 2020-11-27:
Thanks for the quick response! Glad to see that the missing functionality will be soon available.
I will give you some more details on my implementations, my network is an convolutional autoencoder made of 2 convolutions and dense layer and 2 convolution with upsampling to return to the original size. The inputs are made up of images 16 x 8 and the latent space inside is 16.

The following is the code snippet of the keras model:

```python

def build_model():
        ''' Build CAE model. '''
        # Encoding
        x = Input(shape=(16, 8, 1))
        input = x
        x = Conv2D(filters=16, kernel_size=(4, 2), padding='same', activation="relu")(x)
        x = BatchNormalization()(x)
        x = MaxPooling2D()(x)
        x = Conv2D(filters=32, kernel_size=(2, 4), padding='same', activation="relu")(x)
        x = BatchNormalization()(x)
        x = MaxPooling2D()(x)
        # Fully connected
        shape = x.shape[1:]                    
        x = Flatten()(x)    
        units = x.shape[1]
        x = Dense(16, name="latent")(x)
        x = Dense(units, activation="relu")(x)   
        x = Reshape(target_shape=shape)(x)
        # Decoding
        x = Conv2D(filters=32, kernel_size=(2, 4), padding='same', activation="relu")(x)
        x = BatchNormalization()(x)
        x = UpSampling2D()(x)
        x = Conv2D(filters=16, kernel_size=(4, 2), padding='same', activation="relu")(x)
        x = BatchNormalization()(x)
        x = UpSampling2D()(x)       
        x = Conv2D(filters=1, kernel_size=(4, 2), padding='same', activation='sigmoid')(x)
        output = x
        # Return model
        return Model(input, output, name="CAE")

```

and this is the model summary

```
Model: "CAE"
_________________________________________________________________
Layer (type)                 Output Shape              Param #   
=================================================================
input_1 (InputLayer)         [(None, 16, 8, 1)]        0         
_________________________________________________________________
conv2d (Conv2D)              (None, 16, 8, 16)         144       
_________________________________________________________________
batch_normalization (BatchNo (None, 16, 8, 16)         64        
_________________________________________________________________
max_pooling2d (MaxPooling2D) (None, 8, 4, 16)          0         
_________________________________________________________________
conv2d_1 (Conv2D)            (None, 8, 4, 32)          4128      
_________________________________________________________________
batch_normalization_1 (Batch (None, 8, 4, 32)          128       
_________________________________________________________________
max_pooling2d_1 (MaxPooling2 (None, 4, 2, 32)          0         
_________________________________________________________________
flatten (Flatten)            (None, 256)               0         
_________________________________________________________________
latent (Dense)               (None, 16)                4112      
_________________________________________________________________
dense (Dense)                (None, 256)               4352      
_________________________________________________________________
reshape (Reshape)            (None, 4, 2, 32)          0         
_________________________________________________________________
conv2d_2 (Conv2D)            (None, 4, 2, 32)          8224      
_________________________________________________________________
batch_normalization_2 (Batch (None, 4, 2, 32)          128       
_________________________________________________________________
up_sampling2d (UpSampling2D) (None, 8, 4, 32)          0         
_________________________________________________________________
conv2d_3 (Conv2D)            (None, 8, 4, 16)          4112      
_________________________________________________________________
batch_normalization_3 (Batch (None, 8, 4, 16)          64        
_________________________________________________________________
up_sampling2d_1 (UpSampling2 (None, 16, 8, 16)         0         
_________________________________________________________________
conv2d_4 (Conv2D)            (None, 16, 8, 1)          129       
=================================================================
Total params: 25,585
Trainable params: 25,393
Non-trainable params: 192
```

If you can spot any mistakes on my part wiil be very helpfull to debug this.

Thank you very much for you support

Best regards,

Manuel

sousoux, 2020-11-27:
Filter larger than input ....
```
reshape (Reshape)            (None, **4, 2**, 32)          0        
x = Conv2D(filters=32, kernel_size=**(2, 4)**, padding='same', activation="relu")(x)
```
The batch norm position looks good. Must be a converter problem. Try getting rid of tf.lite.OpsSet.SELECT_TF_OPS in the TOCO options. You should check the resulting TFLITE with something like netron and make sure you get rid of those mul/adds. The mul we might be able to fuse through the relu but the add can't be.

Manucar, 2020-12-01:
Hi @sousoux , 

the last release update fixed my previous error, but now I get this one:

```sh
(NNT) open cae_2d_model.tflite 
open - opening graph file cae_2d_model.tflite
tflite - Importing TFLITE model version 3
strided_slice - reducing STRIDED_SLICE_0_11 to a constant
pack - reducing PACK_0_12 to a constant
strided_slice - reducing STRIDED_SLICE_0_17 to a constant
math_mixin - reducing MUL_0_18 to a constant
Traceback (most recent call last):
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 98, in do_open
    self.__open_graph(args.nnfile, args.tensor_file, args.load_quantization, args.load_dequantized)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 71, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/importer.py", line 53, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite2/tflite.py", line 73, in create_graph
    self._import_tflite_graph(G, TFLiteGraph.from_model(model, 0), opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite2/tflite.py", line 92, in _import_tflite_graph
    self._import_nodes(G, graph, handlers, all_nodes, self._provisional_outputs, opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite2/tflite.py", line 182, in _import_nodes
    raise ValueError("no handler found for %s" % node.op_type)
ValueError: no handler found for LOGISTIC
EXCEPTION of type 'ValueError' occurred with message: 'no handler found for LOGISTIC'
```
I thought sigmoid was a supported activation, am I wrong?
Also I have changed a bit my architecture according to your previous suggestion, it is more suitable now?

.h5 [cae_2d_model.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5621684/cae_2d_model.zip) .tflite [cae_2d_model_tflite.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5621686/cae_2d_model_tflite.zip) (generated with the same script as in my first comment)

Getting rid of the tf.lite.OpsSet.SELECT_TF_OPS in the TOCO options results in the same .tflite model.

Any further suggestions?

Thank you again for your support.

Best regards,

Manuel

Manucar, 2020-12-02:
For the moment i have switched to _tanh_ activation function and it does load into NNTOOL, i have also quantized the .tflite model to fully 8 bit integer (except the input and output) and i get this other error while giving "fusions --scale8" command:

```sh
(NNT) open tan_q32.tflite -q
open - opening graph file tan_q32.tflite
tflite - Importing TFLITE model version 3
strided_slice - reducing STRIDED_SLICE_0_12 to a constant
pack - reducing PACK_0_13 to a constant
strided_slice - reducing STRIDED_SLICE_0_16 to a constant
math_mixin - reducing MUL_0_17 to a constant
nngraph - update graph dimensions
set_aliases - looking for aliased edges
nngraph - calculate liveness
nngraph - update graph dimensions
set_aliases - looking for aliased edges
nngraph - calculate liveness
(NNT tan_q32.tflite 0) adjust
adjust_order - adding transposes to correct tensor order for AT kernels
set_aliases - looking for aliased edges
eliminate_transposes - eliminating unnecessary transposes
eliminate_transposes - search for transposes
eliminate_transposes - ++ Starting up from CONV_2D_0_1[0]
eliminate_transposes - looking up at QUANTIZE_0_0[0] transpose [2, 0, 1]
eliminate_transposes - looking up at input_1[0] transpose [2, 0, 1]
eliminate_transposes - accepted input_1 - input without fixed order - transpose input [2, 0, 1]
eliminate_transposes - ++ Found results for CONV_2D_0_1[0]
eliminate_transposes - ++ Starting down from CONV_2D_0_1[0]
eliminate_transposes - looking down at CONV_2D_0_1_activation[0] transpose [1, 2, 0]
eliminate_transposes - looking down at MUL_0_2[0] transpose [1, 2, 0]
eliminate_transposes - accepted MUL_0_2 - propagated transpose to broadcasted linear op
eliminate_transposes - ++ Found results for CONV_2D_0_1[0]
eliminate_transposes - ++ Starting up from CONV_2D_0_4[0]
eliminate_transposes - looking up at ADD_0_3[0] transpose [2, 0, 1]
eliminate_transposes - rejected ADD_0_3 - transpose out - cannot pass
eliminate_transposes - Rejected up edges from CONV_2D_0_4
eliminate_transposes - ++ Starting down from CONV_2D_0_4[0]
eliminate_transposes - looking down at CONV_2D_0_4_activation[0] transpose [1, 2, 0]
eliminate_transposes - looking down at MUL_0_5[0] transpose [1, 2, 0]
eliminate_transposes - accepted MUL_0_5 - propagated transpose to broadcasted linear op
eliminate_transposes - ++ Found results for CONV_2D_0_4[0]
eliminate_transposes - ++ Starting up from MAX_POOL_2D_0_7[0]
eliminate_transposes - looking up at ADD_0_6[0] transpose [2, 0, 1]
eliminate_transposes - rejected ADD_0_6 - transpose out - cannot pass
eliminate_transposes - Rejected up edges from MAX_POOL_2D_0_7
eliminate_transposes - ++ Starting down from MAX_POOL_2D_0_7[0]
eliminate_transposes - looking down at RESHAPE_0_8[0] transpose [1, 2, 0]
eliminate_transposes - reshape RESHAPE_0_8 trans: old [1, 2, 0] new None shape: old 8x4x32 new 1024
eliminate_transposes - rejected RESHAPE_0_8 - transpose out - does not reverse
eliminate_transposes - Rejected down edges from MAX_POOL_2D_0_7[0]
eliminate_transposes - ++ Starting up from CONV_2D_0_19[0]
eliminate_transposes - looking up at RESIZE_NEAREST_NEIGHBOR_0_18[0] transpose [2, 0, 1]
eliminate_transposes - rejected RESIZE_NEAREST_NEIGHBOR_0_18 - transpose out - cannot pass
eliminate_transposes - Rejected up edges from CONV_2D_0_19
eliminate_transposes - ++ Starting down from CONV_2D_0_19[0]
eliminate_transposes - looking down at CONV_2D_0_19_activation[0] transpose [1, 2, 0]
eliminate_transposes - looking down at MUL_0_20[0] transpose [1, 2, 0]
eliminate_transposes - accepted MUL_0_20 - propagated transpose to broadcasted linear op
eliminate_transposes - ++ Found results for CONV_2D_0_19[0]
eliminate_transposes - ++ Starting up from CONV_2D_0_22[0]
eliminate_transposes - looking up at ADD_0_21[0] transpose [2, 0, 1]
eliminate_transposes - rejected ADD_0_21 - transpose out - cannot pass
eliminate_transposes - Rejected up edges from CONV_2D_0_22
eliminate_transposes - ++ Starting down from CONV_2D_0_22[0]
eliminate_transposes - looking down at CONV_2D_0_22_activation[0] transpose [1, 2, 0]
eliminate_transposes - looking down at MUL_0_23[0] transpose [1, 2, 0]
eliminate_transposes - accepted MUL_0_23 - propagated transpose to broadcasted linear op
eliminate_transposes - ++ Found results for CONV_2D_0_22[0]
eliminate_transposes - ++ Starting up from CONV_2D_0_25[0]
eliminate_transposes - looking up at ADD_0_24[0] transpose [2, 0, 1]
eliminate_transposes - rejected ADD_0_24 - transpose out - cannot pass
[... 387 lines cut ...]
  File "/home/manucar/gap_sdk/tools/nntool/graph/matches/matcher.py", line 128, in match
    has_modified_graph = match_instance.match(G, set_identity=False)
  File "/home/manucar/gap_sdk/tools/nntool/graph/matches/remove_relus.py", line 104, in match
    qrec = G.quantization[NodeId(node)]
  File "/home/manucar/gap_sdk/tools/nntool/quantization/quantization_set.py", line 44, in __getitem__
    item = self.qset[key]
KeyError: <utils.node_id.NodeId object at 0x7f75b1f36550>
EXCEPTION of type 'KeyError' occurred with message: '<utils.node_id.NodeId object at 0x7f75b1f36550>'
(NNT tan_q32.tflite 0) 

```

This is my quantized .tflite model [cae_2d_tanh_quant.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5628305/cae_2d_tanh_quant.zip).

I have also tried generating : 

- a full 8 bit integer quantized model (also input and output layer) [cae_2d_tanh_quant_int8.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5628362/cae_2d_tanh_quant_int8.zip)

- eliminating all the batch normalization layers [nobatch_tanh_quant_int8.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5628365/nobatch_tanh_quant_int8.zip)

but i get the same error as above.

Thank you again,

Manuel

sousoux, 2020-12-03:
Acknowledged. It's a regression introduced by constant folding in the latest release. We will resolve and push a fix.

sousoux, 2020-12-03:
I've fixed that problem and will get it pushed to Github shortly but there is an outstanding issue. We do not have tanh implemented in the fused convolution generator. Tanh is a pretty inefficient activation if you want something very accurate. A min(1, max(-1, v)) approximates it pretty well but obviously not perfectly. We call that a htanh. We could add support for htanh fairly easily or you could experiment with another activation.

Manucar, 2020-12-03:
Tanh was a temporary solution to replace sigmoid because when i tried to import my .tflite model [cae_q.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5635519/cae_q.zip) into NNTOOL, it gives me this error:

```
(NNT) open cae_q.tflite -q
open - opening graph file cae_q.tflite
tflite - Importing TFLITE model version 3
strided_slice - reducing STRIDED_SLICE_0_12 to a constant
pack - reducing PACK_0_13 to a constant
strided_slice - reducing STRIDED_SLICE_0_18 to a constant
math_mixin - reducing MUL_0_19 to a constant
Traceback (most recent call last):
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 98, in do_open
    self.__open_graph(args.nnfile, args.tensor_file, args.load_quantization, args.load_dequantized)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/open.py", line 71, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/importer.py", line 53, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite2/tflite.py", line 73, in create_graph
    self._import_tflite_graph(G, TFLiteGraph.from_model(model, 0), opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite2/tflite.py", line 92, in _import_tflite_graph
    self._import_nodes(G, graph, handlers, all_nodes, self._provisional_outputs, opts)
  File "/home/manucar/gap_sdk/tools/nntool/importer/tflite2/tflite.py", line 182, in _import_nodes
    raise ValueError("no handler found for %s" % node.op_type)
ValueError: no handler found for LOGISTIC
EXCEPTION of type 'ValueError' occurred with message: 'no handler found for LOGISTIC'

```
So I thought sigmoid activation function was not yet supported, am I wrong?

sousoux, 2020-12-03:
Sigmoid is supported. I think you were hitting the same regression that you hit with tanh. That graph converts, adjusts, fuses and generates fine now. The fix will be pushed to github tonight I think. I will check.

Manucar, 2020-12-03:
That's great to hear! I will let you know if the fix solves the issue.

Thank you,

Manuel

Manucar, 2020-12-04:
Hi @sousoux, I downloaded the last release but I get the same error. Do you have any idea why?

sousoux, 2020-12-06:
Try the 3.8 dev branch. I missed the last dot release I'm afraid.

sousoux, 2020-12-06:
Try the 3.8.0 Dev branch

Manucar, 2020-12-06:
Hi @sousoux,

I tried the "3.8.0_dev" branch, but still get the same error as in my prevoius comment:

> Tanh was a temporary solution to replace sigmoid because when I tried to import my .tflite model [cae_q.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5635519/cae_q.zip) into NNTOOL, [...]

sousoux, 2020-12-08:
Hi. We appear to have a very weird git issue between myself and the person looking after Github updates. There is a file missing. We are checking what is going on.

sousoux, 2020-12-08:
OK. It should be fixed in the 3.8.0_dev branch. It turned out to be a too general gitignore rule in the sdk root so all regressions tests pass and the integration tests had no network with a sigmoid in it! Go figure. Sorry for the delay.

Manucar, 2020-12-08:
It loads fine now thank you!

closed 2020-12-08 by Manucar