GreenWaves-Technologies/gap_sdk #185: nntool quantization of custom tensor shapes

gemenerik, 2020-12-02:
Hi all,

Thanks for the ONNX support! It's great to have the option to use other deep learning frameworks, such as PyTorch.
Support for PyTorch to ONNX conversion of quantized models is... limited. So, I would like to quantize in `nntool`. 

However, my inputs are 6 channel tensors, and afaik there is currently no input data type support for such tensors. 
Simply being able to read a custom tensor from `.txt` or similar would be ideal, because it allows any tensor shape without requiring support for loading the data directly into `nntool`.

As workaround I can think of making the network multi-input, does `aquant` support that?
Any other suggestions?

sousoux, 2020-12-02:
Yes. Multi input works. aquant expects a list of files with [num of file] % [num of inputs] == 0.

gemenerik, 2020-12-02:
Brilliant, thanks for the quick response!

closed 2020-12-02 by gemenerik

gemenerik, 2020-12-03:
from `.../gap_sdk/tools/nntool/utils/data_importer.py`, which is used by `nntool`
```python
VALID_IMAGE_EXTENSIONS = ['.pgm', '.png', '.ppm', '.jpg', '.jpeg']
```
`.pgm` supports 6 channels, I concatenated my input images with [Fiji](https://imagej.net/Fiji), an ImageJ distribution.
Import > Image sequence