GreenWaves-Technologies/gap_sdk #186: Graph stacked tensor is also defined as graph local or graph argument

gemenerik, 2020-12-03:
I am working with a 2 input to 1 output `ONNX` model. `nntool` succesfully loads, quantizes.

<details>
  <summary>Model summary from nntool</summary>
  
  ```shell
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
| Step |        Step name        | Operation | Input Dims | Output Dims | Inputs | Active | Params |  Ops   |                    Params                     |        Hints         |
|      |                         |           |  (0x1x2)   |   (0x1x2)   |        |  size  |  size  |        |                                               |                      |
+======+=========================+===========+============+=============+========+========+========+========+===============================================+======================+
|  0   | input_1                 | input     |  3x60x60   |   3x60x60   |        | 10800  |   0    |        | I 3x60x60  FIXED_ORDER=0                      | in: none out: none   |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|  1   | input_2                 | input     |  3x60x60   |   3x60x60   |        | 21600  |   0    |        | I 3x60x60  FIXED_ORDER=0                      | in: none out: none   |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|  2   | Concat_0                | concat    |  3x60x60   |   6x60x60   |   0    | 43200  |   0    |        | A 0                                           | in: none out: none   |
|      |                         |           |  3x60x60   |             |   1    |        |        |        |                                               |                      |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|  3   | Conv_1                  | conv2d    |  6x60x60   |  12x30x30   |   2    | 32400  |  3540  |  3.18M | F 12x6x7x7 S 2x2 D 1x1 G 1 M 1 P 3x3x3x3 zero | in: cxhxw out: cxhxw |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|  4   | Relu_2                  | relu      |  12x30x30  |  12x30x30   |   3    | 21600  |   0    |        | Activation relu                               | in: none out: none   |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|  5   | Conv_3                  | conv2d    |  12x30x30  |  24x15x15   |   4    | 16200  |  7224  |  1.62M | F 24x12x5x5 S 2x2 D 1x1 G 1 M 1 P 2x2x2x2     | in: cxhxw out: cxhxw |
|      |                         |           |            |             |        |        |        |        | zero                                          |                      |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|  6   | Relu_4                  | relu      |  24x15x15  |  24x15x15   |   5    | 10800  |   0    |        | Activation relu                               | in: none out: none   |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|  7   | Conv_5                  | conv2d    |  24x15x15  |   2x15x15   |   6    |  5850  |  434   | 97.20K | F 2x24x3x3 S 1x1 D 1x1 G 1 M 1 P 1x1x1x1 zero | in: cxhxw out: cxhxw |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|  8   | output_1                | output    |  2x15x15   |   2x15x15   |   7    |  450   |   0    |        | O 2x15x15  FIXED_ORDER=0                      | in: none out: none   |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|      | Totals (#)              |           |            |             |        | 43200  | 11198  |  4.89M |                                               |                      |
|      | Max active/Total params |           |            |             |        |        |        |        |                                               |                      |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
|      | Totals (#)              |           |            |             |        |        | 54398  |  4.89M |                                               |                      |
|      | Max mem usage           |           |            |             |        |        |        |        |                                               |                      |
+------+-------------------------+-----------+------------+-------------+--------+--------+--------+--------+-----------------------------------------------+----------------------+
```
  
</details>

In the `hints` column of the model summary in `nntool` it shows `in: none out: none` for several layers. After `fusions` this is only true for the inputs, concatenation and output. Could this cause a problem?

AutoTiler correctly tiles all convolutions.
<details>
  <summary>Successful tiling</summary>
  
  ```shell
echo "GENERATING AUTOTILER MODEL"
GENERATING AUTOTILER MODEL
nntool -g -M BUILD_MODEL_SQ8BIT -m tinyflownetModel.c -T BUILD_MODEL_SQ8BIT/tensors -H tinyflownetInfo.h  BUILD_MODEL_SQ8BIT/tinyflownet.json
echo "COMPILING AUTOTILER MODEL"
COMPILING AUTOTILER MODEL
gcc -g -o BUILD_MODEL_SQ8BIT/GenTile -I. -I/home/rik/gap_sdk/tools/autotiler_v3/Autotiler -I/home/rik/gap_sdk/tools/autotiler_v3/Emulation -I/home/rik/gap_sdk/tools/autotiler_v3/CNN_Generators -I/home/rik/gap_sdk/tools/autotiler_v3/CNN_Generators_SQ8 -I/home/rik/gap_sdk/tools/nntool/autotiler/generators -I/home/rik/gap_sdk/tools/autotiler_v3/CNN_Libraries -I/home/rik/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8 -I/home/rik/gap_sdk/tools/nntool/autotiler/kernels -I/home/rik/gap_sdk/tools/autotiler_v3/Generators/BilinearResizes /home/rik/gap_sdk/tools/autotiler_v3/CNN_Generators/CNN_Generator_Util.c /home/rik/gap_sdk/tools/autotiler_v3/CNN_Generators_SQ8/CNN_Generators_SQ8.c /home/rik/gap_sdk/tools/autotiler_v3/CNN_Generators_SQ8/RNN_Generators_SQ8.c /home/rik/gap_sdk/tools/nntool/autotiler/generators/nntool_extra_generators.c /home/rik/gap_sdk/tools/autotiler_v3/Generators/BilinearResizes/ResizeGenerator.c /home/rik/gap_sdk/tools/autotiler_v3/Autotiler/LibTile.a BUILD_MODEL_SQ8BIT/tinyflownetModel.c /home/rik/gap_sdk/tools/autotiler_v3/Autotiler/LibTile.a -lSDL2 -lSDL2_ttf 
echo "RUNNING AUTOTILER MODEL"
RUNNING AUTOTILER MODEL
BUILD_MODEL_SQ8BIT/GenTile -o BUILD_MODEL_SQ8BIT -c BUILD_MODEL_SQ8BIT --L1 48804 --L2 250000 --L3 8388608
InFeat: 6, OutFeat: 12
Conv => W:  60, Pad:[3,2] PadT:[3,2] => Wc: 30, Filter:[7,7]
     => H:  60, Pad:[3,2] PadT:[3,2] => Hc: 30
Pool => Wc: 30, Pad:[0,0] => Wo: 30, Filter:[1,1]
     => Hc: 30, Pad:[0,0] => Ho: 30
OverlapC: 5
OverlapP: 0
TileCons: 2
UsedIn  : [60 x 60]
UsedC   : [30 x 30]
      SetBiasKerName: KerParSetBiasB32_SQ8
         ConvKerName: KerParConv7x7StrideS_SQ8
  DPReductionKerName: KerParReduct_CC_ReLU_SQ8
Nb Oper : 3186000
[... 49 lines cut ...]
TileCons: 1
UsedIn  : [15 x 15]
UsedC   : [15 x 15]
      SetBiasKerName: KerParSetBiasB32_SQ8
         ConvKerName: KerParConv3x3Stride1_SQ8
  DPReductionKerName: KerParReduct_CC_SQ8
Nb Oper : 97200

==== Process Tiling For User Kernel:                   S5_Conv2d_2x24x3x3 =======================
  S5_Conv2d_2x24x3x3 Partition[0] Size =   4121 (Min:     90, Max:  15265), Fraction:       1.00, Giving:  48804 bytes out of  48804 bytes
Kernel:   S5_Conv2d_2x24x3x3, Total Raw Memory: 8112 fits into L1 memory 48804. Promoting all kernel arguments to initialized buffers.
[... cut ...] Iteration count:   1, Given L1 Memory:  48804, Used L1 Memory:   8112, Reusable Memory: 40692, Used L2 Memory: 0
=================================================================================================
```
  
</details>

But fails at the output layer:

```shell
Error: Graph stacked tensor S2_Output, input tensor Input_1 is also defined as a graph local or graph argument
Execution aborted
```

I am unable to identify what is causing the error, and what the error message tries to tell me.

sousoux, 2020-12-03:
Could you attach the model or the autotiler model created by nntool please? Or send me a PM with it.

gemenerik, 2020-12-03:
Sent both in an e-mail.

sousoux, 2020-12-03:
OK. I've identified the issue. It is to do with the concat of the input at the start. You can work around it by having a single input and concating your data. It looks silly but it is tricky to fix due to the way we do concats which are really just memory location aliases. Can't promise a fix this week.

gemenerik, 2020-12-03:
Thanks for looking into it! I originally had a single input, but split into two so I can use `aquant` in `nntool` #185
I'll look into ONNX quantization support and otherwise wait for a fix.

Thanks again! Really appreciate the effort.

sousoux, 2020-12-03:
You can still use aquant. Just preprocess all of your input files by concating them together.

closed 2020-12-03 by sousoux

gemenerik, 2020-12-03:
You are right, `.pgm` supports 6 channels, and can be imported by `nntool`.

If anyone stumbles upon this, see #185 if you are interested in how I concatenated 2 images and saved as a `.pgm` file.