GreenWaves-Technologies/gap_sdk #188: Autotiler compilation error

Manucar, 2020-12-08:
Hi,

I have generated an ATModel.c [ATModel.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5659823/ATModel.zip) from my custom .tflite model [cae_q.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5659830/cae_q.zip) with the following command:
```sh
open cae_q.tflite -q
adjust
fusions --scale8
gen ATModel.c -t -T tensors
```
and all went fine, except when i try to compile this generated autotiler model:

```python
manucar@45U5:~/Desktop/nntool_test$ gcc ATModel.c -g -o Gentile -I. -I${TILER_INC} -I${TILER_EMU_INC} -I${TILER_CNN_GENERATOR_PATH} -I${TILER_CNN_GENERATOR_PATH_SQ8} -I${NNTOOL_GENERATOR_PATH} -I${TILER_CNN_KERNEL_PATH} -I${TILER_CNN_KERNEL_PATH_SQ8} -I${NNTOOL_KERNELS_PATH} ${TILER_LIB} ${TILER_CNN_GENERATOR_PATH}/CNN_Generator_Util.c ${TILER_CNN_GENERATOR_PATH_SQ8}/CNN_Generators_SQ8.c ${NNTOOL_GENERATOR_PATH}/nntool_extra_generators.c
ATModel.c:6:10: fatal error: ResizeGenerator.h: No such file or directory
    6 | #include "ResizeGenerator.h"
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
```
Those are my enviromental variables:
```python
manucar@45U5:~/Desktop/nntool_test$ env | grep TILER 
TILER_INTEGRAL_GENERATOR_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/Generators/IntegralImage
TILER_CNN_GENERATOR_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/CNN_Generators
TILER_INTEGRAL_KERNEL_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/Generators/IntegralImage
TILER_FFT2D_TWIDDLE_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/Generators/FFT2DModel
TILER_CNN_GENERATOR_PATH_SQ8=/home/manucar/gap_sdk/tools/autotiler_v3/CNN_Generators_SQ8
TILER_BILINEAR_RESIZE_KERNEL_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/Generators/BilinearResizes
TILER_LIB=/home/manucar/gap_sdk/tools/autotiler_v3/Autotiler/LibTile.a
TILER_FFT2D_GENERATOR_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/Generators/FFT2DModel
TILER_CNN_KERNEL_PATH_SQ8=/home/manucar/gap_sdk/tools/autotiler_v3/CNN_Libraries_SQ8
TILER_INC=/home/manucar/gap_sdk/tools/autotiler_v3/Autotiler
TILER_CNN_KERNEL_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/CNN_Libraries
TILER_GENERATOR_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/Generators
TILER_PATH=/home/manucar/gap_sdk/tools/autotiler_v3
TILER_BILINEAR_RESIZE_GENERATOR_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/Generators/BilinearResizes
TILER_FFT2D_KERNEL_PATH=/home/manucar/gap_sdk/tools/autotiler_v3/Generators/FFT2DModel
TILER_EMU_INC=/home/manucar/gap_sdk/tools/autotiler_v3/Emulation
manucar@45U5:~/Desktop/nntool_test$ env | grep NNTOOL
NNTOOL_GENERATOR_PATH=/home/manucar/gap_sdk/tools/nntool/autotiler/generators
NNTOOL_KERNELS_PATH=/home/manucar/gap_sdk/tools/nntool/autotiler/kernels
NNTOOL_PATH=/home/manucar/gap_sdk/tools/nntool
NNTOOL_DIR=/home/manucar/gap_sdk/tools/nntool

```

I have tried the last release 3.8.1 (master ) and also the 3.8.0_dev, but both gave me the same error.  The "_ResizeGenerator.h_" file is present in the environmental variables directories, so I have no clue why it cannot be found.

I am on a Ubuntu 20.04 machine, and i have sourced _GAPUINO_V2_.

If you have any suggestions please let me know.

Best regards,

Manuel

sousoux, 2020-12-08:
Looks like we are missing an -I${TILER_BILINEAR_RESIZE_KERNEL_PATH} in the makefile

Manucar, 2020-12-08:
Adding that argument, fixed the error, but now it gives me:
```python
gcc ATModel.c -g -o Gentile -I. -I${TILER_INC} -I${TILER_EMU_INC} -I${TILER_CNN_GENERATOR_PATH} -I${TILER_CNN_GENERATOR_PATH_SQ8} -I${NNTOOL_GENERATOR_PATH} -I${TILER_CNN_KERNEL_PATH} -I${TILER_CNN_KERNEL_PATH_SQ8} -I${NNTOOL_KERNELS_PATH} -I${TILER_BILINEAR_RESIZE_KERNEL_PATH} ${TILER_LIB} ${TILER_CNN_GENERATOR_PATH}/CNN_Generator_Util.c ${TILER_CNN_GENERATOR_PATH_SQ8}/CNN_Generators_SQ8.c ${NNTOOL_GENERATOR_PATH}/nntool_extra_generators.c
/usr/bin/ld: /tmp/ccEuwKZv.o: in function `cae_qModel':
/home/manucar/Desktop/nntool_test/ATModel.c:33: undefined reference to `Load_RNN_SQ8_Library'
/usr/bin/ld: /home/manucar/Desktop/nntool_test/ATModel.c:34: undefined reference to `LoadResizeLibrary'
/usr/bin/ld: /home/manucar/Desktop/nntool_test/ATModel.c:77: undefined reference to `GenerateResizeNew'
collect2: error: ld returned 1 exit status

```

sousoux, 2020-12-08:
You need to use a custom template. Have a look at one of the examples that is using that. The template will include Load_RNN_SQ8_Library but it is unnecessary since you are not using RNN kernels.

closed 2020-12-08 by sousoux

Manucar, 2020-12-08:
Yes _'Load_RNN_SQ8_Library'_ was not necessary, thank you.
But the other two functions _'LoadResizeLibrary'_ and '_GenerateResizeNew_' are declared in the '_ResizeGenerator_.h' file, which I though was included with your previous suggestion, but it doesn't seem so, because it gives me those undefined reference errors.

Any idea on that?

Thank you again @sousoux