GreenWaves-Technologies/gap_sdk #189: Error running Gentile executable

Manucar, 2020-12-08:
Hi,

in my previous #188 issue i finally managed to compile the ATModel.c, but when I try to execute the generated Gentile executable, i get this error:

```
manucar@45U5:~/Desktop/nntool_test$ ./Gentile -o . -c . --L1 50000 --L2 300000 --L3 8000000
...
...
...
------------------------------------------------------------------------------------------------------------------------------------------------
Graph stacked tensors
------------------------------------------------------------------------------------------------------------------------------------------------

Generating Code For User Kernel:             S1_Conv2d_32x1x4x2_Relu
Generating Code For User Kernel:                   S3_MatMul_32x16x8
Generating Code For User Kernel:                   S5_MatAdd_32x16x8
Generating Code For User Kernel:            S6_Conv2d_16x32x2x4_Relu
Generating Code For User Kernel:                   S8_MatMul_16x16x8
Generating Code For User Kernel:                  S10_MatAdd_16x16x8
Generating Code For User Kernel:                     S11_MaxPool_2x2
Generating Code For User Kernel:       S12_Op_MAX_POOL_2D_0_7_TOUT_0
Generating Code For User Kernel:                   S14_Linear_16x512
Generating Code For User Kernel:  S15_Op_FULLY_CONNECTED_0_10_fusion
Generating Code For User Kernel:                   S18_MatMul_8x4x16
Generating Code For User Kernel:                   S20_MatAdd_8x4x16
Generating Code For User Kernel: S21_Op_RESIZE_NEAREST_NEIGHBOR_0_20
Generating Code For User Kernel:           S22_Op_CONV_2D_0_21_TIN_0
Generating Code For User Kernel:           S23_Conv2d_16x16x2x4_Relu
Generating Code For User Kernel:                  S25_MatMul_16x16x8
Generating Code For User Kernel:                  S27_MatAdd_16x16x8
Generating Code For User Kernel:           S28_Conv2d_32x16x4x2_Relu
Generating Code For User Kernel:        S29_Conv2d_1x32x4x2_Hsigmoid
Loading coefficient file ./tensors/S1_Infos.tensor: 9 Byte items
Loading coefficient file ./tensors/S1_Weights.tensor: 256 Byte items
Loading coefficient file ./tensors/S1_Biases.tensor: 128 Byte items
Loading coefficient file ./tensors/S1_Mul_scale.tensor: 32 Byte items
Loading coefficient file ./tensors/S1_Mul_shift.tensor: 32 Byte items
Loading coefficient file ./tensors/Cae_batch_normalization_fusedbatchnormv31_0f68.tensor: 32 Byte items
Loading coefficient file ./tensors/S3_Infos.tensor: 9 Byte items
Loading coefficient file ./tensors/Cae_batch_normalization_fusedbatchnormv3_be95.tensor: 4096 Byte items
Error: For coefficient file ./tensors/Cae_batch_normalization_fusedbatchnormv3_be95.tensor expecting 4096 Byte items, got 32
Execution aborted
```
it seems a shaping error in the very last last sigmoid layer, but can't really tell why.

Thank you for your support.

Manuel

sousoux, 2020-12-08:
Sorry for all the problems. The ONNX support is rather new so there are not enough regression tests on this. I will check this tomorrow and get back to you.

Manucar, 2020-12-08:
Thank you @sousoux for the continuous support, really appreciate it.
However i am not using ONNX but tflite instead, or ONNX is implicitly used also for tflite?

sousoux, 2020-12-08:
Hmm. The fused batch norm names was confusing me then. With ONNX we fuse batch norm at import. This must be coming from TFLITE. Have you already given me this graph? If yes could you point me at it? I'm surprised that tensor dimensions would be off with TFLITE.

Manucar, 2020-12-08:
Here it is the .tflite model [cae_q.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5662246/cae_q.zip)

sousoux, 2020-12-09:
OK. The problem is linked to a missing matrix x vector add kernel. NNTOOL is incorrectly selecting a kernel that operates on the entire tensor so the generator expects a larger input. I still thing you should be able to fuse the batch normalization into the convolutional layers above as part of the conversion which would make this work. I'm interested in what size you expect to come out of this network. Is 1 x 16 x 8 correct? The parameters to the resize cause this. Netron actually displays the wrong size since it doesn't do the shape calculation properly. It shows a 1 x 1 x 1 output. Is that what you are expecting?

Manucar, 2020-12-09:
Yes the correct output size is 1 x 16 x 8, however how can achieve the fusion of the batch normalization layer with the convolution one that you are mentioning?

sousoux, 2020-12-09:
Read the tensorflow toco documentation. I've added a better error message for this problem. We will be adding better support for broadcasted piecewise operations in a future version. For now you need to figure out how to get the batch norm to fuse properly.

sousoux, 2020-12-09:
Which version of TF are you using? It looks like TF 1.X did not support batch norm folding in quantized models at one point. I would suggest you use the latest TF 2.X release.

closed 2020-12-09 by sousoux

Manucar, 2020-12-09:
I am using TF 2.3.0, on google colab service, I will investigate further on this.

Thank you as always.