GreenWaves-Technologies/gap_sdk #190: AIdeck simple SPI transmission

Manucar, 2020-12-10:
Hi,

I am trying to implement a simple SPI transmission on the AIdeck board from the GAP8 to the NINA module. I just need to send simple 1D signal, not needs of all the JPEG stuff.
I am following [this example](https://github.com/bitcraze/AIdeck_examples/blob/master/GAP8/test_functionalities/wifi_jpeg_streamer/test.c) for the GAP8 side and [this example](https://github.com/bitcraze/AIdeck_examples/blob/master/NINA/firmware/main/ai-deck-jpeg-streamer-demo.c) for the NINA side.

However on the GAP8 side, i need to use FreeRTOS (which is not used by the example) for my specifications, and if try to initialize the nina struct:

```C
#include "bsp/transport/nina_w10.h"

struct pi_nina_w10_conf nina_conf;
pi_nina_w10_conf_init(&nina_conf);
``` 
it gives me a compilation error 'undefined reference to `pi_nina_w10_conf_init''.

I found out that using FreeRTOS as OS, it inlcudes the 'freertos_bsp_rules.mk' in which there is a line of code:

```
EXCLUDE_FROM_SRCS= transport/transport.c transport/nina_w10/nina_w10.c

``` 
and the second one is the file i think i need for using the NINA api, but if I comment this line, I get these errors:
```sh
/home/manucar/gap_sdk/rtos/pmsis/pmsis_bsp/transport/nina_w10/nina_w10.c:196:31: error: 'pi_task_t {aka struct pi_task}' has no member named 'implem'
     nina->pending_first = task->implem.next;
                               ^~
/home/manucar/gap_sdk/rtos/pmsis/pmsis_bsp/transport/nina_w10/nina_w10.c:197:49: error: 'pi_task_t {aka struct pi_task}' has no member named 'implem'
     __nina_w10_send_packet(nina, (uint8_t *)task->implem.data[0], task->implem.data[1], task);
                                                 ^~
/home/manucar/gap_sdk/rtos/pmsis/pmsis_bsp/transport/nina_w10/nina_w10.c:197:71: error: 'pi_task_t {aka struct pi_task}' has no member named 'implem'
     __nina_w10_send_packet(nina, (uint8_t *)task->implem.data[0], task->implem.data[1], task);
                                                                       ^~
/home/manucar/gap_sdk/rtos/pmsis/pmsis_bsp/transport/nina_w10/nina_w10.c: In function '__nina_w10_send_async':
/home/manucar/gap_sdk/rtos/pmsis/pmsis_bsp/transport/nina_w10/nina_w10.c:332:25: error: 'pi_task_t {aka struct pi_task}' has no member named 'implem'
       nina->pending_last->implem.next = task;
                         ^~
/home/manucar/gap_sdk/rtos/pmsis/pmsis_bsp/transport/nina_w10/nina_w10.c:337:9: error: 'pi_task_t {aka struct pi_task}' has no member named implem'
     task->implem.next = NULL;
         ^~
/home/manucar/gap_sdk/rtos/pmsis/pmsis_bsp/transport/nina_w10/nina_w10.c:339:9: error: 'pi_task_t {aka struct pi_task}' has no member named implem'
     task->implem.data[0] = (int)buffer;
         ^~
/home/manucar/gap_sdk/rtos/pmsis/pmsis_bsp/transport/nina_w10/nina_w10.c:340:9: error: 'pi_task_t {aka struct pi_task}' has no member named implem'
     task->implem.data[1] = size;
```
Do you have any suggestions on this?

I am on Ubuntu 20.04 machine with the last sdk release.

Thank you for your support.

Best regards,

Manuel

P.S: remember to include [this fix](https://github.com/GreenWaves-Technologies/gap_sdk/issues/177#issuecomment-728025153) in the master branch, because at the moment is not present, even if it was in a previous version.

Manucar, 2020-12-10:
Ok I found this piece of code helpful:

```
#if defined(PMSIS_DRIVERS) || defined(__PULPOS2__)
    hyperflash->waiting_first = task->next;
#else
    hyperflash->waiting_first = task->implem.next;
#endif  /* PMSIS_DRIVERS */
```
Getting rid of the 'implem' field like the else case works now.

sivasivarajah, 2020-12-11:
Hello @Manucar ,

Ai deck is not yet supported by FreeRTOS, that is the reason why some BSP drivers are not included in sources.
Do you have any special requirements to not use PulpOs? PulpOs fully supports AI deck and all of BSP drivers, NINA wifi module included.

FreeRTOS support will be done soon. Will try to update and test on board as soon as possible.
Your fix should avoid compilation errors, warnings, though it has not been tested with boards. With PulpOs-2, there are less fragmentation due to implementation.

Best.

Manucar, 2020-12-11:
Thank you for your clarification, please let me know when the support will be ready.

Best regards,

Manuel

closed 2021-03-24 by sousoux