GreenWaves-Technologies/gap_sdk #191: ONNX bilinear upsampling support

gemenerik, 2020-12-10:
In ONNX opset version 9 and lower ONNX converts bilinear upsampling layers in PyTorch to `Upsample` layers with a 'mode' attribute.
```diff
-EXCEPTION of type 'ValueError' occurred with message: 'no handler found for Upsample'
```

In ONNX opset version 10 and up ONNX converts these to a `Resize` layer, again with a 'mode' attribute
```diff
-EXCEPTION of type 'ValueError' occurred with message: 'no handler found for Resize'
```

Neither is recognized by `nntool` as a bilinear upsampling layer.

sousoux, 2020-12-10:
Upsample is deprecated since v10. I didn't bother looking at it. I will.

sousoux, 2020-12-10:
Two interpolation modes: nearest (default), and linear (including bilinear, trilinear, etc)

I love the etc. Implement this packet header witth 1234, etc.

sousoux, 2020-12-10:
Resize we need to do. We have kernels that support many of its modes. We already handle them from TF.

sousoux, 2020-12-11:
Could you let me have your graph with 'resize' in it please? I'm particularly interested in something that has the scale tensor rather than the size tensor.

sousoux, 2020-12-12:
I found something to test against. The importer for resize is implemented. It does have limitations which you can check in nntool with 'help onnx'. It will go to the dev_3.8.0 branch shortly.

closed 2020-12-12 by sousoux