GreenWaves-Technologies/gap_sdk #192: Wrong output from custom CNN network

Manucar, 2020-12-19:
Hi,

I have designed a simple 2D convolutional auto-encoder that recreates in output a sine wave for testing the functionalities of the GAP8 before I bring to it my real input that handles EEG signals. 

The input of the network is a 2D matrix 16x8 where 16 are the samples from the sine wave and 8 is the number of signals (for simplicity they are 8 equal sine waves, this is in order to simulate my real input of 8 channels EEG).
The sine wave is bounded from 0 to 1, and the samples are stored in float64, so I converted my .h5 model into .tflite without quantizing, and let nntool do the full integer 8 bit quantization. 

So in the 'aquant' command I provided a slice of the sine wave from the numpy array:
```python
np.save('signals.npy', sine_wave[0:100]) 
```
and I have also tried to quantized directly in python, but the final results is the same.
 
All the GAPFlow steps work fine (I used a makefile as suggested for automatizing the process), but when I set some data in the input of the network, the output buffer gives me always wrong results, e.g:
```C
Input buffer:  (16 hard-coded sample of a sine wave equal for each of the 8 channels)
0 0 0 0 0 0 0 0 
6 6 6 6 6 6 6 6 
12 12 12 12 12 12 12 12 
19 19 19 19 19 19 19 19 
25 25 25 25 25 25 25 25 
31 31 31 31 31 31 31 31 
37 37 37 37 37 37 37 37 
43 43 43 43 43 43 43 43 
49 49 49 49 49 49 49 49 
55 55 55 55 55 55 55 55 
61 61 61 61 61 61 61 61 
66 66 66 66 66 66 66 66 
72 72 72 72 72 72 72 72 
77 77 77 77 77 77 77 77 
82 82 82 82 82 82 82 82 
86 86 86 86 86 86 86 86 
Entering main controller
Constructor
Call cluster
Running on cluster
Runner completed

result buffer:  (this should be equal to the input buffer)
1 3 19 21 15 21 21 21 
4 12 21 21 21 21 21 21 
10 21 21 21 21 21 21 21 
21 21 21 21 21 21 21 21 
21 21 21 21 21 21 21 21 
21 21 21 21 21 21 21 21 
21 21 21 21 21 21 21 21 
21 21 21 21 21 21 21 21 
21 21 21 21 21 21 21 21 
14 15 12 21 21 21 21 21 
16 19 14 21 21 21 21 21 
13 16 14 21 21 21 21 21 
14 13 9 21 21 21 21 21 
15 8 2 21 14 21 21 21 
18 14 6 21 16 21 21 21 
20 19 13 21 20 21 21 21 
```
Even if I change the input buffer with other values, it gives me more or less this same wrong output.

My model in python infers correctly every sample of the given sine waves, but honestly I don't know why here is so wrong.

I upload here my project directory with all the files I use [sine_wave.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5720105/sine_wave.zip)

I am on a Ubuntu 20.04 machine with _GAPUINO_V2_, running the 3.8.0_dev branch (because here it's fixed a previous [issue](https://github.com/GreenWaves-Technologies/gap_sdk/issues/182#issuecomment-740507261) I had importing the model) 

Please let me know if you can spot any issue.

Thank you very much for your time.

Best regards,

Manuel

marco-fariselli, 2020-12-22:
Hi Manuel !! Sorry for the late reply! I looked at your code and indeed there were some problems related to this network which in part I solved and will be released soon. These were the main issues:

- Our resizers could not handle multi channel inputs and signed char input/output types (now it's done and will be released). From my side the execution of the network in nntool quantized has the same results as the run on chip;
- The network has at the end a sigmoid which is by default converted by the GapFlow to a HARDSigmoid when you quantize the network. It seems to degradate a lot the accuracy. Up to the layer before the sigmoid the quantization seems to not affect much numerically.

Manucar, 2020-12-22:
Hi @marco-fariselli! Thank you for your reply.

Sorry if I mislead you on my input, it's not multichannel, it is only a 2D image 16x8.

However regarding the input type, I can leave it float32 and let the model quantize, or I must quantize every time my input from float32 to int8? Because I tested the quantized model in python with only input and output NOT quantized it gives me good results, but with also input and output quantized to int8 :
```
converter.inference_input_type = tf.int8
converter.inference_output_type = tf.int8 
```
lead to extra loss of accuracy. So I am wondering, with the aquant command nntool performs also input and ouput int8 quantization? Because would be handy if I can leave my input as it is in float32 and let the model itself taking care of the quantization of the input. If it is not the case I will think to switch to a 16bit quantization scheme.

In your opinion there is something that can be done in order to fix this HARDSigmoid conversion loss of accuracy?

Thank you for your support.

marco-fariselli, 2020-12-22:
Hi @Manucar !! When you give nntool any input it should be in its real value. For instance if you trained a network with [-1:1] input images you should take the original images (typically [0:255] values) and apply a normalization function to convert them to the real value, i.e. "x:x/128-1". In your case you just need to fed the network with your npy files. Beware here because I see you have a single file with 100 samples in it, it won't work, you should convert it into 100 files. However when you eventually deploy your network then the inputs should be in the quantized values, the network will expect `signed char` or `signed short` as inputs. 

For the Sigmoid issue you could do several things:

- remove it from the graph and compute that calculation after the network finished in your main.c. 
- retrain the network with a HARDSigmoid instead of a sigmoid, shouldn't effect much the overall accuracy
- wait for support of a Sigmoid in GapFlow

Manucar, 2020-12-23:
Hi @marco-fariselli! 
As you suggested I switched to hard_sigmoid activation function in my last layer, but when I try to quantized it with 'aquant' in nntool (I have also generated 100 files for correctness ), it gives me this error now:

```python
(NNT q8.tflite 0) aquant data/* --nptype float64 --scheme SQ8
aquant - input file ['data/sine43.npy']
graph_executer - execute uncached: quantization mode none
Traceback (most recent call last):                                    
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/aquant.py", line 75, in do_aquant
    stats_collector.collect_stats(self.G, data)
  File "/home/manucar/gap_sdk/tools/nntool/stats/stats_collector.py", line 96, in collect_stats
    stat = self._collect(G, input_tensors, step_idx)
  File "/home/manucar/gap_sdk/tools/nntool/stats/activation_stats_collector.py", line 50, in _collect
    for _, node, fnode, output_tensors, details in\
  File "/home/manucar/gap_sdk/tools/nntool/execution/graph_executer.py", line 232, in execute_iterator
    output_tensors = switch.execute(node, output_tensors, qrec, details)
  File "/home/manucar/gap_sdk/tools/nntool/quantization/kernels/kernel_executer.py", line 60, in execute
    raise ValueError("no handler found for %s quantization %s" %
ValueError: no handler found for BinaryOpParameters quantization float32
EXCEPTION of type 'ValueError' occurred with message: 'no handler found for BinaryOpParameters quantization float32'

```
I think it could be something related to the converted .tflite graph, because as the last layer now it has not a '_Logistic_' as my previous model and so nntool it's not able to convert it in the _hard_sigmoid_.

Any thoughts on that?

P.S.: Do you approximately know when the sigmoid  support for gap flow will be planned? Because this is part of my master thesis and I have to manage the time scheduling, and in case redesign my topology correctly.

Thank you for your support, really appreciate it.

Best regards,

Manuel

Manuel

marco-fariselli, 2020-12-28:
Hi Manuel! Sorry for the late reply. I took a look at your code (first attached files, sine_wave.zip) and indeed there was a bug in our flow regarding non RELU activations. The HSigmoid does not show too much drop in accuracy now wrt to the standard Sigmoid. Let me know if it's fine for you. I pushed the fixes in a development branch (https://github.com/GreenWaves-Technologies/gap_sdk/tree/3.8.0_dev). The official release will come soon. 

The problem with your latest files is that there is no support now for the Minimum operator. It will be released soon. However it seems that your conversion process ended with a strange minimum + Relu instead of an HSigmoid layer. Can you share the keras code which produces the .h5 graph?

Best regards, Marco

Manucar, 2020-12-30:
Hi @marco-fariselli, thank you for your reply!

The code I used for the conversion is simply this:

```python
from google.colab import drive
drive.mount("/content/drive")

import tensorflow as tf
from tensorflow import keras

model = keras.models.load_model('drive/My Drive/models/sine.h5')

converter = tf.lite.TFLiteConverter.from_keras_model(model)

tfmodel = converter.convert() 
open ('drive/My Drive/models/sine.tflite' , "wb") .write(tfmodel)
```

Regarding the HSigmoid, I will do more testing to see if it's suitable enough. 
Do you know if it is planned to implement standard sigmoid activation in the next 2 months?

Thank you very much.

Best regards,

Manuel

Manucar, 2021-02-10:
Hi @marco-fariselli! Hope everything is good!

I am back to work on the Gapuino, and i was wondering if Sigmoid activation function has now been implemented, and if not, approximately when will it be?

Thanks.

Best regards,

Manuel

marco-fariselli, 2021-02-10:
Hi Manuel !! 
The sigmoid support is at the moment on hold due to other, more prioritized, tasks on going. I suggest however to use the latest 3.8.3_dev branch of the gap_sdk where we fixed some bugs regarding HARD sigmoid support. It should work much better now. To use it you just need to provide a tflite graph with a standard sigmoid, during quantization the HARD version will replace it automatically.

Cheers, Marco

Manucar, 2021-02-10:
@marco-fariselli Thanks for the quick reply!

So I don't need to train my original network with an HARD Sigmoid, am i right?

Best regards,

Manuel

marco-fariselli, 2021-02-10:
It shouldn't be necessary

Manucar, 2021-02-11:
Hi @marco-fariselli!

I tried again my autoencoder with a sine wave, as in my previous posts, and indeed now the output seems more realistic then before, so thank you for the fix!

However, I still have two doubts:

- the output on some samples has a lot less accuracy respect to other, especially when the value range is wider. For example:

  ```
  Input buffer:
  53 53 53 53 53 53 53 53 
  59 59 59 59 59 59 59 59 
  64 64 64 64 64 64 64 64 
  70 70 70 70 70 70 70 70 
  75 75 75 75 75 75 75 75 
  80 80 80 80 80 80 80 80 
  85 85 85 85 85 85 85 85 
  89 89 89 89 89 89 89 89 
  94 94 94 94 94 94 94 94 
  98 98 98 98 98 98 98 98 
  102 102 102 102 102 102 102 102 
  106 106 106 106 106 106 106 106 
  109 109 109 109 109 109 109 109 
  112 112 112 112 112 112 112 112 
  115 115 115 115 115 115 115 115 
  117 117 117 117 117 117 117 117 
  
  result buffer:
  55 55 56 56 56 57 57 57 
  58 58 58 58 58 59 59 59 
  60 60 60 59 60 61 61 61 
  62 62 62 61 62 62 63 63 
  64 64 64 63 63 63 64 64 
  66 66 65 64 64 64 65 65 
  69 68 67 66 66 66 67 67 
  71 70 69 68 68 68 69 69 
  74 72 71 71 70 71 72 71 
  76 75 74 74 73 73 74 73 
  77 77 76 76 75 75 76 76 
  80 79 79 78 78 78 79 79 
  82 82 82 81 81 81 82 81 
  86 86 85 84 85 85 85 84 
  90 89 88 88 88 88 88 87 
  93 92 90 89 89 90 90 90 
  MSE = 411
  ```
  while

  ```
  Input buffer:
  120 120 120 120 120 120 120 120 
  122 122 122 122 122 122 122 122 
  123 123 123 123 123 123 123 123 
  124 124 124 124 124 124 124 124 
  125 125 125 125 125 125 125 125 
  126 126 126 126 126 126 126 126 
  126 126 126 126 126 126 126 126 
  126 126 126 126 126 126 126 126 
  126 126 126 126 126 126 126 126 
  126 126 126 126 126 126 126 126 
  125 125 125 125 125 125 125 125 
  123 123 123 123 123 123 123 123 
  122 122 122 122 122 122 122 122 
  120 120 120 120 120 120 120 120 
  118 118 118 118 118 118 118 118 
  115 115 115 115 115 115 115 115 
  
  result buffer:
  118 118 117 116 116 116 116 116 
  124 123 122 120 120 121 120 121 
  127 127 127 125 126 126 125 125 
  127 127 127 127 127 127 127 127 
  127 127 127 127 127 127 127 127 
  127 127 127 127 127 127 127 127 
  127 127 127 127 127 127 127 127 
  127 127 127 127 127 127 127 127 
  127 127 127 127 127 127 127 127 
  127 127 127 127 127 127 127 127 
  127 127 127 127 127 127 127 127 
  125 126 125 124 123 123 124 124 
  119 119 118 118 117 117 119 119 
  113 113 113 113 113 113 114 114 
  109 109 109 108 109 109 110 110 
  106 105 105 104 105 106 107 107 
  MSE = 17
  
  ```
  seems more feasible (even though I would expect less errors, because this is a simple sine wave, and next I am going to implement this on EEG signals, so better accuracy is needed, may think to try 16bit quantization later).

- when I insert samples with negative values, the output is always zero for them

  ```
  -46 -46 -46 -46 -46 -46 -46 -46 
  -40 -40 -40 -40 -40 -40 -40 -40 
  -34 -34 -34 -34 -34 -34 -34 -34 
  -27 -27 -27 -27 -27 -27 -27 -27 
  -21 -21 -21 -21 -21 -21 -21 -21 
  -15 -15 -15 -15 -15 -15 -15 -15 
  -8 -8 -8 -8 -8 -8 -8 -8 
  -2 -2 -2 -2 -2 -2 -2 -2 
  3 3 3 3 3 3 3 3 
  10 10 10 10 10 10 10 10 
  16 16 16 16 16 16 16 16 
  23 23 23 23 23 23 23 23 
  29 29 29 29 29 29 29 29 
  35 35 35 35 35 35 35 35 
  41 41 41 41 41 41 41 41 
  47 47 47 47 47 47 47 47 
  
  result buffer:
  0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 
  0 0 0 0 0 0 0 0 
  5 4 4 3 2 1 0 0 
  15 13 11 11 12 11 7 6 
  22 19 19 20 22 19 13 12 
  28 24 23 24 27 25 21 18 
  32 29 26 28 32 30 26 23 
  41 38 36 39 41 39 34 30 
  51 48 46 47 47 48 43 39 
  MSE = 412
  
  ```
  and really don't know why, because I though the output should be signed char as the input.

I am on the 3.8.0_dev branch, and here is my project folder with all the files [sine_wave.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/5967379/sine_wave.zip).

Please let me know if you can spot any issue in my implementation.

Thank you very much as always.

Best regards,

Manuel

marco-fariselli, 2021-02-17:
Ciao Manuel! 

You should always check the nntool execution on your input samples to see the result. Try with dump command with float and quant (-q -d flags) execution on your input sample. Remember that in nntool the inputs must always be in their floating point representation. 

- I tried your negative values example and also the floating point execution is returning me 0s. Maybe it is a training issue? Try the same input sample in keras/tflite and see if it differs. 
- On the different range I suggest you to always calibrate the quantization with samples that exploit the whole range of values. If not the network will clamp some of your layers to wrong min/max values resulting in large errors.

Best regards, Marco

Manucar, 2021-02-17:
Hi @marco-fariselli !

I tried the dump and tensors commands with 100 numpy samples (float64) of the sine wave, and the resulting QSNR is the following:

```python
(NNT q8.tflite 0) dump data/* -S float
(NNT q8.tflite 0) dump data/* -S quant -q -d
(NNT q8.tflite 0) tensors -t float quant -Q

000 009:  44, 23, 25, 25, 25, 30, 29, 29, 29, 28
010 019:  28, 25, 38, 38, 

```
increasing the number of the samples to 1000, it gives worse values
```python
(NNT q8.tflite 0) dump data/* -S float
(NNT q8.tflite 0) dump data/* -S quant -q -d
(NNT q8.tflite 0) tensors -t float quant -Q

000 009:  22, 21, 24, 24, 24, 25, 22, 22, 22, 24
010 019:  24, 25, 5, 5, 

```
and also the predictions are worse than before:

```
result buffer:
55 55 55 56 56 56 56 56 
57 57 57 57 57 58 57 57 
58 59 58 58 59 59 59 59 
60 60 60 60 60 60 60 60 
62 62 62 61 62 61 62 62 
63 63 63 62 63 63 64 64 
64 64 64 63 63 64 65 65 
65 65 65 65 65 65 66 66 
67 66 66 66 66 67 68 68 
69 68 68 67 68 69 70 70 
70 70 70 70 70 71 72 72 
73 73 72 73 73 74 74 75 
75 76 75 76 76 76 77 77 
78 78 78 78 79 79 79 79 
80 81 81 81 81 81 81 81 
83 83 83 83 83 84 84 84 
MSE = 610
```

100 samples are the minimum number to capture at least a whole dynamic range of the sine wave. Seems increasing the number of samples in input at the aqaunt command in NNTOOL worsen the QSNR. 

The negative inputs work fine in the quantized int8 tflite model in keras

```python
INPUT BUFFER:
array([[-46, -46, -46, -46, -46, -46, -46, -46],
       [-40, -40, -40, -40, -40, -40, -40, -40],
       [-34, -34, -34, -34, -34, -34, -34, -34],
       [-27, -27, -27, -27, -27, -27, -27, -27],
       [-21, -21, -21, -21, -21, -21, -21, -21],
       [-15, -15, -15, -15, -15, -15, -15, -15],
       [ -8,  -8,  -8,  -8,  -8,  -8,  -8,  -8],
       [ -2,  -2,  -2,  -2,  -2,  -2,  -2,  -2],
       [  3,   3,   3,   3,   3,   3,   3,   3],
       [ 10,  10,  10,  10,  10,  10,  10,  10],
       [ 16,  16,  16,  16,  16,  16,  16,  16],
       [ 23,  23,  23,  23,  23,  23,  23,  23],
       [ 29,  29,  29,  29,  29,  29,  29,  29],
       [ 35,  35,  35,  35,  35,  35,  35,  35],
       [ 41,  41,  41,  41,  41,  41,  41,  41],
       [ 47,  47,  47,  47,  47,  47,  47,  47]], dtype=int8)

OUTPUT BUFFER:
array([[[-42, -42, -42, -42, -42, -42, -42, -46],
        [-39, -36, -39, -39, -39, -36, -36, -39],
        [-32, -32, -32, -32, -32, -32, -32, -32],
        [-25, -25, -25, -25, -25, -25, -25, -25],
        [-22, -22, -22, -22, -22, -18, -18, -18],
        [-15, -15, -15, -15, -15, -15, -15, -15],
        [ -7,  -7,  -7,  -7,  -7,  -7,  -7,  -7],
        [  0,   0,   0,   0,   0,   0,   0,   0],
        [  4,   4,   7,   4,   4,   7,   4,   7],
        [ 11,  11,  11,  11,  11,  11,  11,  11],
        [ 18,  18,  18,  18,  18,  18,  18,  18],
        [ 25,  25,  25,  25,  25,  25,  25,  25],
        [ 32,  29,  29,  29,  29,  29,  32,  29],
        [ 36,  36,  36,  36,  36,  36,  36,  36],
        [ 42,  42,  42,  42,  42,  42,  42,  42],
        [ 49,  49,  49,  49,  49,  49,  49,  49]]], dtype=int8)
```

So to me doesn't seem a training error, but maybe I am wrong. What do you think ?
If you need some others information or files feel free to ask!

Thank you for your time.

Best regards,

Manuel

marco-fariselli, 2021-02-17:
Dump and comparison should be used on single inputs. To test what is going on. I see float inference going to 0. Are you sure the quantization is done properly? If your network is already quantized you can try to export quantization in tflite and open it in nntool already quantized with:
open model_quantized.tflite -q

Manucar, 2021-02-17:
Ok thanks for the clarification of the dump command. My network is not already quantized, I let NNTOOL quantize with the following commands:

```
adjust
fusions --scale8
aquant model/data/* --nptype float64 --scheme SQ8   (100 samples of the sine wave, circa all range of one period)
```
and everything seems to work fine. What is a feasible number of samples to fed into NNTOOL for the quantization?

However I also tried to quantized it directly in python to full int8 and import into NNTOOL with the -q flag, and the results compared to the model quatized by NNTOOL are the following:

```python
Input buffer:
-46 -46 -46 -46 -46 -46 -46 -46 
-40 -40 -40 -40 -40 -40 -40 -40 
-34 -34 -34 -34 -34 -34 -34 -34 
-27 -27 -27 -27 -27 -27 -27 -27 
-21 -21 -21 -21 -21 -21 -21 -21 
-15 -15 -15 -15 -15 -15 -15 -15 
-8 -8 -8 -8 -8 -8 -8 -8 
-2 -2 -2 -2 -2 -2 -2 -2 
3 3 3 3 3 3 3 3 
10 10 10 10 10 10 10 10 
16 16 16 16 16 16 16 16 
23 23 23 23 23 23 23 23 
29 29 29 29 29 29 29 29 
35 35 35 35 35 35 35 35 
41 41 41 41 41 41 41 41 
47 47 47 47 47 47 47 47 

result buffer:
0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 
0 0 0 13 17 8 2 10 
29 7 20 41 49 39 32 45 
67 53 62 80 85 75 67 78 
95 96 107 118 121 110 99 107 
119 127 127 127 127 127 122 124 
127 127 127 127 127 127 127 127 
MSE = 1710
```
Worst result for the first sample.
```python
Input buffer:
53 53 53 53 53 53 53 53 
59 59 59 59 59 59 59 59 
64 64 64 64 64 64 64 64 
70 70 70 70 70 70 70 70 
75 75 75 75 75 75 75 75 
80 80 80 80 80 80 80 80 
85 85 85 85 85 85 85 85 
89 89 89 89 89 89 89 89 
94 94 94 94 94 94 94 94 
98 98 98 98 98 98 98 98 
102 102 102 102 102 102 102 102 
106 106 106 106 106 106 106 106 
109 109 109 109 109 109 109 109 
112 112 112 112 112 112 112 112 
115 115 115 115 115 115 115 115 
117 117 117 117 117 117 117 117 

result buffer:
54 54 54 55 56 56 56 55 
57 57 57 59 60 61 60 59 
61 61 62 62 63 64 63 63 
65 65 65 67 67 68 67 67 
69 69 69 69 70 71 70 70 
71 71 71 71 71 72 74 72 
74 74 74 74 74 75 76 76 
76 76 76 75 76 77 78 78 
78 78 78 77 78 79 82 82 
82 82 82 80 82 84 84 84 
85 85 85 85 86 87 87 87 
90 88 90 90 91 92 91 91 
94 93 94 94 95 95 94 94 
98 98 99 99 100 99 98 98 
101 101 102 103 103 102 101 102 
106 106 107 108 108 107 106 106 
MSE = 127
```
Better result for the second sample, where the range dynamic is wider.

```python
Input buffer:
120 120 120 120 120 120 120 120 
122 122 122 122 122 122 122 122 
123 123 123 123 123 123 123 123 
124 124 124 124 124 124 124 124 
125 125 125 125 125 125 125 125 
126 126 126 126 126 126 126 126 
126 126 126 126 126 126 126 126 
126 126 126 126 126 126 126 126 
126 126 126 126 126 126 126 126 
126 126 126 126 126 126 126 126 
125 125 125 125 125 125 125 125 
123 123 123 123 123 123 123 123 
122 122 122 122 122 122 122 122 
120 120 120 120 120 120 120 120 
118 118 118 118 118 118 118 118 
115 115 115 115 115 115 115 115 

result buffer:
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
127 127 127 127 127 127 127 127 
125 125 125 124 124 124 124 124 
MSE = 23

```
slightly worse result.

So overall the quantization in NNTOOL seems better than the one quantized directly in python, but still I can't get where this drop in accuracy could be introduced, maybe in the adjust and fusions commands? or something related to the tiling operations?
Thanks.

Best regards,

Manuel

gemenerik, 2021-02-24:
> ... For instance if you trained a network with [-1:1] input images you should take the original images (typically [0:255] values) and apply a normalization function to convert them to the real [...]

Does this imply that (for deployment) we should train networks on integer input values, @marco-fariselli ?

marco-fariselli, 2021-02-25:
> Does this imply that (for deployment) we should train networks on integer input values, @marco-fariselli ?

No the network can be trained with float inputs as in normal training process. When you quantize the network within nntool you should provide the same inputs as in the training. The nntool will figure out the activation ranges by running the NN float and will quantize the tensors within that range. When you run inference in nntool you should always provide the inputs in their float representation. If the quantized inference is used nntool will take care of quantize the inputs in the discrete domain.

> 
> EDIT: Also, how does one change whether the network expects `signed char` or `signed short` as input? My generated kernels specify `signed char` as input to the first convolution.
> 
> ```c
> void S1_Conv2d_6x1x3x3(
> 		signed char * __restrict__ In,
>                 ...) {
> ...
> }
> ```

When you generate the code and run it on the platform, on the other hand, you should provide the quantized values, i.e. you need to take care of discretize the inputs. If you run qshow in nntool you will see, for each layer, the quantization strategy. In this case you will need the information about the first layer.

`+======+==============+======+==============+==============+==============+==============+=============+=======+=======+
|  0   | input_1      | SQ8  |              | -0.85<(i8-0. |              |              |             |       |       |
|      |              |      |              | 00)*0.006632 |              |              |             |       |       |
|      |              |      |              | 08<0.84      |              |              |             |       |       |
+------+--------------+------+--------------+--------------+--------------+--------------+-------------+-------+-------+
`

In this example for instance the quantized deployed network expects the inputs in int8 representation and the [-127:128] range will be interpreted as [-0.85:0.84]. Hence -127=-0.84, ....

Manucar, 2021-02-25:
> > 
> EDIT: Also, how does one change whether the network expects `signed char` or `signed short` as input? My generated kernels specify `signed char` as input to the first convolution.
> 
> ```c
> void S1_Conv2d_6x1x3x3(
> 		signed char * __restrict__ In,
>                 ...) {
> ...
> }
> ```
I think he was referring to the quantization scheme, if you quantize with 8bit the generated code will have _signed char_, as in your case, instead if you quantized with 16bit, the generated code will have _signed short_ as input.

Best,

Manuel

closed 2021-03-19 by Manucar