GreenWaves-Technologies/gap_sdk #193: Issue with viewing PULP-OS manual documentation

alialkhatatbih, 2021-01-21:
Hello,

I'm having issues when trying to view the PULP-OS manual reference. I get a blank/empty reference back. For example, when I click on Reference->Kernal or Reference->Drivers or Reference->Cluster. Here is a link to one of the examples: https://greenwaves-technologies.com/manuals/BUILD/PULP-OS/html/group__groupKernel.html

Your help is much appreciated.

Thanks.

Yaooooo, 2021-01-21:
Hi,

I see the problem, sorry for that. We will fix it soon.
BTW, since the doc is also integrated in gap_sdk/doc, you can run command: `make all` to compile it. 
Then you can open the file `gap_doc.html` with your browser.

alialkhatatbih, 2021-01-21:
Thank you for the fast response! I'll try that in the meantime to get going. I hope to see it fixed soon.

Thanks.