GreenWaves-Technologies/gap_sdk #194: nntool - resize .onnx opset 11 - 'scales' is an array

gemenerik, 2021-01-26:
Running the latest `3.8.0_dev`
Using an `onnx` opset version 11 model with a bilinear resize layer
```
Traceback (most recent call last):
  File "/home/rik/miniconda3/envs/gap_sdk/lib/python3.6/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/rik/miniconda3/envs/gap_sdk/lib/python3.6/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/rik/miniconda3/envs/gap_sdk/lib/python3.6/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/rik/gap_sdk/tools/nntool/interpreter/commands/open.py", line 98, in do_open
    self.__open_graph(args.nnfile, args.tensor_file, args.load_quantization, args.load_dequantized)
  File "/home/rik/gap_sdk/tools/nntool/interpreter/commands/open.py", line 71, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/rik/gap_sdk/tools/nntool/importer/importer.py", line 53, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 51, in create_graph
    return self._import_onnx_model(G, model.graph, opset_import, opts)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 62, in _import_onnx_model
    self._import_nodes(G, graph, handlers, all_nodes, outputs, opts)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 153, in _import_nodes
    used_tensors=used_tensors)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/handlers/handler.py", line 61, in handle
    return ver_handle(node, **kwargs)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/handlers/backend/resize.py", line 107, in version_11
    return cls.common_11_13(node, **kwargs)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/handlers/backend/resize.py", line 96, in common_11_13
    return cls._common(node, scales, None, nearest_mode=nearest_mode, **kwargs)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/handlers/backend/resize.py", line 43, in _common
    if scales:
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()
EXCEPTION of type 'ValueError' occurred with message: 'The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()'
```

In `resize.py`:
```python
if scales:
    sizes = np.array(x_shape) * np.array(scales)

...

# common_11_13
scales_inp = inputs[2]
scales_shape = scales_inp[2].shape if scales_inp else None
if scales_inp and len(scales_shape) > 0 and scales_shape[0] == len(x_shape):
    scales = cls.get_constant(inputs[2])
    return cls._common(node, scales, None, nearest_mode=nearest_mode, **kwargs)
else:
    sizes_inp = inputs[2]
    sizes_shape = sizes_inp[2].shape if sizes_inp else None
    if not sizes_inp or len(sizes_shape) < 1 or sizes_shape[0] == len(x_shape):
        raise ValueError('neither sizes nor scales input is valid for reshape')
    sizes = cls.get_constant(inputs[3])
    return cls._common(node, None, sizes, nearest_mode=nearest_mode, **kwargs)
```

It seems like it's getting different inputs than expected. I can share the `onnx` model.

Cheers

sousoux, 2021-01-26:
Try changing that line (43) to `if scales is not None:`

We'll check it out as well

gemenerik, 2021-01-26:
That makes sense. Now it fails as x_shape is NoneType.

```
Traceback (most recent call last):
  File "/home/rik/miniconda3/envs/gap_sdk/lib/python3.6/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/rik/miniconda3/envs/gap_sdk/lib/python3.6/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/rik/miniconda3/envs/gap_sdk/lib/python3.6/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/rik/gap_sdk/tools/nntool/interpreter/commands/open.py", line 98, in do_open
    self.__open_graph(args.nnfile, args.tensor_file, args.load_quantization, args.load_dequantized)
  File "/home/rik/gap_sdk/tools/nntool/interpreter/commands/open.py", line 71, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/rik/gap_sdk/tools/nntool/importer/importer.py", line 53, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 51, in create_graph
    return self._import_onnx_model(G, model.graph, opset_import, opts)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 62, in _import_onnx_model
    self._import_nodes(G, graph, handlers, all_nodes, outputs, opts)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 153, in _import_nodes
    used_tensors=used_tensors)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/handlers/handler.py", line 61, in handle
    return ver_handle(node, **kwargs)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/handlers/backend/resize.py", line 107, in version_11
    return cls.common_11_13(node, **kwargs)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/handlers/backend/resize.py", line 96, in common_11_13
    return cls._common(node, scales, None, nearest_mode=nearest_mode, **kwargs)
  File "/home/rik/gap_sdk/tools/nntool/importer/onnx/handlers/backend/resize.py", line 44, in _common
    sizes = np.array(x_shape) * np.array(scales)
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'
EXCEPTION of type 'TypeError' occurred with message: 'unsupported operand type(s) for *: 'NoneType' and 'float''
```

gemenerik, 2021-02-08:
Any update? I can share the `onnx` model.

sousoux, 2021-03-24:
Is this still a problem? If it is please send us the graph. I will close for now.

closed 2021-03-24 by sousoux

gemenerik, 2021-03-30:
I do not get this error anymore.

The only way nntool accepts the layer for me right now is if I specify the exact size to which it should be resized. A scale factor does not work, which is a big shame because that means I cannot work with arbitrarily sized inputs anymore. I can send some example models later this week.

```python
torch.nn.Upsample(size=(16, 21), mode='bilinear', align_corners=True)  # works
torch.nn.Upsample(scale=2, mode='bilinear', align_corners=True)  # does not work
```