GreenWaves-Technologies/gap_sdk #196: CNN construction fails with looping application code

gemenerik, 2021-02-18:
`__PREFIX(CNN_Construct)()` fails when `__PREFIX(CNN)(Input_1, Output_1)` is called anywhere in the application code. By commenting out the call to the CNN, the code runs successfully. 

Below are several snippets of my application code. The code is designed to wait for an image to be loaded into a buffer, do some manipulations on it, then input it into the CNN for inference, and wait for that to finish to start the loop from the top with an image capture. The CNN construction also fails when the CNN is called right after construction, before the loop initiates. The CNN construction and execution is (obviously) successful when the image capture/manipulation loops are never started. I somewhat suspect there may be some conflicts in the way I handle having multiple tasks?

I managed to get the same CNN running in different application code that had a single image capture followed by a single CNN inference.

Any obvious errors in my approach? Otherwise; any recommendations on how to debug this?

```c
/* Define includes... */
#include <stdio.h>
...

/* Define constants */
#define CAM_WIDTH 324
...

/* Define global variables and pointers */
static unsigned char * imgBuff0;
...

void run_inference(pi_task_t *return_task)
{
    /* Demosaick images */
    ...

    /* Run inference */
    pi_cluster_send_task_to_cl(&cluster_dev, task);  // Send RunNetwork to cluster

    /* Write images to host */
    ...

    /* Return to cam_handler */
    pi_task_push(return_task);
}

static void cam_handler(void *arg)
{
    pi_camera_control(&camera, PI_CAMERA_CMD_STOP, 0);
    inference_done = 0;
    run_inference(pi_task_callback(&task_camera, network_handler, (void *)&inference_done));
    return;
}

static void network_handler(void *arg)
{   
    *(int *)arg = 1;
    if (inference_done)
    {
        pi_camera_capture_async(&camera, imgBuff0, CAM_WIDTH*CAM_HEIGHT, pi_task_callback(&task_camera, cam_handler, NULL));
        pi_camera_control(&camera, PI_CAMERA_CMD_START, 0);
        printf("Capture image\n");
    }
}

static int open_camera(struct pi_device *device)
{
    struct pi_himax_conf cam_conf;
    pi_himax_conf_init(&cam_conf);
    cam_conf.format = PI_CAMERA_QVGA;
    pi_open_from_conf(device, &cam_conf);
    
    if (pi_camera_open(device))
        return -1;

    pi_camera_control(device, PI_CAMERA_CMD_AEG_INIT, 0);

    return 0;
}

static void RunNetwork()
{
    printf("Running network\n");
    __PREFIX(CNN)(Input_1, Output_1);  // The line that if inserted, breaks the CNN construction.
}

void neural_network(void)
{
    /* Assign memory */
    imgBuff0 = (unsigned char *)pmsis_l2_malloc((CAM_WIDTH*CAM_HEIGHT)*sizeof(unsigned char));
    if (imgBuff0 == NULL) 
    {
        printf("Failed to allocate memory for camera buffer\n");
        return 1;
    }
    ...

    /* Configure CNN task */
    pi_cluster_conf_init(&conf);
    pi_open_from_conf(&cluster_dev, (void *)&conf);
    pi_cluster_open(&cluster_dev);
    pi_freq_set(PI_FREQ_DOMAIN_CL,175000000);
    pi_freq_set(PI_FREQ_DOMAIN_FC,250000000);

    task = pmsis_l2_malloc(sizeof(struct pi_cluster_task));
    if (!task) {
        printf("failed to allocate memory for task\n");
    }
    memset(task, 0, sizeof(struct pi_cluster_task));
    task->entry = &RunNetwork;
    task->stack_size = STACK_SIZE;
    task->slave_stack_size = SLAVE_STACK_SIZE;
    task->arg = NULL;

    /* Construct CNN */
    if (__PREFIX(CNN_Construct)())
    {
        printf("Failed to construct CNN\n");
        pmsis_exit(-5);
    }

    // pi_cluster_send_task_to_cl(&cluster_dev, task);
    // Construction also fails when RunNetwork is called here, directly after the construction and outside of any of the loops

    /* Open camera */
    if (open_camera(&camera))
    {
        printf("Failed to open camera\n");
        return -1;
    }
    printf("Opened camera\n");

    /* Start camera */
    pi_camera_control(&camera, PI_CAMERA_CMD_STOP, 0);
    pi_camera_capture_async(&camera, imgBuff0, CAM_WIDTH*CAM_HEIGHT, pi_task_callback(&task_camera, cam_handler, NULL));
    pi_camera_control(&camera, PI_CAMERA_CMD_START, 0);

    /* Start main loop */
    while(1)
    {   
        pi_yield();
    }
    pmsis_exit(0);
}

int main(void)
{
    return pmsis_kickoff((void *) neural_network);
}
```

Yaooooo, 2021-02-18:
Hi Rik,

Seems your idea and implementations are good. However, I doubt you have
enough memory on the chip to have multi tasks here running asynchronously...
Be aware that we have only 512KB of L2 embedded, please check how much
memory (L2 and L1) is needed by the NN, plus the image buffer size,
demosaicking
image size...
I am not sure you can handle multi tasks like this.
There's one way you can try, is to create a pipeline, like:

   1. Minimize the buffer for the camera, and stream the image to L3 (ext
   memory).
   2. Use one or several small buffers to bring the slice of images back to
   L2 and do the demosaicking/resizing asynchronously.
   3. Change the memory budget for autotiler to use smaller memory pool for
   NN.

In this way, you may have a chance to handle multi tasks in a pipeline way,
but it still needs to check the smallest memory size you need.

Hope this helps

gemenerik, 2021-02-18:
Thanks for the quick reply Yao!

I already ran into memory constraints before, so it seems likely that that is the issue here, indeed. I'll investigate if that is indeed the case and report here.

gemenerik, 2021-02-18:
It was indeed a lack of memory. Slightly reducing the L2 memory budget for the CNN was sufficient to make it work.

Thanks for the help!

closed 2021-02-18 by gemenerik