GreenWaves-Technologies/gap_sdk #197: Probleme HWCE WaterMeter

Julie33380, 2021-02-22:
Hi, we are students working on the gap8 board for some project. We are currently looking for some informations about how to use the HWCE core, but when we want to launch the WaterMeter example which is using the HWCE (in the MakeFile, USE_HARDWARE_CE = -DRT_HAS_HWCE=1) we have an issue “Error: home/yao/Projects/SDK/autotiler-top/AutoTilerGen/ModelCreate.c:499. Can't find Kernel Lib KerPoolNxNStrideS_fp
Execution aborted”.

Thank you for your help !

Pafrak, 2021-02-22:
Hello,
thanks for reporting this issue. 

We are looking into it. In the meanwhile you can take a look at this branch that is were we are fixing the problem. 

https://github.com/GreenWaves-Technologies/gap_sdk/tree/water_meter_hwce

Pafrak, 2021-02-24:
Should be solved now in branch mentioned above.

closed 2021-02-24 by Pafrak

Julie33380, 2021-02-24:
Hello,

Thank you for your help. It's now working