GreenWaves-Technologies/gap_sdk #199: 16bit quantization Gentile execution error

Manucar, 2021-02-24:
Hi,

from my previous issue #192 I am trying to generate a 16bit quantization model instead of my previous 8bit in order to test the accuracy drop, but when my script launches the Gentile executable, it gives me this error:

```
==== Process Tiling For User Kernel:         S12_Conv2d_1x32x7x5_Hsigmoid =======================
S12_Conv2d_1x32x7x5_Hsigmoid Partition[0] Size =   7876 (Min:    224, Max:  23796), Fraction:       1.00, Giving:  48804 bytes out of  48804 bytes
Kernel: S12_Conv2d_1x32x7x5_Hsigmoid, Total Raw Memory: 11204 fits into L1 memory 48804. Promoting all kernel arguments to initialized buffers.
S12_Conv2d_1x32x7x5_Hsigmoid For Iter Space: 0 Iteration count:   1, Given L1 Memory:  48804, Used L1 Memory:  11204, Reusable Memory: 37600, Used L2 Memory: 10692
=================================================================================================

Error: ProcessGraphEdges, Graph local argument declared, written but not read S6_Output
Execution aborted
```
I honestly don't know what it could be linked to, because initially I thought it was due to the last layer which has the Sigmoid activation, but I also tried to change the activation to Linear, but it gives me the same error:
```
==== Process Tiling For User Kernel:                  S12_Conv2d_1x32x7x5 =======================
 S12_Conv2d_1x32x7x5 Partition[0] Size =   7876 (Min:    224, Max:  23796), Fraction:       1.00, Giving:  48804 bytes out of  48804 bytes
Kernel:  S12_Conv2d_1x32x7x5, Total Raw Memory: 11204 fits into L1 memory 48804. Promoting all kernel arguments to initialized buffers.
 S12_Conv2d_1x32x7x5 For Iter Space: 0 Iteration count:   1, Given L1 Memory:  48804, Used L1 Memory:  11204, Reusable Memory: 37600, Used L2 Memory: 10692
=================================================================================================

Error: ProcessGraphEdges, Graph local argument declared, written but not read S6_Output
Execution aborted
```
 
I am on a Ubuntu 20.04 machine with GAPUINO_V2, running the 3.8.0_dev branch.
Here is my project directory with all the files [sine_16.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/6035459/sine_16.zip)

If you can spot something wrong in my implementation please let me know!
Thank you for your support.

Best regards,

Manuel

marco-fariselli, 2021-02-25:
Hello Manuel !! Thank you for the report, this is indeed a bug in our tool. It will be fixed and integrated in the incoming sdk release.

Manucar, 2021-03-11:
Hi @marco-fariselli! Do you know approximately when will be released the next sdk?

Pafrak, 2021-03-16:
New SDK has been released yesterday!

closed 2021-03-16 by Pafrak