GreenWaves-Technologies/gap_sdk #201: UART with a RN2483 LoRa module.

Orel138, 2021-03-09:
Hello, 

I am working on a GAPPoC-A and I am trying to use the GAP8 with a LoRa RN2483 module.
My module works correctly and I can send in LoRa using the RN2483 with an FTDI and CuteCom or CoolTerm.

I have tried the examples 'uart_helloworld' (modified to send a 'sys get vdd' command followed by Carriage Return and Flow Control) 
and 'uart_input' in 'gap_sdk/examples/pmsis/periph/uart' and I get no response from my LoRa module. (in case of an error he should send me "invalid_param").

I also tried to modify the 'ble_read_test' examples to convert the AT commands which communicate with the BLE/WIFI nina module into commands understandable by the RN2483.
However, I manage to retrieve on the Tx signal of the GAP8 with an FTDI the messages I ask it to send and I noticed that a <break> corresponding to "0x00" always appears at the initialization of the UART. 
(cf images)
[image]
[image]

I have the feeling that the module doesn't understand the commands sent because if I connect the Tx signal from the GAP8 to the Rx of the RN2483 and the Tx of the RN2483 to an FTDI, I receive nothing on my FTDI.
I saw on a GAP8 documentation (part 10. "UART FLOW CONTROL") that it could be necessary to add a "simulated" hardware flow control with CTS, RTS signals and an input GPIO pin (I can't find which one?).

Can the GAP8 send a command such as "sys get ver" with <CR+LF> in UART at a speed of 57600 bauds 8N1 without flow control?

(I am using UART at a speedrate of 57600 bps, 8 bits, no parity, 1 Stop bit, no flow control.)

Thank you very much,
Best regards,

Aurélien

Yaooooo, 2021-03-09:
Hi @Orel138 

This <break> is sent due to the system init. 
If you boot it from JTAG, you may see it twice. 
You can try to send the command several times, and you will see there's no more <break> added.
For example:
[image]

Therefore, I think there should be a way to send some AT Command to abord previous communication (to let lora module ignore the previous <br>), and start it again. 

In such a speed, 57600 is super slow, you don't really need flow control, it should works fine.

Best,
Yao

Orel138, 2021-03-11:
Hello,

Thank you very much for your reply.
I have managed to get my LoRa RN2483 module to work.

The module interprets the "break" as a request to change the baudrate and waits for a response. 
To continue, it is necessary to send 0x55 (the character "U" in ASCII) and then, it is possible to send the other commands.

Thank you very much,
Best regards,

Aurélien

closed 2021-03-11 by Orel138

Yaooooo, 2021-03-11:
Great! Thank you for telling us this good news :)
Best,
Yao