GreenWaves-Technologies/gap_sdk #204: SDK 3.9.1: Flashing hangs, if image contains files for hyperflash

asmorkalov, 2021-03-16:
System: Ubuntu 18.04
Hardware: Gapuino v2 (hyperflash, hyperram)
gap8-openocd: gap8 (b84d97ec4d2e601e704b54351e954b1c58d41683)

`make all` for Face REID and othe samples hangs is the same place:
```
gap8-openocd -c "gdb_port disabled; telnet_port disabled; tcl_port disabled" -c "script interface/ftdi/gapuino_ftdi.cfg; script target/gap8revb.tcl; script tcl/flash_image.tcl; script tcl/jtag_boot.tcl; gap_flash_raw_hyper /home/alexander/Projects/Xperience/GAP8/licence_plate_recognition/BUILD/GAP8_V2/GCC_RISCV/target.board.devices.flash.img 5139860 /mnt/projects/Projects/Xperience/GAP8/gap_sdk/tools/gap8-openocd-tools; exit;"
Open On-Chip Debugger 0.10.0+dev-gb84d97ec4 (2021-03-16-09:45)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
--------------------------
begining flash session (hyperflash)
--------------------------
load flasher to L2 memory
Loading binary through JTAG
Warn : Burst read timed out
[Flahser]: Hyperflash flasher is ready
Instruct flasher to begin flash per se
device struct address is 470047304
Warn : Burst read timed out
going to wait on addr GAP_RDY
wait on gap_rdy done witg buff ptr 0 469784956
loading image with addr 469784956 addr_min 469784956 and size 262144
load image done
^CError: libusb_handle_events() failed with LIBUSB_ERROR_INTERRUPTED
Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/mnt/projects/Projects/Xperience/GAP8/gap_sdk/tools/gapy/gapy", line 162, in <module>
    main()
  File "/mnt/projects/Projects/Xperience/GAP8/gap_sdk/tools/gapy/gapy", line 152, in main
    operationFunc(args, config)
  File "/mnt/projects/Projects/Xperience/GAP8/gap_sdk/tools/gapy/run.py", line 146, in operationFunc
    raise RuntimeError('Runner has failed with value: %d' % status)
RuntimeError: Runner has failed with value: -1
/mnt/projects/Projects/Xperience/GAP8/gap_sdk/tools/rules/pulp_rules.mk:232: recipe for target 'flash_fs' failed
make: *** [flash_fs] Error 1
```

Yaooooo, 2021-03-16:
The same on the issue #203

Yaooooo, 2021-03-17:
Same as #203

closed 2021-03-17 by Yaooooo

asmorkalov, 2021-03-17:
Yes, the solution works for me. Checked with branch 3.9_dev