GreenWaves-Technologies/gap_sdk #208: Improve CNN performance

Manucar, 2021-03-24:
Hi!

I implemented my custom CNN and deployed it into the GAP8 thanks to the _GAPflow_. This model has actually 100k parameters, file size ~400Kb and the the inference time ~10ms, but this delay does not fulfill my real-time constraint with the rest of the application (I have to reduce this time by 2-3ms).

Reading through the previous issues I understand that the HWCE of the GAP8 is not used during the standard _GAPflow_ execution, and so I am wondering if I can exploit this to reduce the inference time of my network? I also read that deploying the network on the HWCE will not use the others 8 cores for the computation, so it's a choice between the 8 cores or the single HWCE?

Thank you for your time!

Best regards,

Manuel

sousoux, 2021-03-24:
How have you quantized your network? Are you using the SQ8 quantization scheme? What are the performance stats for cycles and cycles/op by layer?

Manucar, 2021-03-24:
The network is 16bit quantized by NNTool and the performance counter with -DPERF are the following:

```
                      S4_Conv2d_16x1x3x3_Relu: Cycles:      27227, Operations:      81920, Operations/Cycle: 3.008778
                           S5_AveragePool_2x1: Cycles:      19332, Operations:       8192, Operations/Cycle: 0.423753
                     S8_Conv2d_32x16x3x3_Relu: Cycles:     315541, Operations:    1187840, Operations/Cycle: 3.764455
                           S9_AveragePool_2x1: Cycles:      19543, Operations:       8192, Operations/Cycle: 0.419178
                    S12_Conv2d_64x32x3x3_Relu: Cycles:     582272, Operations:    2367488, Operations/Cycle: 4.065948
                          S13_AveragePool_2x1: Cycles:      20125, Operations:       8192, Operations/Cycle: 0.407056
                   S16_Conv2d_128x64x3x3_Relu: Cycles:    1400328, Operations:    4726784, Operations/Cycle: 3.375484
                          S17_AveragePool_2x1: Cycles:      21808, Operations:       8192, Operations/Cycle: 0.375642
                  S20_Conv2d_256x128x3x3_Relu: Cycles:    3792061, Operations:    9445376, Operations/Cycle: 2.490829
                          S21_AveragePool_2x2: Cycles:      10427, Operations:       8192, Operations/Cycle: 0.785653
            S22_Op_AVERAGE_POOL_2D_0_9_TOUT_0: Cycles:       5398, Operations:       2048, Operations/Cycle: 0.379400
            S22_Op_AVERAGE_POOL_2D_0_9_TOUT_0: Cycles:       4937, Operations:       2048, Operations/Cycle: 0.414827
                            S26_Linear_8x2048: Cycles:      42467, Operations:      16384, Operations/Cycle: 0.385805

                                        Total: Cycles:    6261466, Operations:   17870848, Operations/Cycle: 2.854100
```

Manucar, 2021-03-26:
And why _autotiler_v2_ seems to implement functions for the HWCE, while _autotiler_v3_ does not? It is no longer supported the HWCE in the new version?

Why one should use HWCE instead of the cluster of 8 cores? Or am I missing something?

sousoux, 2021-04-03:
Why are you using 16 bit quantization? Is it absolutely necessary?

Manucar, 2021-04-03:
@sousoux Yes at the moment we are dealing with EEG signals, so we need to keep as much as we can of the signal precision.

So it's feasible to use the HWCE or at the moment it's not supported at all?

Best,

Manuel

Gualor, 2021-04-13:
Hi @sousoux , I am interested in using the HWCE too, but the lack of updated examples using the new `autotiler_v3` makes it really hard for me in doing so. 

> Hi!
>
> I implemented my custom CNN and deployed it into the GAP8 thanks to the _GAPflow_. This model has actually 100k parameters, file size ~400Kb and the the inference time ~10ms, but this delay [...]

It would be really helpful if the question in this issue is finally addressed. Thanks for your time.

BR,

Lorenzo

sousoux, 2021-04-15:
I'm afraid we have no plans to add HWCE support in those kernels currently.

closed 2021-04-15 by sousoux