GreenWaves-Technologies/gap_sdk #209: Debug: Cannot run to main function

MiguelCosta94, 2021-03-24:
I'm facing some issues while debugging the pmsis/helloworld example. After building the program using the makefile you provide, I am not able to configure the debug to run to a breakpoint set at the main function. The behavior of gdb suggests that the "c" command is working as "si", as only one instruction is stepped when I enter the command. Below is the list of commands I have run to debug:

1. Start OPENOCD: /usr/bin/gap8-openocd -f <gap8_openocd install path>/tcl/interface/ftdi/gapuino_ftdi.cfg -f <gap_sdk install path>/tools/gap8-openocd/tcl/target/gap8.tcl

2. Start GDB: /lib/gap_riscv_toolchain/bin/riscv32-unknown-elf-gdb

3. Configure GDB:
3.1: set arch riscv:rv32
3.2: target extended-remote :3333
3.3: file <gap_sdk install path>/examples/pmsis/helloworld/BUILD/GAP8/GCC_RISCV/test
3.4 load
3.5 hb main
3.6 c

Yaooooo, 2021-03-24:
Hi @MiguelCosta94 

We are sorry that gap8 doesn't support gdb debug due to some hw limitation.
For debugging, you could try:
- Simple debug with printf, gpio. etc.
- Using our virtual platform which can produce trace log, which can help for debugging:
https://gvsoc.readthedocs.io/en/latest/
https://greenwaves-technologies.com/gvsoc-the-full-system-simulator-for-profiling-gap-applications/

Hope this can help you. 
If you have further questions about how to debug on GAP8, please let us know. 

Sorry again for the inconvenience.

MiguelCosta94, 2021-03-25:
Hi @Yaooooo 

I am working with Gapuino v1. Do you know if Gapuino v2 supports debug via GDB?

Yaooooo, 2021-03-25:
Hi @MiguelCosta94 

No, this is not related to the board, but the chip. Therefore, all the boards based on GAP8 cannot support GDB for now.

MiguelCosta94, 2021-03-25:
Hi @Yaooooo 

Sorry, I thought it was supported. The documentation of GAP-SDK encourages the use of GDB and OPENOCD to debug applications in GAP8 chips. Nevertheless, I was able to debug applications using the PlatformIO plug-in for VS Code.

closed 2021-04-29 by Yaooooo