GreenWaves-Technologies/gap_sdk #210: 16bit NNTool quantization: 'number cannot be represented with this many bits'

Manucar, 2021-04-02:
Hi!

I am trying to implement my new CNN and quantize it (16bit) with NNTOOL, but when I use the _'aquant'_ command it gives me this error:

```python
Traceback (most recent call last):                                    
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/manucar/.local/lib/python3.8/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/manucar/gap_sdk/tools/nntool/interpreter/commands/aquant.py", line 88, in do_aquant
    qrecs = quantizer.quantize(self.G)
  File "/home/manucar/gap_sdk/tools/nntool/quantization/unified_quantizer.py", line 413, in quantize
    self.quantize_forwards(G, state, params, **quant_kwargs)
  File "/home/manucar/gap_sdk/tools/nntool/quantization/unified_quantizer.py", line 211, in quantize_forwards
    cur_qrec = self.handle(handler,
  File "/home/manucar/gap_sdk/tools/nntool/quantization/unified_quantizer.py", line 168, in handle
    return handler.quantize(params, in_qs, stats, fusion=parent_params, **kwargs)
  File "/home/manucar/gap_sdk/tools/nntool/quantization/unified_quantization_handler.py", line 40, in quantize
    return cls._quantize(params, in_qs, stats, **kwargs)
  File "/home/manucar/gap_sdk/tools/nntool/quantization/symmetric/quantizers/from_stats_multi_input_pow2.py", line 40, in _quantize
    o_q = QType.from_min_max_pow2(stats['range_out'][0]['min'],
  File "/home/manucar/gap_sdk/tools/nntool/quantization/qtype.py", line 456, in from_min_max_pow2
    return cls.from_min_max(min_val, max_val, dtype=dtype, scaled=False)
  File "/home/manucar/gap_sdk/tools/nntool/quantization/qtype.py", line 439, in from_min_max
    raise ValueError("number cannot be represented with this many bits")
ValueError: number cannot be represented with this many bits
EXCEPTION of type 'ValueError' occurred with message: 'number cannot be represented with this many bits'
```

I tried using 60 samples of my original input .npy data from Python (dtype='int16', I have also tried saving them with other types, but no luck), and the range it's between [+19209,  -12671], while using other samples in the range of [+16100, -12671] seems to work.

Checking the NNTool code I noticed this error it's caused by the following function:

```python
def calc_bits(num, signed=True):
    abs_num = math.floor(math.fabs(num))
    if num < 0 and abs_num > 0:
        abs_num -= 1
    if abs_num == 0:
        return 1
    # calculate number of bits to represent absolute number
    return math.floor(math.log(abs_num) / math.log(2)) + 2

```
and with the above calculation my maximum number [+19209] returns 16 bit, and then with this next check:

```python
int_bits = min_max_to_bits(max_val, min_val, signed=signed)
            if int_bits + (1 if signed else 0) > bits:
                raise ValueError("number cannot be represented with this many bits")
```
It returns 16+1=17 bits for representing my signed [+19209] which is not the right bit value for the representation, and raises the error.

I tried also with the 8 bit quantization option in NNTool and it does not give the previous error, any idea why?

This is my folder with all the files of my project [test_CNN_16_short_input.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/6250093/test_CNN_16_short_input.zip)

Thank you for your support!

Best,

Manuel

sousoux, 2021-04-03:
Thanks. POW2 is rather rusty and is currently having a refresh. Will be fixed in next version.

closed 2021-04-03 by sousoux

Gualor, 2021-06-03:
Any news on this issue? I am also having a hard time with this issue, I have a model that strictly requires 16-bit quantization that I need for an incoming publication. @sousoux any information would be welcomed.

BR,
Lorenzo