GreenWaves-Technologies/gap_sdk #211: Problems in using Platformio with GAP-SDK 3.9.1

MiguelCosta94, 2021-04-06:
Hi,

Is it possible to update the build-pulp-os.py, used by Platformio, for the new version of GAP-SDK (3.9.1). When I try to use this script with the new version of GAP-SDK, I get the following assertion error:
[Error1](https://community.platformio.org/uploads/default/original/2X/a/af88dc90a22a21c4d657085d525bff4729480c1e.png)

To solve it, I updated the directory of pulp-os in build-pulp-os.py script:
`FRAMEWORK_DIR = join(SDK_DIR, “rtos/pulp/pulp-os”)`

After that Platformio did not find the file gap_config.h. One more time, I updated build-pulp-os.py script:
`CCFLAGS=[
…,
("-include", join(SDK_DIR, “tools/pulp_tools/gap_flasher/include”, “gap_config.h”))]`

After doing this, the Build task in Platformio is performed “successfully”. However, it also returns a warning and the program does not execute
`riscv32-unknown-elf/bin/ld: warning: cannot find entry symbol _start; defaulting to 000000001c000000`

How can I solve this? How can I update build-pulp-os.py to use it along GAP-SDK 3.9.1?

Yaooooo, 2021-04-07:
HI,

We stop the maintenance of platformio since a while, therefore, all the script, toolchain, tools cannot use in platformIO directly.
Instead, we have add VScode support in our SDK, you can read the readme "VScode integration" section.

Best

MiguelCosta94, 2021-04-07:
Hi,

Thanks for the reply. I was using Platformio because it was the only way I found to debug code on Gapuino v1. Is Pulp debug bridge still operational? I am facing some issues with OpenOCD.

Best regards

Yaooooo, 2021-04-15:
Hi,

We stopped maintenance on pulp-debug-bridge since a while. 
It should be operational, but the actual sdk install process and the runner doesn't use it anymore. 
Therefore, you may need to compile and use it manually. 

 BTW, for debugging GAP8, unfortunately, the gdb doesn't work well with it, so we are using GVSOC and instruction trace for debug. You can take a look of this video: https://greenwaves-technologies.com/gvsoc-the-full-system-simulator-for-profiling-gap-applications/

BR

closed 2021-04-15 by Yaooooo