GreenWaves-Technologies/gap_sdk #212: undefined reference to `Load_RNN_SQ8_Library'

qlawyer, 2021-04-15:
Hi, 

I am new to the platform and try to get used to it. I followed the exact instructions [here](https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html) to install the SDK. Now I follow the GAPflow tutorial [here](https://youtu.be/0JtxdWdrMpY).

 The nntool instructions worked (almost) as described in the video. But when I hit the gcc ATModel.c -g -o Gentile -I. -I${TILER_INC} -I${TILER_EMU_INC} -I${TILER_CNN_GENERATOR_PATH} -I${TILER_CNN_GENERATOR_PATH_SQ8} -I${NNTOOL_GENERATOR_PATH} -I${TILER_CNN_KERNEL_PATH} -I${TILER_CNN_KERNEL_PATH_SQ8} -I${NNTOOL_KERNELS_PATH} ${TILER_LIB} ${TILER_CNN_GENERATOR_PATH}/CNN_Generator_Util.c ${TILER_CNN_GENERATOR_PATH_SQ8}/CNN_Generators_SQ8.c ${NNTOOL_GENERATOR_PATH}/nntool_extra_generators.c command it returns the following error log. 
 
 /tmp/ccWEk8kK.o: In function `visual_wake_quantModel':
/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/examples/nntool/visual_wake/ATModel.c:32: undefined reference to `Load_RNN_SQ8_Library'
collect2: error: ld returned 1 exit status

I use a Ubuntu 18.04 VM with WM Workstation 16 PRO. The disk (vmdk file) is from osboxes that I accessed from the link given in the documentation and installed the latest anaconda3 2020.11. I could have run the Hello World example successfully. Nntool and tiler paths are accessible by grep. I always source the source.sh whenever I open the terminal, just in case. 

Do you have any recommendation about this issue? I have listed the entire anomalies I have faced to during the installation. In case they could be helpful. Thank you for the help in advance!

The SDK installation did not proceed if I follow the exact steps given in the [documentation](https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html). For example /gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/requirements.txt has the last line ./extern/bfloat16-1.0.tar.gz but it is not accessible from gap_sdk path as there two more folders in between, namely tools and nntool. I edited it as  .tools/nntool/extern/bfloat16-1.0.tar.gz. Then one of the python scripts returned "ImportError: No module named configparser", I changed the import line as it is suggested [here](https://github.com/packt-cli/Packt-Publishing-Free-Learning/issues/70). When I enter the https://greenwaves-technologies.com/autotiler/ URL for the autotiler installation, edited requirement.txt caused an error so that I edit the last line back to ./extern/bfloat16-1.0.tar.gz. Installation finished than I started watching the GAP flow tutorial I mention in the very beginning. During the tutorial [inference section](https://youtu.be/0JtxdWdrMpY?t=467) did not work too due to a mismatch between network input and image dimension format, i.e. the network was adjusted to chw format but the image still was hwc. I assumed this is trivial and skipped. Then I faced the error log that I opened issue for. Once again, thank you for your help in advance.

sousoux, 2021-04-15:
I will check the example but for now just manually remove the call to Load_RNN_SQ8_Library in ATModel.c. The RNN basic kernels are not needed for this example.

sousoux, 2021-04-15:
On the installation issues the makefile in the nntool directory should be called with make sdk and the path should be correct.

qlawyer, 2021-04-15:
> 
> 
> I will check the example but for now just manually remove the call to Load_RNN_SQ8_Library in ATModel.c. The RNN basic kernels are not needed for this example.

It seems it worked out, thanks. 

> 
> 
> On the installation issues the makefile in the nntool directory should be called with make sdk and the path should be correct.

Currently I am not facing a problem, but we can discuss if you are willing to. I can start a new virtual machine and send you the exact log. Otherwise you can close the issue.

closed 2021-04-16 by qlawyer