GreenWaves-Technologies/gap_sdk #216: coudn't compile gap_sdk

eladpar-rnd, 2021-04-26:
hey, im trying to compile gap8 sdk version 3.8.1 
after following the instructions in the README and getting to `make sdk` command im running into an error:
`ImportError: No module named pulp_config:
  File "/home/eladpar/gap_sdk/rtos/pulp/archi_pulp/SConstruct", line 4:
    import pulp_config as plpconfig`

Yaooooo, 2021-04-29:
Hi 
Can you please provide a more specific log? 
-  Which config file or which one you have chosen when doing the sourceme?
-  And the full log when you run `make sdk`?

Furthermore, can you please check: 
- If you have installed all the packages?
- How you manage the python (pip) packages?

For python package management, we strongly recommend use Anaconda, which can help you easily get the clean depandency in your environement.

Yaooooo, 2021-07-02:
No feedback, close.
Please feel free to reopen it if you have further question.

closed 2021-07-02 by Yaooooo