GreenWaves-Technologies/gap_sdk #218: Missing FlatBuffers flexbuffers module

nbaddi, 2021-04-29:
Hi,

I'm trying to run the sdk3.9_shl1.5 gap8_project in occupancy_management,
and I got the following message :

`    from flatbuffers.flexbuffers import GetRoot
ModuleNotFoundError: No module named 'flatbuffers.flexbuffers'
model_rules.mk:41: recipe for target 'BUILD_MODEL_16BIT/lynred.json' failed
make: *** [BUILD_MODEL_16BIT/lynred.json] Error 1`

I'm using a GAPoc B_2.3 board
The sdk version is 3.9.1
platform : Ubuntu 18.04.5 LTS

PS : The sdk3.9 gap8_project works perfectly for me.
And I already installed the nntool requirements listed here: https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/tools/nntool/requirements.txt

Thank you in advance for your help.

Yaooooo, 2021-04-29:
Hi,

This is because you haven't installed the NNTool in your SDK.
You can run these commands below to resolve this:

* make sdk  - this will install all the necessities in your sdk
OR
* make nntool - this will install the nntool in your sdk

nbaddi, 2021-04-29:
Hello, 
This fixes it for me!

Thank you for your fast answer.
Have a nice day!

closed 2021-04-29 by nbaddi