GreenWaves-Technologies/gap_sdk #219: nntool - RESIZE_BILINEAR MissMatchedInputsError - TensorFlow Lite model

gemenerik, 2021-04-30:
When opening my quantized tflite model with RESIZE_BILINEAR layers:

```bash
(NNT) open model/model.tflite -q
open - opening graph file model/model.tflite load_quantization = True
tflite - Importing TFLITE model version 3
Traceback (most recent call last):
  File "/home/rik/miniconda3/envs/gap_sdk_391/lib/python3.7/site-packages/cmd2/cmd2.py", line 1661, in onecmd_plus_hooks
    stop = self.onecmd(statement, add_to_history=add_to_history)
  File "/home/rik/miniconda3/envs/gap_sdk_391/lib/python3.7/site-packages/cmd2/cmd2.py", line 2081, in onecmd
    stop = func(statement)
  File "/home/rik/miniconda3/envs/gap_sdk_391/lib/python3.7/site-packages/cmd2/decorators.py", line 223, in cmd_wrapper
    return func(cmd2_app, args)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/interpreter/commands/open.py", line 118, in do_open
    self.__open_graph(args)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/interpreter/commands/open.py", line 92, in __open_graph
    G = create_graph(graph_file, opts=opts)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/importer/importer.py", line 52, in create_graph
    graph = importer.create_graph(filename, opts)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/importer/tflite2/tflite.py", line 74, in create_graph
    self._import_tflite_graph(G, TFLiteGraph.from_model(model, 0), opts)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/importer/tflite2/tflite.py", line 106, in _import_tflite_graph
    self._import_nodes(G, graph, handlers, all_nodes, self._provisional_outputs, opts)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/importer/tflite2/tflite.py", line 210, in _import_nodes
    params = handler.handle(node, all_nodes=all_nodes, G=G, opts=opts, importer=self)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/importer/tflite2/handlers/handler.py", line 62, in handle
    return ver_handle(node, **kwargs)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/importer/tflite2/handlers/backend/resize_bilinear.py", line 39, in version_1
    return cls._common(node, **kwargs)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/importer/tflite2/handlers/backend/resize_bilinear.py", line 35, in _common
    **kwargs)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/importer/tflite2/handlers/backend/resize_mixin.py", line 43, in _common
    out_shape = params.get_output_size([Dim.unnamed(x[2].known_shape)])[0]
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/graph/types/resizers.py", line 38, in get_output_size
    in_dims = self.clone_dim_with_hints(in_dims)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/graph/types/base.py", line 221, in clone_dim_with_hints
    cloned_dim.apply_naming_hints(hint)
  File "/home/rik/gap_sdk-release-v3.9.1/tools/nntool/graph/dim.py", line 258, in apply_naming_hints
    raise MissMatchedInputsError()
graph.dim.MissMatchedInputsError
EXCEPTION of type 'MissMatchedInputsError' occurred with message: ''

```

[tflite_model.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/6404011/tflite_model.zip)

sousoux, 2021-04-30:
Hi. Sorry about the error. The next version of NNTOOL gives a better error message and describes the problem:
(base) martin:~/projects/nntool$ ./nntool tests/graph/resize_shape_error.tflite 
Welcome to NNTOOL. Type help or ? to list commands.

settings - set log level to INFO
log_level - was: 'INFO'
now: 'INFO'
open - opening graph file tests/graph/resize_shape_error.tflite load_quantization = False
tflite - Importing TFLITE model version 3
EXCEPTION of type 'ValueError' occurred with message: 'multi batch (n=8) convolutions are not supported CONV_2D_0_1'

You have exported your model with a batch size of 8. We do not support multi batch evaluation.

gemenerik, 2021-04-30:
Brilliant! Curious about the new version.

Will certainly help filter out silly mistakes like this. Thanks!

closed 2021-04-30 by gemenerik