GreenWaves-Technologies/gap_sdk #221: Pre-quantized network - autotiler expects different size coefficient files

gemenerik, 2021-05-04:
My on-board network performance was terrible with post-training quantization in nntool. Since PyTorch does not support exporting a quantized model to onnx, I reimplemented my pipeline in TensorFlow for quantization aware training. My model uses depthwise separable convolutions, but since nntool does not support depthwiseconv2d layers, I manually created the depthwise seperable convolutions by using a depthwise conv using groups=no_input_channels, followed by a pointwise conv.  Exported as .tflite (int8, is that okay?) file. nntool seems to correctly read and process my network, but autotiler expects 6 times the data for the layer with groups=6, so it does not seem aware of the fact I'm using a grouped conv.

```bash
Generating Code For User Kernel:                   S3_Conv2d_6x6x3x3
Generating Code For User Kernel:             S6_Conv2d_24x6x1x1_Relu
Generating Code For User Kernel:                  S9_Conv2d_2x24x3x3
Loading coefficient file ./BUILD_MODEL_SQ8BIT/tensors/S1_Model_quant_conv1d_conv2d;model_quant_conv1d_lastvaluequant_fakequantwithminmaxvarsperchannel.tensor: 324 Byte items
Error: For coefficient file ./BUILD_MODEL_SQ8BIT/tensors/S1_Model_quant_conv1d_conv2d;model_quant_conv1d_lastvaluequant_fakequantwithminmaxvarsperchannel.tensor expecting 324 Byte items, got 54
Execution aborted
```

Attached is a dummy network that has the same error as my more complex model;
[tflite_model.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/6421785/tflite_model.zip)

gemenerik, 2021-05-04:
Either way, it would be great if you could add support for tflite depthwiseconv2d layers.

sousoux, 2021-05-04:
Hi. We actually do have support for tflite depthwiseconv2d layers however here it appears that tflite's exporter has been updated with another undocumented feature. Depthwise convs seem to be being exported as Conv2Ds with the only indication the number of input channels versus the number of filter channels. I've added in the ability to spot this. While you are waiting you can try the following patch (which may not be entirely accurate since I'm on a different branch).

`--- a/importer/tflite2/handlers/backend/conv_2d.py
+++ b/importer/tflite2/handlers/backend/conv_2d.py
@@ -71,13 +71,14 @@ class Conv2D(FilterMixin, BackendHandler):
             bias_node = ConstantInputParameters(f'{node.name}_bias',
                                                 dims=Dim.unnamed([filt_out_c]),
                                                 value=np.zeros([filt_out_c], dtype=np.float32))  # TODO - check
-
+        groups = in_c // filt_in_c
         params = Conv2DParameters(node.name,
                                   filt=filt_dim,
                                   stride=StrideDim(
                                       node_opts.StrideH(), node_opts.StrideW()),
                                   dilation=DilationDim(node_opts.DilationHFactor(),
                                                        node_opts.DilationWFactor()),
+                                  groups=groups,
                                   padding=pad,
                                   has_bias=True,
                                   in_dims_hint=[['h', 'w', 'c'], cls.TF_LITE_FILTER_ORDER.copy(), [
`

sousoux, 2021-05-05:
Looking a bit more at your first message I now see that the way you were structuring your model caused this issue but the fix above will support how you were doing it. We *do* support depthwiseconv2d layers and the depthwiseconv2d tflite operator. Not sure why you thought that we did not.

gemenerik, 2021-05-05:
I cannot recall. I initially used them and nntool raised an error that I thought was because of lack of support, must have been something else, probably in the tflite conversion. Thank you for the great support!

closed 2021-05-05 by gemenerik