GreenWaves-Technologies/gap_sdk #222: Cannot install pyhton depencencies

plat-zhaw, 2021-05-05:
Hi there

I'm trying to install the newest gap_sdk. As in the set-up guide  ([https://greenwaves-technologies.com/setting-up-sdk/](url) ) recommended I installed Anaconda. Then I tried to install the python dependencies as instructed in the guides section "Python requirements". This is the error I get when executing the first bash command:

`(base) immanuel@ubuntu1-VirtualBox:~/gap8/gap_sdk$ pip install -r tools/nntool/requirements.txt
WARNING: Requirement './extern/bfloat16-1.0.tar.gz' looks like a filename, but the file does not exist
Processing ./extern/bfloat16-1.0.tar.gz
ERROR: Could not install packages due to an EnvironmentError: [Errno 2] No such file or directory: '/home/immanuel/gap8/gap_sdk/extern/bfloat16-1.0.tar.gz'
`

The second bash command in this section worked without any errors.

I'm using Ubuntu LTS 20.04. Has anyone an idea, what I'm doing wrong?

Thanks for the help.

Regards, Immanuel

sousoux, 2021-05-05:
Just do a 'make sdk' after having sourced sourceme.sh

neoamos, 2021-05-06:
You need to run pip from inside the tools/nntools folder.

Yaooooo, 2021-05-06:
That's why we integrated this command inside of our install process. So you can do just as @sousoux suggested:
1. using "make sdk", it will install all the packages and all necessarities in SDK.
2. If you have already installed other things, but just want to install NNTOOL, then you can run:
make nntool 

Both of these 2 command will installed all the packages for python, which includes this bfloat16-1.0.tar.gz

neoamos, 2021-05-06:
Maybe the readme can be updated to reflect this.  Under the "Full Install" the python installation commands are still listed.  It also lists the command

`pip install -r tools/nntool/requirements.txt`

But this results in the error that @plat-zhaw was having because it is not executed in the nntool folder.

qlawyer, 2021-05-26:
I had the same issue. Edit the gap_sdk/tools/nntool/requirement.txt changing the ./extern with ./tools/nntool/extern. After the packages are installed edit it back. There is a path conflict between package installation and sdk make.

Yaooooo, 2021-07-02:
Resolved in the latest version of SDK.

closed 2021-07-02 by Yaooooo