GreenWaves-Technologies/gap_sdk #223: Bidirectional transport for NINA

neoamos, 2021-05-05:
I noticed in the nina_w10 transport API, the __nina_w10_receive_async function is not implemented.  Is it planned to be implemented?

https://github.com/GreenWaves-Technologies/gap_sdk/blob/master/rtos/pmsis/pmsis_bsp/transport/nina_w10/nina_w10.c

I am using the AI deck on a Crazyflie and I would like to route communication between the GAP8 and the AI deck over the NINA so it is compatible with the lighthouse deck.  I have it working in the GAP8->NINA->CF direction, but I found that the SPI transport functions in the opposite direction is not implemented.  We can do CF->GAP8 communication over UART1, but with the caveat that the LED on the lighthouse has to be disabled and UART1 interferes with the hyperflash memory on the AI deck.

Yaooooo, 2021-05-06:
Hi 

Sorry for that, it's really very low priority for us, so we don't have a date about when it will be implemented. 
I suggest you to implement it by yourself, we have lots of examples on using SPI and UART, and if you have any question, we are happy to help.

neoamos, 2021-05-06:
Ok, that's good to know.  I can look into implementing it myself and possibly submit a pull request.

Yaooooo, 2021-07-02:
Thanks, please feel free to post your question and pull request.

closed 2021-07-02 by Yaooooo