GreenWaves-Technologies/gap_sdk #225: Unfused relu causes nntool error

neoamos, 2021-05-07:
If you have a network with a relu activation that isn't fused to another operator by the TFlite converter, nntool throws this error:
```
Traceback (most recent call last):
  File "/gap_sdk/tools/nntool/nntool", line 101, in <module>
    main()
  File "/gap_sdk/tools/nntool/nntool", line 85, in main
    mod.generate_code(args)
  File "/gap_sdk/tools/nntool/interpreter/generator.py", line 93, in generate_code
    write_template(G, code_gen, opts['model_directory'], opts['model_file'], code_template, "model")
  File "/gap_sdk/tools/nntool/interpreter/generator.py", line 47, in write_template
    model = template(G, code_generator=code_gen)
  File "/gap_sdk/tools/nntool/generation/default_template.py", line 193, in default_template
    return execute_template(generator_template_v3, G, naming_convension, code_generator)
  File "/gap_sdk/tools/nntool/generation/default_template.py", line 187, in execute_template
    return template_function(G, code_generator)
  File "/gap_sdk/tools/nntool/generation/default_template.py", line 100, in generator_template_v3
    ${gen.global_generator(indent=2)},
  File "/gap_sdk/tools/nntool/generation/code_generator.py", line 252, in global_generator
    self.generate_constants()
  File "/gap_sdk/tools/nntool/generation/code_generator.py", line 291, in generate_constants
    self.execute_phase("globals", anode, qrec, pnode, fnode)
  File "/gap_sdk/tools/nntool/generation/generators/generator_decorators.py", line 79, in execute_phase
    this_res = gen['func'](self, param, qrec, *args, **kwargs)
  File "/gap_sdk/tools/nntool/generation/generators/globals/mult8_infos_generator.py", line 50, in mult8_infos_generator
    act_infos(gen, pnode, pnode, pnode, gen.G.quantization[NodeId(pnode)])
  File "/gap_sdk/tools/nntool/generation/generators/globals/mult8_infos_generator.py", line 141, in act_infos
    actscale = act_q.scale_mul_biases_q.qbiases[0]
TypeError: 'NoneType' object is not subscriptable
```

This is the definition of the model:
```
def unfused_relu(img_width, img_height, img_channels, output_dim):
   # Input
   img_input = Input(shape=(img_height, img_width, img_channels))

   x1 = Conv2D(32, (5, 5), strides=[2,2], padding='same')(img_input)
   x1 = MaxPooling2D(pool_size=(3, 3), strides=[2,2])(x1)

   # First residual block
   x2 = Activation('relu')(x1) # Unfused relu causes error
   x2 = Conv2D(32, (3, 3), strides=[2,2], padding='same')(x2)

   x2 = keras.layers.normalization.BatchNormalization()(x2)
   x2 = Activation('relu')(x2)
   x2 = Conv2D(32, (3, 3), padding='same')(x2)

   x1 = Conv2D(32, (1, 1), strides=[2,2], padding='same')(x1)
   x3 = add([x1, x2])

   x = Flatten()(x3)
   x = Activation('relu')(x)

   x = Dense(output_dim)(x)
   nums = Activation('softmax')(x)

   model = Model(inputs=[img_input], outputs=[nums])
   print(model.summary())

   return model
```

And here is a visualization of the network:
[image]

This is with gap_sdk 3.9.1.  I tested it with sigmoid and it does not have the same issue

sousoux, 2021-05-08:
COuld you attach the tflite model (untrained/dummy weights) to this issue please?

neoamos, 2021-05-08:
[model.tflite.gz](https://github.com/GreenWaves-Technologies/gap_sdk/files/6445139/model.tflite.gz)
Yeah, here it is

sousoux, 2021-05-09:
This is fixed and will be in the next release

closed 2021-05-09 by sousoux