GreenWaves-Technologies/gap_sdk #227: Error management Autotiler install

knmcguire, 2021-05-24:
Hi!

We had two reports of users installing the autotiler wrong or used a wrong URL. Since the installer did not give an error message , t took a long time to figure out what was actually happening and figured out that they needed to install it again. Perhaps it would be better to give a clear error if the wrong URL is used.

Yaooooo, 2021-05-25:
Hi,

Actually, we do have some error messages like this:
```
(base) yao@yao-tuxedo:~/projects/sdk/gap_sdk_github$ make autotiler 
if [ -e /home/yao/projects/sdk/gap_sdk_github/tools/autotiler_v2/Makefile ]; then make -C /home/yao/projects/sdk/gap_sdk_github/tools/autotiler_v2 all; fi
make[1]: Entering directory '/home/yao/projects/sdk/gap_sdk_github/tools/autotiler_v2'
python3 get_tiler.py
Registration is required to load the GAP8 AutoTiler library

In case you have already registered, you can directly enter the link
to the AutoTiler library that you received from an email.
Otherwise just press Enter.
Enter URL from email: aiudkhkefhnkf@kasdnjksefd
Please enter a valid URL, it must start with https:// and
end with /.
Enter URL from email: https://greenwaves-technologies.com
Please enter a valid URL, it must start with https:// and
end with /.
Enter URL from email: https://greenwaves-technologies.com/.
Please enter a valid URL, it must start with https:// and
end with /.
Enter URL from email: https://greenwaves-technologies.com/
mkdir -p lib
rm -rf lib/libtile*
./download_tiler.sh
--2021-05-25 11:56:43--  https://greenwaves-technologies.com/libtile.2.0.1.a
Resolving greenwaves-technologies.com (greenwaves-technologies.com)... 35.197.241.59
Connecting to greenwaves-technologies.com (greenwaves-technologies.com)|35.197.241.59|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2021-05-25 11:56:43 ERROR 404: Not Found.

make[1]: *** [Makefile:23: lib/libtile.a] Error 1
make[1]: Leaving directory '/home/yao/projects/sdk/gap_sdk_github/tools/autotiler_v2'
make: *** [Makefile:115: autotiler] Error 2
```

And you think it's not enough? Can you be more precise on what kind of message we can add for help?

Yao

knmcguire, 2021-05-26:
Ah that seems to be sufficient indeed. The issue that I linked from the other repository: https://github.com/bitcraze/AIdeck_examples/issues/43, indicated that it didn't see anything at all... So I wonder how that is possible. However they did use a older version of the SDK

knmcguire, 2021-06-07:
Hi! Probably this issue is not super urgent to check out so let's just close it for now. I'm not sure how to fix the issue that one of our users had either. if I'm able to recreate it properly i'll open it again

closed 2021-06-07 by knmcguire

Yaooooo, 2021-06-07:
Pls let us know if you have more clear idea. Just fyi, we may provide an official container in 1-2 months