GreenWaves-Technologies/gap_sdk #231: 16-bit quantization doubt

Manucar, 2021-06-04:
Hi everyone!

While waiting the fix for my previous issue #210, (that for the moment I fixed with a quick adjustment but not knowing if this could break something else), I was experimenting with NNTOOL the quantization scheme for the input of the network.

The .tflite model is loaded into NNTOOL and quantized witht the following commands:
```
open model.tflite
adjust
fusions --pow2
aquant model/data/* --nptype float64 --scheme POW2
```

However, when i run the 'qshow' command, to see the quantization strategy for my inputs and outputs, it shows:
```
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
| Step |          Name          |   Type    |        In         |  Out   | Weights | Bias  | Mulbias |  Calc  |  Acc   |
+======+========================+===========+===================+========+=========+=======+=========+========+========+
|  0   | input_1                | symmetric |                   | Q15.1  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  1   | CONV_2D_0_0_r_hwc_chw  | symmetric | Q15.1             | Q15.1  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  2   | ADEF_LMconv2d_9Conv2D  | symmetric |                   | Q0.16  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  3   | ADEF_LMconv2d_9BiasAdd | symmetric |                   | Q7.9   |         |       |         |        |        |
|      | ADEF_LMc               |           |                   |        |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  4   | CONV_2D_0_0            | symmetric | Q15.1,Q0.16,Q7.9  | Q15.17 | Q0.16   | Q7.9  | None    | Q15.17 | Q15.17 |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  4   | CONV_2D_0_0_activation | symmetric | Q15.17            | Q13.3  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  4   | CONV_2D_0_0_fusion     | symmetric | Q15.1,Q0.16,Q7.9  | Q13.3  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  5   | AVERAGE_POOL_2D_0_1    | symmetric | Q13.3             | Q13.3  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  6   | ADEF_LMconv2d_10Conv2D | symmetric |                   | Q5.11  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  7   | ADEF_LMconv2d_10BiasAd | symmetric |                   | Q6.10  |         |       |         |        |        |
|      | dADEF_LM               |           |                   |        |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  8   | CONV_2D_0_2            | symmetric | Q13.3,Q5.11,Q6.10 | Q18.14 | Q5.11   | Q6.10 | None    | Q18.14 | Q18.14 |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  8   | CONV_2D_0_2_activation | symmetric | Q18.14            | Q14.2  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  8   | CONV_2D_0_2_fusion     | symmetric | Q13.3,Q5.11,Q6.10 | Q14.2  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  9   | AVERAGE_POOL_2D_0_3    | symmetric | Q14.2             | Q14.2  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  10  | ADEF_LMconv2d_11Conv2D | symmetric |                   | Q4.12  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  11  | ADEF_LMconv2d_11BiasAd | symmetric |                   | Q5.11  |         |       |         |        |        |
|      | dADEF_LM               |           |                   |        |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  12  | CONV_2D_0_4            | symmetric | Q14.2,Q4.12,Q5.11 | Q18.14 | Q4.12   | Q5.11 | None    | Q18.14 | Q18.14 |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  12  | CONV_2D_0_4_activation | symmetric | Q18.14            | Q13.3  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  12  | CONV_2D_0_4_fusion     | symmetric | Q14.2,Q4.12,Q5.11 | Q13.3  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  13  | AVERAGE_POOL_2D_0_5    | symmetric | Q13.3             | Q13.3  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  14  | AVERAGE_POOL_2D_0_5_TO | symmetric | Q13.3             | Q13.3  |         |       |         |        |        |
|      | UT_0                   |           |                   |        |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  15  | ADEF_LMdense_3MatMul   | symmetric |                   | Q3.13  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  16  | ADEF_LMdense_3BiasAddR | symmetric |                   | Q2.14  |         |       |         |        |        |
|      | eadVaria               |           |                   |        |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  17  | FULLY_CONNECTED_0_7    | symmetric | Q13.3,Q3.13,Q2.14 | Q13.3  | Q3.13   | Q2.14 | None    | Q16.16 | Q16.16 |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  18  | RESHAPE_0_11           | symmetric | Q13.3             | Q13.3  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+
|  19  | output_1               | symmetric | Q13.3             | Q13.3  |         |       |         |        |        |
+------+------------------------+-----------+-------------------+--------+---------+-------+---------+--------+--------+

```
So, no quantization strategy is suggested for the input layer. Could it be because I acquired my data directly on int16 variables and the network is also trained with these data, and the input doesn't need a quantization at all, and provided as is?

Thanks for your time!

Best,
Manuel

marco-fariselli, 2021-07-02:
Ciao @Manucar 

In this case the quantized network expects the input in the Q1.15 format (1 integer bit and 15 decimal bits). In nntool you still need to provide float inputs, the tool will make the conversion from float to Q1.15 for you, but on the platform you will need to be sure that the short int input will match this Q1.15 scheme.

Manucar, 2021-07-02:
Ciao @marco-fariselli, thank you for your answer!

Since I am new to this type of quantization, could you please explain me how this conversion to this Q15.1 scheme can actually be done? Considering for example I have a signed short value of 20000, how this number should be converted?

marco-fariselli, 2021-07-02:
The quantization scheme applied in the 16bits quantization is a standard DSP quantization scheme QM.N, which means that each value of the tensor is represented with M (1 of which is for the sign) integer bits and N decimal bits (M+N=16). During nntool quantization (calibration) we collect the statistic of each tensor and we make sure that the max/min values will be representable in int16, so if a tensor goes from -18.82:20.32 we need to be sure to represent 20.32 --> M=6 integer bits (-32:31) and thus N=10 (Q6.10). To change from float to quantized values and viceversa you can use the formulas:
f = S * q
Where S is the scaling factor (the minimum representable float value) S = 2^(-N)
and:
q = clip(round(f/S), -32768, 32767))

Manucar, 2021-07-02:
Thank you for your explanation!

closed 2021-07-02 by Manucar