GreenWaves-Technologies/gap_sdk #237: Informations about the pmsis interface

MattAbrs, 2021-06-11:
Hi,
I'm writing this message because I need some help about the pmsis interface.
In fact, what I'm trying to do is to create an interruption in my code.
For that, I'm using your gpio_irq_cb project with the Gapoc_b and the Raspberry Pi 3b+.
I made something which is waiting that the raspberry write on the output GPIO but I'm not sure it's the good way to do that.

First I configured my gpio output and input with the code below :
I'm using the variables  RASPBERRY_ACTIVITY_TRIGGER as the Raspberry output and RASPBERRY_ACTIVITY_ACQ as the Raspberry input.
```
int32_t errors;
int32_t value;
struct pi_gpio_conf gpio_conf = { 0 };
pi_gpio_conf_init(&gpio_conf);
pi_open_from_conf(&gpio, &gpio_conf);
errors = pi_gpio_open(&gpio);
if (errors) {
    printf("Error opening GPIO %d\n", errors);
    pmsis_exit(errors);
}

/* Configure gpio input */
pi_gpio_e gpio_raspberry_trigger = RASPBERRY_ACTIVITY_TRIGGER;
pi_gpio_notif_e irq_type = PI_GPIO_NOTIF_RISE;
pi_gpio_flags_e cfg_flags = PI_GPIO_INPUT | PI_GPIO_PULL_DISABLE | PI_GPIO_DRIVE_STRENGTH_LOW;
pi_task_callback(&cb_gpio, __pi_cb_gpio, (void*)gpio_raspberry_trigger);
pi_gpio_pin_configure(&gpio, gpio_raspberry_trigger, cfg_flags);
pi_gpio_pin_task_add(&gpio, gpio_raspberry_trigger, &cb_gpio, irq_type);
pi_gpio_pin_notif_configure(&gpio, gpio_raspberry_trigger, irq_type);

/* Configure gpio outputs */
pi_gpio_pin_configure(&gpio, RASPBERRY_ACTIVITY_ACQ, PI_GPIO_OUTPUT);
pi_gpio_pin_write(&gpio, RASPBERRY_ACTIVITY_ACQ, 0);
```

My __pi_cb_gpio function is the next one :

```
void __pi_cb_gpio(void *arg)
{
    done++;
    pi_gpio_e gpio_pin = (pi_gpio_e) arg;
    uint32_t val = 0;
    pi_gpio_pin_read(&gpio, gpio_pin, &val);
    printf("First GPIO callback GPIO_A%d %d %d\n", gpio_pin & 0xFF, done, val);
}
```

The code of the test is the following one:

```
pi_task_block(&cb_gpio);
while (1)
{
    pi_gpio_pin_write(&gpio, RASPBERRY_ACTIVITY_ACQ, 1);
    pi_task_callback(&cb_gpio, __pi_cb_gpio, (void*)gpio_raspberry_trigger);
    pi_task_wait_on(&cb_gpio);
    pi_gpio_pin_write(&gpio, RASPBERRY_ACTIVITY_ACQ, 0);
    pi_time_wait_us(2000000);
}
pi_gpio_close(&gpio);
pmsis_exit(errors);
```

What I wanted is to use the callback function to stop the code. In your docs, you write that we can't use callback function to interrupt the code. So, What I thought is that I did the interruption in the wrong way. Even if the interruption works thanks to the pi_task_wait_on function. 

I hope you can help me to understand more clearly the pmsis interface.

Thanks, have a good day.

PaulLuperini, 2021-07-05:
Hi @MattAbrs,
Sorry for the late answer.
Is your issue still relevant? Or did you already fix it?

I see that you are using the same `&gpio` for input and output.
In order to use 2 GPIOs, you must use 2 different `struct device`.
Also `cb_gpio` is initialized with `pi_task_block ` and then with `pi_task_callback`, maybe this is causing some issue :thinking:.

If this fixes your issue, feel free to close it.

MattAbrs, 2021-07-06:
Thanks for your answer !

closed 2021-07-06 by MattAbrs