GreenWaves-Technologies/gap_sdk #239: efuse_burner example not working

Skydell, 2021-06-24:
Hello, while using the Gappoc_A 0.3 i was trying to boot from flash. But it didn't work with examples like blnk_led so I tried the efuse_burner example (in examples/pmsis/features/efuse_burner)  to see if everything was working correctly.

I did the following inputs in the above repository:
make clean all run EFUSE_WRITE=1

And the result i got was :
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
Loading binary through JTAG
Info : tcl server disabled
Info : telnet server disabled
Warn : Burst read timed out
Entering main controller
Writing efuse 50 with value 0x12
Read efuse 50: 0x0

This result shows that it doesn't write correctly. 
Do you know something to fix this ? Is there a special input to do to be able to use this example ?
Kind regards.

Yaooooo, 2021-07-02:
Hi,

The efuse example is just show you how to burn one efuse in the chip.
For more details, you can read this document from our website: https://greenwaves-technologies.com/wp-content/uploads/2021/03/AN004_BootFromFlash.pdf

We have also a fuser tool in the tools: https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/tools/gap_fuser

Hope this can help you.

Skydell, 2021-07-06:
Hey,

Thanks for your answer. 
We managed to make the efuse burner example work by closing the dip switch 3.
This made us burn the expected efuse.
I still have a question around flashing, but that is not related to efuse burner. I will send you an email about it.

Thanks for your help.

closed 2021-07-06 by Skydell