GreenWaves-Technologies/gap_sdk #241: Error when running MNIST Example

rallen10, 2021-06-28:
I am trying to run the [mnist example](https://github.com/GreenWaves-Technologies/gap_sdk/tree/master/examples/nntool/mnist) but running into a `FileNotFoundError` when trying to execute `make run all platform=gvsoc`.

Note that I am using a docker environment to run the gap_sdk examples as described here: https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/docker-gap8/

Here are my steps:

Pafrak, 2021-06-29:
Hello Rallen10,
 can you try with `make all run platform=gvsoc` ?

rallen10, 2021-06-29:
@Pafrak I feel so stupid. Yes `make all run platform=gvsoc` gets past the `FileNotFoundError`; however now I am getting a module error.

```bash
Traceback (most recent call last):
  File "/gap_sdk/tools/nntool/nntool", line 28, in <module>
    from interpreter.commands.open import add_open_options
  File "/gap_sdk/tools/nntool/interpreter/commands/open.py", line 22, in <module>
    from importer.importer import create_graph
  File "/gap_sdk/tools/nntool/importer/importer.py", line 19, in <module>
    from .onnx.onnx import OnnxImporter
  File "/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 19, in <module>
    from graph.matches.matchers.duplicate_constants import MatchDuplicateConstants
  File "/gap_sdk/tools/nntool/graph/matches/matchers/duplicate_constants.py", line 18, in <module>
    from graph.types import ConstantInputParameters, NNEdge
  File "/gap_sdk/tools/nntool/graph/types/__init__.py", line 16, in <module>
    from graph.types.activations import (ActivationParameters,
  File "/gap_sdk/tools/nntool/graph/types/activations.py", line 19, in <module>
    from expressions.symbolic.basic import HSigmoid, HTanh, Relu, Sigmoid, TanH
  File "/gap_sdk/tools/nntool/expressions/symbolic/basic.py", line 17, in <module>
    from bfloat16 import bfloat16
ModuleNotFoundError: No module named 'bfloat16'
```
Perhaps my docker container is just missing this dependency? I'll try to install the necessary dependencies and report back

Thank you for the help

Yaooooo, 2021-06-29:
Yes, this package is required in python3, suggested with version 1.0.
Please check the file: gap_sdk/tools/nntool/requirements.txt

BTW, all the packages will be installed when you run "make nntool" in the
sdk.
However, our install process won't take care of package version conflict.
That's way we suggest to use something like Anaconda or Pipenv.

rallen10, 2021-06-29:
I went through the process of installing `bfloat16` and some other missing dependencies (`iteration_utilities` and `graphviz`) but then ran into new errors. However when I restarted my docker container, everything seemed to "just work" without needing to perform any pip installs. I think my build had been fouled by my earlier mistakes within the docker container, but restarting it cleared out those mistakes.

So in the end, everything seemed to work as intended when using [this docker container](https://www.bitcraze.io/documentation/repository/AIdeck_examples/master/getting-started/docker-gap8/) and the following instructions:

```bash
  export GAP_SDK_VERSION=3.8.1
  sudo docker run --rm -it gapsdk:${GAP_SDK_VERSION} /bin/bash
  source /gap_sdk/configs/ai_deck.sh
  cd /gap_sdk/examples/nntool/mnist
  make all run platform=gvsoc
  ```

Thank you for the help everyone!

closed 2021-06-29 by rallen10