GreenWaves-Technologies/gap_sdk #242: can't install gap_sdk

SpeakNow-Wu, 2021-07-01:
i'm using ubuntu 20.04，and without the GAP8 board(i just need to use emulator will it influence?).when I install the sdk,this step especially
[image]
it shows like
[image]
though I updated sphinx it doesn't work.
and when I tried to run the "helloword" example,it shows
[image]
until I type "make clean all run platform=gvsoc runner_args=--vcd" it showed the correct result.
what should I do for it. thanks

PaulLuperini, 2021-07-02:
Hi @SpeakNow-Wu,

Sorry for the documentation generation inconvenience.
It seems that new versions of `breathe` package do not support old versions of `Sphinx`, according to this [issue](https://github.com/sphinx-doc/sphinx/issues/7664).
What is your version of `Sphinx` ?
Could you try to update it with `pip3 install --upgrade Sphinx` ? You may also need to do the same with `breathe`.

We will add more specific versions in `doc/requirements.txt` to avoid this issue.

Your second issue seems to be related to the default selected platform.
By default, the SDK assumes that you are using a board (equivalent to `platform=board`).
If you want to use GVSoC simulator, you need to specify `platform=gvsoc` when doing a `make clean all run`.
So to run the `helloworld` example on GVSoC, use `make clean all run platform=gvsoc`.

If all your questions are answered and your issues fixed, feel free to close this issue.
Thank you for your feedback,
Have a good day :)

SpeakNow-Wu, 2021-07-02:
hi @PaulLuperini thanks for your response, my Sphinx version is 1.8.5, after I update both sphinx and breathe it still shows the same result. one more silly question, is this software like an application or just I can work it on the terminal? thanks

PaulLuperini, 2021-07-02:
Weird :thinking: 
What is the output of `sphinx-build --version`?
Can you try to go to the `doc` folder and run `make clean html`?

Can you give me more details about your second question? What is this software you are talking about?
The GAP SDK allows you to create applications that are intended to run on GAP processors, not on your computer.
GVSoC is a tool used to simulate GAP processors, to ease development.

PaulLuperini, 2021-07-05:
Hi @SpeakNow-Wu,
`1.8.5` is the version of the Ubuntu package `python3-sphinx`. It may be overriding your local python3 packages.
If you installed it, could you try uninstalling it? Is `~/.local/bin` in your `PATH` ?

SpeakNow-Wu, 2021-07-06:
thank you Paul,  I just run it on Ubuntu 18.04 and it works now, I don't know what happened on 20.04, but the package of sphinx really doesn't work.

closed 2021-07-06 by SpeakNow-Wu