GreenWaves-Technologies/gap_sdk #245: SDK Installation Error

qlawyer, 2021-07-13:
Hi, 

I am working with the latest master repository and latest version of Anaconda. During the SDK installation I get such an error: 

make[2]: Entering directory '/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/rtos/pulp/archi_pulp'
scons
scons: Reading SConscript files ...
ImportError: No module named configparser:
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/rtos/pulp/archi_pulp/SConstruct", line 4:
    import pulp_config as plpconfig
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/gap-configs/python/pulp_config.py", line 19:
    import json_tools as js
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/gapy/json_tools.py", line 23:
    import configparser
Makefile:2: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/rtos/pulp/archi_pulp'
Makefile:7: recipe for target 'do.archi' failed
make[1]: *** [do.archi] Error 2
make[1]: Leaving directory '/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/rtos/pulp'
Makefile:131: recipe for target 'pulp-os' failed
make: *** [pulp-os] Error 2 

However, I can import configparser in the very terminal. I am clueless about how I can complete the installation. Can you help me? 

Thank you for your support in advance.

qlawyer, 2021-07-13:
Hi @Pafrak, 

Apparently the error occurred because I haven't get it installed for python2. I first installed the pip2 then I installed the configparser using all conda install configparser, pip3 install configparser and pip2 install configparser again. Then it worked. It also errored out no module named sphinx-rtd-theme and breathe. I installed them too. Now it finished the installation as before. 

I am not sure what changed since the previous branch but it could be nice to put these two library in the requirements.txt and pip2 to [the installation guide](https://greenwaves-technologies.com/manuals/BUILD/HOME/html/index.html) too. Just a suggestion though. 

Thank you for your assistance!  

Bests

closed 2021-07-13 by qlawyer

crissmath, 2021-07-22:
Thank you very much, it worked for me.

Apparently, you need to install configparser for python2.

:)

Yaooooo, 2021-07-22:
Hi,

I found this issue remains only for < Ubuntu 18, so we have to install the
package in python2 as well.
From Ubuntu 20, we don't need to do it anymore.