GreenWaves-Technologies/gap_sdk #247: Issue with nntool for custom model

brupa9, 2021-07-13:
Hello,

I created an SSD model by using transfer learning on the efficientdet_lite0 model. When I try to generate the ATModel.c I get the following issue : 

>  EXCEPTION of type 'NotImplementedError' occurred with message: 'Don't know how to generate kernel for parameter type DEQUANTIZE_0_265 quantize. Perhaps you need to run fusions -a expression_matcher.' 

Running fusions doesn't change anything, is there a way to workaround this error ?

Thanks for your help

sousoux, 2021-12-07:
Close. No activity.

closed 2021-12-07 by sousoux

jessicatingley, 2024-08-20:
Hi, I am wondering how this issue was resolved. I am currently getting the same issue with a different parameter type:
[image]

Running fusions doesn't change anything as in the previous user's issue
Thank you for your help.

brupa9, 2024-08-21:
@jessicatingley Unfortunately I've never resolved this issue