GreenWaves-Technologies/gap_sdk #248: Sending message through the UART

Lambpaul, 2021-07-13:
Hi,

I'm trying to send message using the UART. I flashed a program on my card and I tried to use a FTDI to see those messages on CoolTerm. I can't see any characters on CoolTerm but I can see exchanges. I tried to use minicom instead of CoolTerm and in minicom I can see unknown symbols with question marks... 
My code is this one :
```
/**
 * This example blinks a user led on Gapuino boards(Gapuino_V2),
 * Gapoc_a and Gapoc_b boards.
 */

/* PMSIS includes */
#include "pmsis.h"
#include "bsp/bsp.h"

void blink_led(void)
{
    printf("Entering main controller\n");
    
    struct pi_device uart;
    struct pi_uart_conf conf;
    pi_uart_conf_init(&conf);
    conf.enable_tx = 1;
    conf.enable_rx = 1;
    conf.baudrate_bps = 115200;
    conf.stop_bit_count=1;
    conf.parity_mode=0;
    conf.word_size=8;
    pi_open_from_conf(&uart, &conf);
    if (pi_uart_open(&uart))
    {
        printf("Uart open failed !\n");
        pmsis_exit(-1);
    }

    printf("Blinking USER LED\n");
    pi_gpio_pin_configure(NULL, GPIO_USER_LED, PI_GPIO_OUTPUT);
    while (1)
    {
        pi_gpio_pin_write(NULL, GPIO_USER_LED, 0);
        pi_time_wait_us(1000000);
        //printf("blink\n");
        char c = 0x55;
        pi_uart_write(&uart, &c , 1);
        pi_gpio_pin_write(NULL, GPIO_USER_LED, 1);
        pi_time_wait_us(1000000);
    }

    pmsis_exit(0);
}

/* Program Entry. */
int main(void)
{
    printf("\n\n\t *** PMSIS Blink LED ***\n\n");
    return pmsis_kickoff((void *) blink_led);
}
```
As you can see my code is a mixed between blink_led.c and uart_helloworld.c .
I tried to modify those 4 parameters:
conf.baudrate_bps = 115200;
conf.stop_bit_count=1;
conf.parity_mode=0;
conf.word_size=8;

I also tried to change those parameters in CoolTerm and minicom but nothing change.

Do you have any idea how to solve thise problem ?

Kind regards.

Pafrak, 2021-07-13:
hi @Lambpaul 
can you try to declare you `char c` variable out of the blink_led function with the prefix PI_L2 and test it again?

Lambpaul, 2021-07-15:
Thanks for you quick response.
Now my code is like this : 
```
/**
 * This example blinks a user led on Gapuino boards(Gapuino_V2),
 * Gapoc_a and Gapoc_b boards.
 */

/* PMSIS includes */
#include "pmsis.h"
#include "bsp/bsp.h"

PI_L2 char c = 0x55;

void blink_led(void)
{
    printf("Entering main controller\n");
    
    struct pi_device uart;
    struct pi_uart_conf conf;
    pi_uart_conf_init(&conf);
    conf.enable_tx = 1;
    conf.enable_rx = 1;
    conf.baudrate_bps = 115200;
    conf.stop_bit_count=1;
    conf.parity_mode=0;
    conf.word_size=8;
    pi_open_from_conf(&uart, &conf);
    if (pi_uart_open(&uart))
    {
        printf("Uart open failed !\n");
        pmsis_exit(-1);
    }

    printf("Blinking USER LED\n");
    pi_gpio_pin_configure(NULL, GPIO_USER_LED, PI_GPIO_OUTPUT);
    while (1)
    {
        pi_gpio_pin_write(NULL, GPIO_USER_LED, 0);
        pi_time_wait_us(1000000);
        //printf("blink\n");
        
        pi_uart_write(&uart, &c , 1);
        pi_gpio_pin_write(NULL, GPIO_USER_LED, 1);
        pi_time_wait_us(1000000);
    }

    pmsis_exit(0);
}

/* Program Entry. */
int main(void)
{
    printf("\n\n\t *** PMSIS Blink LED ***\n\n");
    return pmsis_kickoff((void *) blink_led);
}
```
I have the same problem as before. Only unknown characters appear in minicom.
Maybe it's coming from my flashing procedure which is unconventional. 
Once the program is flashed on my Gappoc-A board I have to shut it down, unplug the sipeed , power the board back on and plug back the sipeed in order to have my program running on the board. Any other procedure doesn't work...

Is it the correct procedure for flashing ? 

Kind regards.

Pafrak, 2021-07-19:
@Lambpaul 
can you try to also add `io=uart` into the Makefile ? because you jtag is not connected printf can get your code stuck.

Lambpaul, 2021-07-21:
I tryed to add io=uart when compiling. Now nothing appears in minicom... 

I can see that nothing is on RX now because on Coolterm there is no data transmitted.
I also tried to print my `PI_L2 char c ` with `printf(&c)` instead of  `pi_uart_write(&uart, &c , 1);` but still nothing appears in minicom.

sivasivarajah, 2021-07-21:
Hi @Lambpaul,

There are few things I noted about your UART config. You should use the enum types declared in *rtos/pmsis/pmsis_api/include/pmsis/drivers/uart.h* to set up the UART, especially the word size. 
UART is using uDMA to transfer data between L2 memory and ext. Therefore your data buffer must be in L2. You can ensure data are in L2 either allocating buffer in L2(or declare global var in L2 as you have done).