GreenWaves-Technologies/gap_sdk #249: Build Model Fail

qlawyer, 2021-07-14:
Hey, I try to run the visual_wave or image classification examples. 

visual_wake throws this error: 

File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/nntool", line 28, in <module>
    from interpreter.commands.open import add_open_options
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/interpreter/commands/open.py", line 22, in <module>
    from importer.importer import create_graph
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/importer/importer.py", line 19, in <module>
    from .onnx.onnx import OnnxImporter
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 17, in <module>
    from graph.constant_store import ConstantStore
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/graph/constant_store.py", line 24, in <module>
    from sklearn.cluster import KMeans
  File "/home/osboxes/anaconda3/lib/python3.8/site-packages/sklearn/__init__.py", line 76, in <module>
    from .base import clone
  File "/home/osboxes/anaconda3/lib/python3.8/site-packages/sklearn/base.py", line 16, in <module>
    from .utils import _IS_32BIT
  File "/home/osboxes/anaconda3/lib/python3.8/site-packages/sklearn/utils/__init__.py", line 15, in <module>
    from .murmurhash import murmurhash3_32
  File "__init__.pxd", line 242, in init sklearn.utils.murmurhash
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject
../common/model_rules.mk:52: recipe for target 'BUILD_MODEL_SQ8BIT/vww.json' failed
make: *** [BUILD_MODEL_SQ8BIT/vww.json] Error 1

[image classification](https://github.com/GreenWaves-Technologies/image_classification_networks) throws this one: 

Traceback (most recent call last):
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/nntool", line 28, in <module>
    from interpreter.commands.open import add_open_options
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/interpreter/commands/open.py", line 22, in <module>
    from importer.importer import create_graph
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/importer/importer.py", line 19, in <module>
    from .onnx.onnx import OnnxImporter
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/importer/onnx/onnx.py", line 17, in <module>
    from graph.constant_store import ConstantStore
  File "/home/osboxes/gap_riscv_toolchain_ubuntu_18/gap_sdk/tools/nntool/graph/constant_store.py", line 24, in <module>
    from sklearn.cluster import KMeans
  File "/home/osboxes/anaconda3/lib/python3.8/site-packages/sklearn/__init__.py", line 76, in <module>
    from .base import clone
  File "/home/osboxes/anaconda3/lib/python3.8/site-packages/sklearn/base.py", line 16, in <module>
    from .utils import _IS_32BIT
  File "/home/osboxes/anaconda3/lib/python3.8/site-packages/sklearn/utils/__init__.py", line 15, in <module>
    from .murmurhash import murmurhash3_32
  File "__init__.pxd", line 242, in init sklearn.utils.murmurhash
ValueError: numpy.ndarray size changed, may indicate binary incompatibility. Expected 88 from C header, got 80 from PyObject
common/model_rules.mk:59: recipe for target 'BUILD_MODEL_SQ8BIT/mobilenet_v2_0_75_160_quant.json' failed
make: *** [BUILD_MODEL_SQ8BIT/mobilenet_v2_0_75_160_quant.json] Error 1 

I guess the origin of the errors are the same. Do you have any recommendation?

Thank you for your help in advance. 

Best Regards

Yaooooo, 2021-07-19:
Hi @qlawyer ,

It's a conflict package problem in python, hard to say what's the reson, but mainly on `numpy, cython and scikit-xxxxx`
I also suffering this when I cannot use anaconda, that's why we strongly recomend it. 

I remember I finally make it work (don't understand how), by starting from scrach and with these package versions:
Add cython==0.29.21
change numpy==1.19.4 (this one is not the same as in the [file](https://github.com/GreenWaves-Technologies/gap_sdk/blob/278fd42563424c3557ed61d8d997d9a7bdcc1530/tools/nntool/requirements.txt) )

Best regards

qlawyer, 2021-07-22:
Hi @Yaooooo, 

Actually I use Anaconda. I followed the setup guide. This must be what happens for everyone who starts over. Maybe there is a conflict due to Anaconda version. Let me try to fix the packages. But it may take a few days to get back to this work. 

Best Regards

closed 2021-07-22 by qlawyer