GreenWaves-Technologies/gap_sdk #250: himax rotation registry

knmcguire, 2021-07-15:
Hi! Th [HIMAX_IMG_ORIENTATION](https://github.com/GreenWaves-Technologies/gap_sdk/blob/278fd42563424c3557ed61d8d997d9a7bdcc1530/rtos/pmsis/pmsis_bsp/camera/himax/himax.h#L32) registry doesn't seem to work properly for the grayscale himax camera on the AIdeck 1.1. Is there a chance that this is supposed to be a different address?

Yaooooo, 2021-07-19:
Hi, due to the NDA with Himax, we cannot share the datasheet. But we can confirm it's aligned with what we are using in Himax BSP (The file you are sharing):
#define         HIMAX_IMG_ORIENTATION     0x0101

knmcguire, 2021-07-29:
Sorry for the delay but I have an update.

If I change the [registry in the driver directly](https://github.com/GreenWaves-Technologies/gap_sdk/blob/278fd42563424c3557ed61d8d997d9a7bdcc1530/rtos/pmsis/pmsis_bsp/camera/himax/himax.c#L136) to 0x03 and remake the pmsis-bsp in the sdk, it indeed rotates. 

However, since the aideck is a bit of a special case, we thought it would be handy to rotate the camera after the camera has been initialized like in this [ wifi image streaming example](https://github.com/bitcraze/AIdeck_examples/blob/91dc28ccb4ea39457262cf287013920153c4bb22/GAP8/test_functionalities/wifi_jpeg_streamer/test.c#L172-L182). This used to be possible in previous SDKs (not exactly sure where it went wrong) but it is not possible anymore.

So when would be the good time to set this registry? Or is it supposed to only been done when initializing?

knmcguire, 2021-08-03:
So on a different thread, @Yaooooo found out that PI_CAMERA_CMD_STOP puts the camera to sleep and that the registries setting don't work in there. So the camera has to be started first before the registry is set. This at least solves our problem for now.

@Yaooooo there was still something weird with the registry setting. Do you want to still discuss that in this thread or shall I close this issue?

knmcguire, 2021-08-05:
oke great thanks!

closed 2021-08-05 by knmcguire