GreenWaves-Technologies/gap_sdk #251: Error: LinearTileNAddressAndSizeInternal: Internal error 3

paulxiongfree, 2021-07-17:
Hello,

I tried to use my own model for the mnist example of nntool. It contains the upsampling and concatenate operations and I got this error:

```
Generating Code For User Kernel:             S1_Op_input_1_formatter
Generating Code For User Kernel:             S5_Conv2d_32x1x3x3_Relu
Generating Code For User Kernel:            S8_Conv2d_32x32x3x3_Relu
Generating Code For User Kernel:           S11_Conv2d_64x32x3x3_Relu
Generating Code For User Kernel:     S13_Op_CONCATENATION_0_3_TOUT_0
Generating Code For User Kernel:                  S16_Linear_10x4704
Error: LinearTileNAddressAndSizeInternal: Internal error 3
Execution aborted
```

Attached please check my .tflite model. I just substitute the original file in the model directory of mnist with this one. 
Could you please help me to get around this error? Thank you!

[mnist.tflite.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/6834175/mnist.tflite.zip)

sousoux, 2021-07-19:
I just tried this (admittedly with our current internal head) and it worked.

`(NNT mnist.tflite 0) performance
generating project from graph
compiling and running project with 8 threads
network performance
+-----------------------------------+---------+-----------+-----------+----------+
| Layer                             | Cycles  |    Ops    | Ops/Cycle | % cycles |
+===================================+=========+===========+===========+==========+
| S4_Conv2d_32x1x3x3_Relu           |  23,775 |    62,720 |     2.638 |    6.987 |
+-----------------------------------+---------+-----------+-----------+----------+
| S7_Conv2d_32x32x3x3_Relu          | 107,558 |   453,152 |     4.213 |   31.611 |
+-----------------------------------+---------+-----------+-----------+----------+
| S10_Conv2d_64x32x3x3_Relu         | 188,795 |   906,304 |     4.800 |   55.486 |
+-----------------------------------+---------+-----------+-----------+----------+
| S14_Op_FULLY_CONNECTED_0_5_fusion |  16,756 |    47,050 |     2.808 |    4.924 |
+-----------------------------------+---------+-----------+-----------+----------+
| S15_SoftMax                       |   3,375 |        10 |     0.003 |    0.992 |
+-----------------------------------+---------+-----------+-----------+----------+
| Total                             | 340,259 | 1,469,236 |     4.318 |    100.0 |
+-----------------------------------+---------+-----------+-----------+----------+
(NNT mnist.tflite 0) 
`

We will check if we have a mismatch between the released code and our current internal head.

paulxiongfree, 2021-07-26:
I tried resetting the head to the former version and it doesn't work. I am at HEAD e6e64e64 Release v4.0.0 now. Could you please help me with that network? If it works, maybe I should the problem on my side.

sousoux, 2021-09-21:
Close due to inactivity

closed 2021-09-21 by sousoux