GreenWaves-Technologies/gap_sdk #252: Flashing hangs, if image contains files for hyperflash on AI Deck with SDK 3.9.1 and 4.0.0

neoamos, 2021-07-18:
OS: Ubuntu 18.04
Board: AI deck 1.0
SDK versions: 3.9.1, 4.0.0

The is the output from flashing the [AI deck MNIST example ](https://github.com/bitcraze/AIdeck_examples/tree/master/GAP8/ai_examples/gapflow_model_extern)
```
Flashing image with command:
gap8-openocd -c "gdb_port disabled; telnet_port disabled; tcl_port disabled" -c "script interface/jlink.cfg; script target/gap8revb.tcl; script tcl/flash_image.tcl; script tcl/jtag_boot.tcl; gap_flash_raw_hyper /module/data/BUILD/GAP8_V2/GCC_RISCV/target.board.devices.flash.img 284164 /gap_sdk/tools/gap8-openocd-tools; exit;"
Open On-Chip Debugger 0.10.0+dev-gb84d97ec4 (2021-07-18-12:02)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
Warn : Interface already configured, ignoring
TARGET create
Info : core 0 selected
Info : gap8_adv_debug_itf tap selected
Info : adv_dbg_unit debug unit selected
Info : Option 7 is passed to adv_dbg_unit debug unit
GAP8 INIT TARGET
Info : J-Link V11 compiled Aug 14 2019 16:21:09
Info : Hardware version: 11.00
Info : VTarget = 2.864 V
Info : clock speed 1500 kHz
Info : JTAG tap: gap8.cpu tap/device found: 0x149511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x1)
GAP8 examine target
Init jtag
Initialising GAP8 JTAG TAP
Info : adv debug unit is configured with option BOOT MODE JTAG
Info : adv debug unit is configured with option ADBG_USE_HISPEED
Info : gdb port disabled
--------------------------
begining flash session (hyperflash)
--------------------------
load flasher to L2 memory
Loading binary through JTAG
Warn : Burst read timed out
[Flahser]: Hyperflash flasher is ready
Instruct flasher to begin flash per se
device struct address is 470047304
Warn : Burst read timed out
going to wait on addr GAP_RDY
wait on gap_rdy done witg buff ptr 0 469784956
loading image with addr 469784956 addr_min 469784956 and size 262144
load image done
```

This is similar to issue #203 but with the AI deck.  I was hoping 4.0.0 would solve the issue, but it did not.  It works with 3.8.1.
Also see [this issue](https://github.com/bitcraze/AIdeck_examples/issues/41) on the AI deck repository.

Yaooooo, 2021-07-29:
Hi,

Sorry for late reply. In fact, there's an regression and has been fixed since a while on branch 4.0_dev.
I just merge the branch to master, you can give a try.

Yao

neoamos, 2021-08-03:
Yes its fixed now!  Thanks.

closed 2021-08-03 by neoamos