GreenWaves-Technologies/gap_sdk #255: Error opening custom tflite model with NNtool

rallen10, 2021-07-22:
This is a follow on issue to https://github.com/GreenWaves-Technologies/gap_sdk/issues/253, but encountering a new error I have been unable to debug. See related discussion [here](https://github.com/mlech26l/keras-ncp/issues/21)

[hello_ncp.zip](https://github.com/GreenWaves-Technologies/gap_sdk/files/6859890/hello_ncp.zip)

When I attempt to open the attached `hello_ncp.tflite` model with `nntool`, I get the error (more traceback can be provided if needed):
```bash
File "/home/ross/Projects/AIIA/crazyflie/gap_sdk/tools/nntool/graph/types/others.py", line 701, in get_output_size
    (self.name, in_dim, self.shape))
NotImplementedError: bad reshape STRIDED_SLICE_0_107: in dim 1x8 does not match reshape scalar
EXCEPTION of type 'NotImplementedError' occurred with message: 'bad reshape STRIDED_SLICE_0_107: in dim 1x8 does not match reshape scalar'
```

I don't understand where the reshape is coming from as there are no reshape operations in the graph at that location. I am at a loss how to debug this.

sousoux, 2021-07-22:
That is a bug in the nntool importer (it has nothing to do with the strided slice node), This diff will solve it. This will be in the next release.
```
diff --git a/importer/tflite2/handlers/backend/reducer_mixin.py b/importer/tflite2/handlers/backend/reducer_mixin.py
index afd4f38..ff1885f 100644
--- a/importer/tflite2/handlers/backend/reducer_mixin.py
+++ b/importer/tflite2/handlers/backend/reducer_mixin.py
@@ -51,7 +51,10 @@ class ReducerMixin(object):
             params = NoOPParameters(node.name)
             pout_shape = x_shape.copy()
         else:
-            pout_shape = [1 if idx in axes and dim is not None else dim for idx, dim in enumerate(x_shape)]
+            if node_opts.KeepDims():
+                pout_shape = [1 if idx in axes and dim is not None else dim for idx, dim in enumerate(x_shape)]
+            else:
+                pout_shape = [dim for idx, dim in enumerate(x_shape) if idx not in axes]
             # subtract 1 from axis for all None's preceeding it and remove
             # axes that are not defined
             axes = [ax - sum([1 if dim is None else 0 for dim in x_shape[:ax:]])

```

you will need to use fusions -a expression_matcher in your nntool script to produce code for this. The performance will not be good at all (that said the performance in the TFLITE runtime would probably be worse). The better way to handle this would be to write a custom GAP kernel for the LTC Cell operation in the NCP network and then match and export that as a custom tflite operation in TOCO. I'm afraid that is a lot of work. This really highlights one of the big problems with (all) the export formats. The lack of a way of tagging a group of operators as originating from a 'meta-operator' so that they can be well matched to a kernel. It is something that we are thinking about and have requested as a feature in ONNX.

sousoux, 2021-07-22:
By the way that is in tools/nntool in the SDK

rallen10, 2021-07-22:
@sousoux thank you for the help; this seems to fix the import process! For posterity, here is the exact commands for applying this patch

```
git clone git@github.com:GreenWaves-Technologies/gap_sdk.git
cd gap_sdk/tools/nntool 
# <copy the above diff into a file called reducer_mixin.diff in gap_sdk/tools/nntool>
patch -p1 < ./reducer_mixin.diff
```

closed 2021-07-22 by rallen10